nathanmarlor/foxess_modbus #529: Exception Response(131, 3, IllegalAddress) with v.1.10.0b4 : Ethernet connected Inverter

Carl-mgb66, 2024-01-28:
## Version of the Custom Component
v.1.10.0b4

## Configuration

```yaml
Standard install - H1-5.0-E Inverter connected via Ethernet
Master Version: 1.48
Slave Version: 1.02
Manager Version: 1.45
```

## Describe the bug

A clear and concise description of what the bug is.

On installing the latest versions, on restart, initially all appears to be OK, but after a short while I get the following on the Connection Status:
```text
Error reading registers. Type: RegisterType.HOLDING; start: 44008; count: 5; slave: 247 from tcp://192.168.50.228:502: Exception Response(131, 3, IllegalAddress)
```
## Debug log

```text

2024-01-28 20:03:15.101 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses on tcp://192.168.50.228:502 247: (31000, 39)
2024-01-28 20:03:15.241 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses on tcp://192.168.50.228:502 247: (44008, 5)
2024-01-28 20:03:15.348 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Modbus error when polling tcp://192.168.50.228:502 247: Exception Response(131, 3, IllegalAddress)
2024-01-28 20:03:15.491 DEBUG (MainThread) [custom_components.foxess_modbus.client.modbus_client] Closing connection to modbus on tcp://192.168.50.228:502
2024-01-28 20:03:25.529 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses on tcp://192.168.50.228:502 247: (31000, 39)
2024-01-28 20:03:25.531 DEBUG (SyncWorker_4) [custom_components.foxess_modbus.client.custom_modbus_tcp_client] Connecting to ModbusBaseClient._params(retries=3, retry_on_empty=False, close_comm_on_error=False, strict=True, broadcast_enable=False, reconnect_delay=None, source_address=None, server_hostname=None)
2024-01-28 20:03:26.653 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses on tcp://192.168.50.228:502 247: (44008, 5)
2024-01-28 20:03:26.755 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Modbus error when polling tcp://192.168.50.228:502 247: Exception Response(131, 3, IllegalAddress)
2024-01-28 20:03:26.755 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Writing registers for tcp://192.168.50.228:502 247: (44000, [0])

2024-01-28 20:03:35.529 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses on tcp://192.168.50.228:502 247: (31000, 39)
2024-01-28 20:03:35.644 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses on tcp://192.168.50.228:502 247: (44008, 5)
2024-01-28 20:03:35.750 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Modbus error when polling tcp://192.168.50.228:502 247: Exception Response(131, 3, IllegalAddress)
2024-01-28 20:03:45.530 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses on tcp://192.168.50.228:502 247: (31000, 39)
2024-01-28 20:03:45.642 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses on tcp://192.168.50.228:502 247: (44008, 5)
2024-01-28 20:03:45.791 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Modbus error when polling tcp://192.168.50.228:502 247: Exception Response(131, 3, IllegalAddress)
2024-01-28 20:03:55.531 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses on tcp://192.168.50.228:502 247: (31000, 39)
2024-01-28 20:03:55.635 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses on tcp://192.168.50.228:502 247: (44008, 5)
2024-01-28 20:03:55.747 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Modbus error when polling tcp://192.168.50.228:502 247: Exception Response(131, 3, IllegalAddress)
2024-01-28 20:04:05.531 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses on tcp://192.168.50.228:502 247: (31000, 39)
2024-01-28 20:04:05.640 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses on tcp://192.168.50.228:502 247: (44008, 5)
2024-01-28 20:04:05.786 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Modbus error when polling tcp://192.168.50.228:502 247: Exception Response(131, 3, IllegalAddress)
2024-01-28 20:04:05.786 WARNING (MainThread) [custom_components.foxess_modbus.modbus_controller] tcp://192.168.50.228:502 247 - 5 failed poll attempts: now not connected. Last error: Error reading registers. Type: RegisterType.HOLDING; start: 44008; count: 5; slave: 247 from tcp://192.168.50.228:502: Exception Response(131, 3, IllegalAddress)

```

canton7, 2024-01-28:
Interesting! Mind jumping over to https://github.com/nathanmarlor/foxess_modbus/discussions/513#discussioncomment-8274389 (so we can see all of the discussion in one place)? Thanks!

FozzieUK, 2024-01-28:
Wondering if that might be because of early firmware versions, it’a working ok on my H1 with Manager 1.70

canton7, 2024-01-28:
@FozzieUK Are you using LAN, or RS485?

canton7, 2024-02-03:
This should be fixed in v1.10.0b6: mind giving that a try?

Gics78, 2024-02-06:
Same problem here! v1.10.0b6. HA Core (Docker) via Waveshare RS485 ethernet. 
I use Modbus to read to Symcon and EVCC. Those work without a problem. I even closed those to rule out interference - no luck!

Is there anything I might try or doing wrong?

[home-assistant_foxess_modbus_2024-02-06T20-47-02.478Z.log](https://github.com/nathanmarlor/foxess_modbus/files/14185980/home-assistant_foxess_modbus_2024-02-06T20-47-02.478Z.log)

canton7, 2024-02-06:
What version of foxess_modbus are you running?

Gics78, 2024-02-06:
> What model of inverter do you have?

Ups, sorry: H3-10

canton7, 2024-02-15:
Try v1.10.0b7, see if that improves things.

Carl-mgb66, 2024-02-15:
> Try v1.10.0b7, see if that improves things.

Installed - I'll update if I see anything.

Gics78, 2024-02-15:
Installed too, H3-10. No errors at first, then when I tried to force charge some errors showed up. But maybe I wasn't patient enough. But sure enough Foxcloud confirmed that I had changened modus from self use to backup and min Soc on Grid to 100% while trying.

Hope the debug log helps.
And last but not least: Thank you very much!

[home-assistant_foxess_modbus_2024-02-15T20-49-06.768Z.log](https://github.com/nathanmarlor/foxess_modbus/files/14302335/home-assistant_foxess_modbus_2024-02-15T20-49-06.768Z.log)

canton7, 2024-02-15:
Interesting, thanks! I'll go though that tomorrow, and how it differs from other H3s I've seen which supported that register.

What manager version do you have?

Gics78, 2024-02-15:
Master 1.83, Slave 1.03, Comm 1.75

canton7, 2024-02-16:
It's really odd. Most of the time you're failing to write register 44000, which we know should be writable on the H3. Then just sometimes it succeeds.

But, then your attempts to change Min Soc on Grid were also failing sometimes. Is that something which normally succeeds?

Gics78, 2024-02-16:
> It's really odd. Most of the time you're failing to write register 44000, which we know should be writable on the H3. Then just sometimes it succeeds.
> 
> But, then your attempts to change Min Soc on Grid were also failing sometimes. Is that something which normally succeeds?

It might be related to a kind of timeout issue: for instance when I try to set Min Soc on Grid by clicking on the up/down arrows multiple times in a row (ie 10 times from 20% to 30%) the errors pop up, but in the end Min Soc on Grid is set correctly as verified in the Cloud App. When I type in the requested % directly it works flawlessly.

MartB, 2024-02-25:
Im not sure if thats fox playing around with my units or a bug in the latest beta but my Remote Control setting changed itself to disabled. Just leaving this here in case it happens to anyone else.

canton7, 2024-02-25:
What exactly do you mean by "my remote control setting"?

MartB, 2024-02-25:
The one in the inverter settings called "Remote Control" Enabled | Disabled

icornish72, 2024-05-17:
I am new to FoxEss / PV etc, having had an install of H1-6.0-E-G2 and ECS batteries and Panels earlier this week. Just connected an Elfin dongle which is communicating well with the box, but I get the same error with the integration.
My Versions are:-
Master: 1.14
Slave: 1.02
Comm: 1.17

FozzieUK, 2024-05-17:
@icornish72 your firmware versions need to be updated.

Can you ask your installer or drop an email to Fox uk service to upgrade your inverter firmware to the latest versions (for the H1 G2 they are - Master 1.22, Manager 1.37, Slave 1.02)

If you could then post back if it’s working 😉, if you are still having problems could you open a new issue and the integration version you are using so we can track it separately please.

JHolah, 2024-05-22:
I had set up all my hardware (Waveshare Modbus POE connected to H1 Inverter) and set everything correctly. Had exactly the same issue as described here. After confirming everything was connected and configured correctly, only a Firmware upgrade fixed it.

Vato9394, 2024-07-10:
I think this issue somehow has to do with that "remote control" setting that is available on the H3 units.
Can you add force charge and discharge again for H3 units (Maybe in a beta version?) so we can test this? Thanks

MuttzNuttz, 2025-03-02:
I am having a similar issue:
KH8 Inverter connected via Waveshare RS485 to Eth (B)
This happens within 2 minutes of restarting HA, I have tried all versions including the beta(s) upto 1.13.5b3 with no luck so far.
Model KH8
Master 1.34
Slave 1.01
ARM 1.28

Connection Status (Solar Monitor) changed to Error reading registers. Type: RegisterType.HOLDING; start: 36001; count: 3; slave: 247 from tcp://10.10.9.245:502: Exception Response(131, 3, IllegalAddress)

FozzieUK, 2025-03-02:
> I am having a similar issue: KH8 Inverter connected via Waveshare RS485 to Eth (B) This happens within 2 minutes of restarting HA, I have tried all versions including the beta(s) upto 1.13.5b3 with [...]

You are on an older manager(ARM) version than is supported by default - you have to go into advanced settings as described here https://github.com/nathanmarlor/foxess_modbus/wiki/Supported-Versions and set the integration for your older firmware version.

MuttzNuttz, 2025-03-04:
> You are on an older manager(ARM) version than is supported by default - you have to go into advanced settings as described here https://github.com/nathanmarlor/foxess_modbus/wiki/Supported-Versions [...]

Thanks for the info, I will give this a try.