nathanmarlor/foxess_modbus #532: Bug Report V1.10.0b5

sleepymaxx, 2024-01-29:
There ar no more Readings with this release sorry

Error reading registers. Type: RegisterType.HOLDING; start: 44008; count: 5; slave: 247 from tcp://192.168.2.49:502: Exception Response(131, 3, IllegalAddress)

-->

V1.10.0b5

-->

## Configuration

```yaml
Normal Configuration through Web Interface
```

## Describe the bug
NO more readings in the Whole integration

## Debug log

Dieser Fehler wurde von einer benutzerdefinierten Integration verursacht

Logger: custom_components.foxess_modbus.modbus_controller
Source: custom_components/foxess_modbus/modbus_controller.py:144
Integration: FoxESS - Modbus (documentation, issues)
First occurred: 19:13:43 (1 occurrences)
Last logged: 19:13:43

Failed to write registers
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 8 bytes (0 received)

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/config/custom_components/foxess_modbus/modbus_controller.py", line 144, in write_registers
    await self._client.write_registers(start_address, values, self._slave)
  File "/config/custom_components/foxess_modbus/client/modbus_client.py", line 179, in write_registers
    raise ModbusClientFailedError(message, self, response) from response
custom_components.foxess_modbus.client.modbus_client.ModbusClientFailedError: Error writing registers. Start: 44000; values: [0]; slave: 247 from tcp://192.168.2.49:502: Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 8 bytes (0 received)

```text

Thank you again very very much for your work on this Integration

```

canton7, 2024-01-29:
Thanks!

What inverter model do you have, what is the manager version, and how are you connected to it?

canton7, 2024-01-29:
Interesting. They must not have implemented the remote control functionality. I'll remove support for that inverter.

sleepymaxx, 2024-01-30:
tried to reinstall the whole thing no really good changings here -went back to v1.9.0 great sucess 

[image]

Logfile of the "false / not functioning version : v1.10.0b5
[home-assistant_2024-01-30T15-47-17.632Z.log](https://github.com/nathanmarlor/foxess_modbus/files/14100804/home-assistant_2024-01-30T15-47-17.632Z.log)

sleepymaxx, 2024-01-30:
you can not imagine how much i apreciate your work  thanks a lot!

canton7, 2024-01-30:
Leaving open until I fix this

commit b99624c6 (2024-02-03): Remove support for the Kuara H3

Fixes: #532
closed 2024-02-03 by canton7