nathanmarlor/foxess_modbus #551: Can't install v1.10

AdamGilchrist317, 2024-02-24:
v1.10beta8 was working fine, 1.10 doesn't seem to. I think error is: "Unable to determine capacity of inverter"

Hope this helps!

Debug log:

2024-02-24 11:54:36.278 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry FoxESS - Modbus for foxess_modbus
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 444, in async_setup
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/foxess_modbus/__init__.py", line 123, in async_setup_entry
    create_controller(client, inverter)
  File "/config/custom_components/foxess_modbus/__init__.py", line 79, in create_controller
    controller = ModbusController(
                 ^^^^^^^^^^^^^^^^^
  File "/config/custom_components/foxess_modbus/modbus_controller.py", line 106, in __init__
    self._inverter_capacity = connection_type_profile.inverter_model_profile.inverter_capacity(
                              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/foxess_modbus/inverter_profiles.py", line 173, in inverter_capacity
    raise Exception(f"Unable to determine capacity of inverter '{inverter_model}'")
Exception: Unable to determine capacity of inverter 'H1-5.0'
2024-02-24 11:54:48.358 WARNING (MainThread) [custom_components.foxess_em.battery.battery_controller] Battery state is unknown
2024-02-24 11:55:13.196 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry FoxESS - Modbus for foxess_modbus
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 444, in async_setup
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/foxess_modbus/__init__.py", line 123, in async_setup_entry
    create_controller(client, inverter)
  File "/config/custom_components/foxess_modbus/__init__.py", line 79, in create_controller
    controller = ModbusController(
                 ^^^^^^^^^^^^^^^^^
  File "/config/custom_components/foxess_modbus/modbus_controller.py", line 106, in __init__
    self._inverter_capacity = connection_type_profile.inverter_model_profile.inverter_capacity(
                              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/foxess_modbus/inverter_profiles.py", line 173, in inverter_capacity
    raise Exception(f"Unable to determine capacity of inverter '{inverter_model}'")
Exception: Unable to determine capacity of inverter 'H1-5.0'
2024-02-24 11:55:13.209 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry FoxESS - Modbus for sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 444, in async_setup
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 141, in async_setup_entry
    return await component.async_setup_entry(entry)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 183, in async_setup_entry
    raise ValueError("Config entry has already been setup!")
ValueError: Config entry has already been setup!
2024-02-24 11:55:13.214 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry FoxESS - Modbus for binary_sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 444, in async_setup
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/binary_sensor/__init__.py", line 238, in async_setup_entry
    return await component.async_setup_entry(entry)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 183, in async_setup_entry
    raise ValueError("Config entry has already been setup!")
ValueError: Config entry has already been setup!
2024-02-24 11:55:13.219 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry FoxESS - Modbus for select
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 444, in async_setup
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/select/__init__.py", line 109, in async_setup_entry
    return await component.async_setup_entry(entry)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 183, in async_setup_entry
    raise ValueError("Config entry has already been setup!")
ValueError: Config entry has already been setup!
2024-02-24 11:55:13.226 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry FoxESS - Modbus for number
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 444, in async_setup
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/number/__init__.py", line 119, in async_setup_entry
    return await component.async_setup_entry(entry)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 183, in async_setup_entry
    raise ValueError("Config entry has already been setup!")
ValueError: Config entry has already been setup!

canton7, 2024-02-24:
Thanks! Fix incoming...

canton7, 2024-02-24:
Right, should be fixed in 1.10.1.

closed 2024-02-24 by canton7

AdamGilchrist317, 2024-02-24:
Many thanks!

PR #552 (2024-02-24): Inverter models don't always end in '-E' etc
Fixes: #551
commit 9b37686f (2024-02-24): Inverter models don't always end in '-E' etc

Fixes: #551