nathanmarlor/foxess_modbus #554: Ver 1.10.1 breaks on KH10.5 & AIO 6KW

MJW911, 2024-02-24:
## Version of the custom_component

 ver 1.10.1

## Configuration

```yaml
Add your logs here.
```

## Describe the bug

the new version fails completely on both versions of inverter. the USB - RS485 device fails to be recognised by the integration, and none of the entities are readable.

## Debug log

```text

This error originated from a custom integration.

Logger: custom_components.foxess_modbus.modbus_controller
Source: custom_components/foxess_modbus/modbus_controller.py:315
Integration: FoxESS - Modbus (documentation, issues)
First occurred: 4:31:46 PM (2 occurrences)
Last logged: 4:41:20 PM

/dev/ttyUSB0 247 - 5 failed poll attempts: now not connected. Last error: Error reading registers. Type: RegisterType.HOLDING; start: 31040; count: 15; slave: 247 from /dev/ttyUSB0: Exception Response(131, 3, IllegalAddress)

```

MJW911, 2024-02-24:
This error originated from a custom integration.

Logger: homeassistant
Source: custom_components/foxess_modbus/client/modbus_client.py:205
Integration: FoxESS - Modbus (documentation, issues)
First occurred: 5:13:05 PM (1 occurrences)
Last logged: 5:13:05 PM

Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/config/custom_components/foxess_modbus/modbus_controller.py", line 328, in _refresh
    await self._remote_control_manager.poll_complete_callback()
  File "/config/custom_components/foxess_modbus/remote_control_manager.py", line 335, in poll_complete_callback
    await self._update()
  File "/config/custom_components/foxess_modbus/remote_control_manager.py", line 92, in _update
    await self._update_disable()
  File "/config/custom_components/foxess_modbus/remote_control_manager.py", line 101, in _update_disable
    await self._disable_remote_control()
  File "/config/custom_components/foxess_modbus/remote_control_manager.py", line 317, in _disable_remote_control
    await self._controller.write_register(self._addresses.remote_enable, 0)
  File "/config/custom_components/foxess_modbus/modbus_controller.py", line 174, in write_register
    await self.write_registers(address, [value])
  File "/config/custom_components/foxess_modbus/modbus_controller.py", line 211, in write_registers
    raise ex
  File "/config/custom_components/foxess_modbus/modbus_controller.py", line 195, in write_registers
    await self._client.write_registers(start_address, values, self._slave)
  File "/config/custom_components/foxess_modbus/client/modbus_client.py", line 168, in write_registers
    response = await self._async_pymodbus_call(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/foxess_modbus/client/modbus_client.py", line 205, in _async_pymodbus_call
    result = await self._hass.async_add_executor_job(call, *args)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pymodbus/client/mixin.py", line 152, in write_register
    return self.execute(
           ^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pymodbus/client/base.py", line 444, in execute
    raise ConnectionException(f"Failed to connect[{self!s}]")
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Failed to connect[ModbusSerialClient(<pymodbus.framer.rtu_framer.ModbusRtuFramer object at 0xffff56ab1190> baud[9600])]
\

MJW911, 2024-02-24:
This error originated from a custom integration.

Logger: custom_components.foxess_modbus.modbus_controller
Source: custom_components/foxess_modbus/modbus_controller.py:195
Integration: FoxESS - Modbus (documentation, issues)
First occurred: 5:13:05 PM (1 occurrences)
Last logged: 5:13:05 PM

Failed to write registers
Traceback (most recent call last):
  File "/config/custom_components/foxess_modbus/modbus_controller.py", line 195, in write_registers
    await self._client.write_registers(start_address, values, self._slave)
  File "/config/custom_components/foxess_modbus/client/modbus_client.py", line 168, in write_registers
    response = await self._async_pymodbus_call(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/foxess_modbus/client/modbus_client.py", line 205, in _async_pymodbus_call
    result = await self._hass.async_add_executor_job(call, *args)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pymodbus/client/mixin.py", line 152, in write_register
    return self.execute(
           ^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pymodbus/client/base.py", line 444, in execute
    raise ConnectionException(f"Failed to connect[{self!s}]")
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Failed to connect[ModbusSerialClient(<pymodbus.framer.rtu_framer.ModbusRtuFramer object at 0xffff56ab1190> baud[9600])]

MJW911, 2024-02-24:
> What manager version does your KH have?

Mnager version 1.13. which version should it have for the new Modbus integration?

MJW911, 2024-02-24:
ive rolled back to ver 1.9 which seems to be working correctly.

canton7, 2024-02-24:
From the first line of the release notes:

> ⚠️  This version adds support for KH Manager version 1.19+, but **removes support for earlier versions**. Update your KH if necessary.

Unfortunately FoxESS broke the KH in Manager version 1.19, and I had to break support in the integration to keep up. Please ask your installer to upgrade your KH.

closed 2024-02-24 by canton7