nathanmarlor/foxess_modbus #558: Exit from Force Discharge to Feed in First

AdamGilchrist317, 2024-02-26:
Just a small point:

Going from force discharge to self use works as expected. Going from force discharge to feed in first doesn't do anything. The change to feed in first is logged, but nothing happens (the inverter still discharges the batteries at full speed).

Hope this helps!

(v1.10.1, H1-5.0)

canton7, 2024-02-26:
Can you turn on debug logging, and then reproduce the problem and send me the logs?

Note that it will take a good ~20 seconds before you'll see force discharge stop.

AdamGilchrist317, 2024-02-26:
I've tried this again. I think the problem might be in the inverter, rather than your software. I've noticed before that it sometimes won't stop exporting after exporting at the export limit (from solar power), and will drain the battery (ie feed in first just feeds in forever). 

I tried limiting the export power to 1kW, and it works perfectly. Setting the export power to 5kW (my export limit is 3.68kW), and it doesn't stop exporting.
Here are the logs, in case they help. Force discharge at 9:48:32, Feed in first 9:49:08, still exporting at 9:51:20, changed to self use.

Many thanks
 
[home-assistant_foxess_modbus_2024-02-26T09-51-28.188Z.zip](https://github.com/nathanmarlor/foxess_modbus/files/14402876/home-assistant_foxess_modbus_2024-02-26T09-51-28.188Z.zip)

canton7, 2024-02-26:
Thanks. Which logs exactly did you enable? I'd expect to see extra debug messages in there when remote control is enabled.

The remote control interface has a watchdog on it, which should automatically disable everything if I don't write the export power for 20s, so even in the absence of anything else it should just stop at most 20s after we disable force discharge.

I guess I could additionally set the export power when disabling remote control, just in case.

AdamGilchrist317, 2024-02-26:
I'd enabled debug logging for the modbus component (settings, devices and services, foxess_modbus, enable debug logging). Should I be doing something different?

Thanks

AdamGilchrist317, 2024-02-28:
I'll close this. I'm pretty sure that it's the firmware that's doing this, and it's easy to get around with an automation.

Thanks very much for your excellent software!

closed 2024-02-28 by AdamGilchrist317