nathanmarlor/foxess_modbus #571: Load sensor artificially low after latest beta release.

ryaaanmorgan, 2024-03-18:
Evening lads, noticed my "load" sensor hit rock bottom despite all other sensors (including the CT) being correct.
I've done some digging and really can't explain it. It's not done wonders for my losses sensor 🤣

Any thoughts? It coincided with the upgrade to the latest beta last night and I think that might be a bug.
Nothing in the logs to suggest an issue neither.

[image]

ryaaanmorgan, 2024-03-18:
HA is at version 2024.3.0

ryaaanmorgan, 2024-03-18:
I've moved back to the most recent non-beta release and the load sensor is fixed.
Any details you need to help recreate the error - let me know.

FozzieUK, 2024-03-19:
Morning Ryan, thanks for spotting that :)

@canton7 - looks like an error crept in when refactoring - see https://github.com/nathanmarlor/foxess_modbus/pull/572

closed 2024-03-19 by canton7