nathanmarlor/foxess_modbus #606: Force Discharge/Charge Power input numbers broken in v11

Ndrinta, 2024-04-27:
[image]

Pretty self-explanatory when looking at the picture.

canton7, 2024-04-27:
Presumably that's an old H1, not a H1 G2?

What's the capacity?

(I have a H1 and don't see this)

canton7, 2024-04-27:
What version were you on prior to upgrading to v1.11.0?

canton7, 2024-04-27:
We read the max value from the inverter itself (register 44008). I wonder whether your inverter's started reporting something odd on that address.

We switched which register we read here: https://github.com/nathanmarlor/foxess_modbus/pull/523 . It looks like ac_power_limit_down might not be responding with the right thing in all cases either... Maybe we just fall back to parsing the inverter capacity from the model name in all cases, and scrap that register.

Ndrinta, 2024-04-28:
Hi Antony, yes its a Gen1 (H1-6.0-E).

I think I was in 1.11.0 when I first noticed it.
I am now in 12.0b2 and it all seems fine, might be the restart thought...

Ndrinta, 2024-04-28:
Is there a way to back track the version of an integration (like an entity)?

iancg, 2024-04-28:
I have previously had the same error, but very rarely (and given I control charge and discharge completely with force charge/discharge it must have been relatively very rare, and would correct itself shortly thereafter - when I'm charging/discharging the automation runs every 20 seconds to set the charge/discharge power). At the time I'd assumed it was because of the code that tries to reduce the maximum throughput so that solar production isn't stopped - I thought the version I was on at the time was still parsing the inverter name.  When I saw it, the upper limit was just below the maximum 6.0 - I don't have the error but from memory it was something like 5.95 and I was trying to set 6.

My H1-G6 (gen 1) inverters were recently upgraded to Firmware versions Manager 1.71/Slave1.02/Master 1.64 and I also upgraded the integration from 1.10.0b8 to 1.11.0 (then 1.11.1).  Since then I have not seen the problem again, but it has only been a few days and was rare before that I can't categorically say it has gone.

Is the value of register 44008 exposed as an sensor? If so I can potentially look back through the history of the values of mine.  If not, it might be useful to do so, both for this, but I could also remove the hard coded limit numbers from my automations.

canton7, 2024-04-28:
> Is there a way to back track the version of an integration (like an entity)?

HACS might keep track, I'm not sure. The integration itself doesn't know its own version.

canton7, 2024-04-28:
I've removed reliance on the ac_power_limit_down register to give the inverter capacity: it's now parsed straight from the inverter model name.

closed 2024-04-28 by canton7

Ndrinta, 2024-04-28:
As always, thanks a lot. You guys are great!
Weekends are always productive! Haha

PR #607 (2024-04-28): Remove use of ac_power_limit_down registers
Fixes: #606
commit 05fe4947 (2024-04-28): Remove the ac_power_limit_down register references

This register seems to not do what we expect occasionally. Fall back to
just parsing the inverter capacity from the model name.

Fixes: #606