nathanmarlor/foxess_modbus #637: since update to HASS 2024.7 there is no more sensor data, the controls are working

Runesmacher, 2024-07-03:
## v1.11.1

## Configuration

```yaml
Set up with UI
```

## Describe the bug

HASS update 2024.7 broke the plugin somehow, the sensors can't be created anymore, see hass log below:

```text
Error while setting up foxess_modbus platform for sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 364, in _async_setup_platform
    await asyncio.shield(awaitable)
  File "/config/custom_components/foxess_modbus/sensor.py", line 26, in async_setup_entry
    async_add_devices(create_entities(SensorEntity, controller))
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/foxess_modbus/inverter_profiles.py", line 330, in create_entities
    return inverter_connection_type_profile_from_config(controller.inverter_details).create_entities(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/foxess_modbus/inverter_profiles.py", line 95, in create_entities
    entity = entity_factory.create_entity_if_supported(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/foxess_modbus/entities/modbus_integration_sensor.py", line 53, in create_entity_if_supported
    return ModbusIntegrationSensor(
           ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/foxess_modbus/entities/modbus_integration_sensor.py", line 84, in __init__
    IntegrationSensor.__init__(
TypeError: IntegrationSensor.__init__() missing 1 required keyword-only argument: 'max_sub_interval'
```

## Debug log

```text

File "/config/custom_components/foxess_modbus/sensor.py", line 26, in async_setup_entry
File "/config/custom_components/foxess_modbus/inverter_profiles.py", line 330, in create_entities
File "/config/custom_components/foxess_modbus/inverter_profiles.py", line 95, in create_entities
File "/config/custom_components/foxess_modbus/entities/modbus_integration_sensor.py", line 53, in create_entity_if_supported
File "/config/custom_components/foxess_modbus/entities/modbus_integration_sensor.py", line 84, in __init__
2024-07-03 21:41:40.064 ERROR (MainThread) [homeassistant.components.sensor] Error while setting up foxess_modbus platform for sensor
File "/config/custom_components/foxess_modbus/sensor.py", line 26, in async_setup_entry
File "/config/custom_components/foxess_modbus/inverter_profiles.py", line 330, in create_entities
File "/config/custom_components/foxess_modbus/inverter_profiles.py", line 95, in create_entities
File "/config/custom_components/foxess_modbus/entities/modbus_integration_sensor.py", line 53, in create_entity_if_supported
File "/config/custom_components/foxess_modbus/entities/modbus_integration_sensor.py", line 84, in __init__
2024-07-03 21:47:00.094 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses on tcp://10.0.20.20:502 247: (11000, 37)
2024-07-03 21:47:00.222 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses on tcp://10.0.20.20:502 247: (41000, 12)
2024-07-03 21:47:00.310 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses on tcp://10.0.20.20:502 247: (44008, 5)
2024-07-03 21:47:00.384 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Refresh of tcp://10.0.20.20:502 247 complete - notifying sensors: {11008, 41000, 41001, 41002, 41003, 41004, 41005, 41006, 44008, 44012, 41009, 41010, 41011, 11000, 11003, 11036}

```

canton7, 2024-07-03:
Dup of #635. Please search open issues first

Runesmacher, 2024-07-03:
I searched the issues ;) Didnt notice there where discussions 😅 Good that you're on it 😄 Thanks for the good work