nathanmarlor/foxess_modbus #638: Hass 2024.7 broke this integration

canton7, 2024-07-03:
See #635.

Please do not open other issues!

R1200G, 2024-07-04:
All my entities are not available. Integration not working anymore..

Ndrinta, 2024-07-04:
To "fix" this, at least until the amazing devs will release an actual fix, simply open the terminal and downgrade using the command:

> --version 2024.6.4

Which is the latest version of HA known working with this integration.

Rill189, 2024-07-04:
Thanks, although my PV sensor won't now display in history?
Will future updates likely be broken due to the extra code added?

muhkuh666, 2024-07-05:
Is a fix already online when upgrading HASS ?

canton7, 2024-07-05:
> Is a fix already online when upgrading HASS ?

No. I might have a chance to do a fix this weekend.

Rill189, 2024-07-05:
> > Is a fix already online when upgrading HASS ?
> 
> No. I might have a chance to do a fix this weekend.

Thanks for the quick response. Would any fix you implement break the integration if the manual fix has been applied above?

Thanks

Rill189, 2024-07-06:
Is it safe to install 2024.7.1?

screenagerbe, 2024-07-06:
> Is it safe to install 2024.7.1?

Nope, I tried it yesterday and I had to revert to v2024.6.4

canton7, 2024-07-06:
Fixed in 1.12.1

sleepymaxx, 2024-07-07:
Again and again --- Thanks for your great work!!!!