nathanmarlor/foxess_modbus #643: unable to connect to modbus

banzoola, 2024-07-12:
Step 1 - wired into modbus using an ethernet cable 1 twisted pair (pin 3 and 4)
Step 2 - wired into modbus to ethernet adapter connecting relevant cables to A+ and B+
Step 3 -  confirmed modbus is accessible via webui and changed relevant settings acording to wiki
Step 4 - Cant intergrate to HA or get a response from modpoll.
Step 5 - I feel like im cursed at the moment!

./modpoll -m enc -p 503 -a 247 -t 3 -0 -1 -c 16 -r 11000 192.168.100.7
modpoll 3.11 - FieldTalk(tm) Modbus(R) Master Simulator
Copyright (c) 2002-2024 proconX Pty Ltd
Visit https://www.modbusdriver.com for Modbus libraries and tools.

Protocol configuration: Encapsulated RTU over TCP, FC4
Slave configuration...: address = 247, start reference = 11000 (PDU), count = 16
Communication.........: 192.168.100.7, port 503, t/o 1.00 s, poll rate 1000 ms
Data type.............: 16-bit register, input register table

Can't reach server/slave! Check TCP/IP and firewall settings.

banzoola, 2024-07-12:
forgot to also confirm slave is set correctly on the inverter

FozzieUK, 2024-07-13:
Just a couple of quick questions / thoughts -

is it an H1 inverter ? (Others have different RS485 pin outs on the meter connector)

Which adapter are you using ?

Is Port 503 correct?, the default setting is normally 502

banzoola, 2024-07-15:
i was testing and changed the port, good spot. I didnt want to add another comment clarifying that. It is a H1 inverter and the guide states pin 3 and pin 4. I haven't confirmed that from the H1 inverters manual though i was assuming this is correct. I have the AC1 however it says its supported here https://github.com/nathanmarlor/foxess_modbus. FoxESS H1 (including AC1 and AIO-H1).

I wonder if the pins are different on the AC1, ill check the user manual and update if its the wrong pins. Thanks

FozzieUK, 2024-07-16:
The AC1 should work just fine if it’s Gen 1, the Gen 2 hybrid works ok if it has latest firmware, so if it’s a Gen 2 AC1 you might need to get your firmware updated to latest.

I can see you are using -enc on the modpoll which suggests your adapter isn’t encoding modbus - which adapter are you using ?

Another thing to check is that your inverter ethernet option is disabled and DHCP is off - it sometimes stops the modbus working if it has been enabled previously.

canton7, 2024-09-18:
Closing as old and no response. Please re-open (with a response to @FozzieUK's comments) if this is still relevant. Thanks!

closed 2024-09-18 by canton7