nathanmarlor/foxess_modbus #650: Inverter fault not registered

MarkRS-UK, 2024-07-19:
My inverter (H1-5.0E-AUX) registered a 10 minute grid voltage fault yet this (otherwise fabulous!) extension doesn't show it.  In fact the entity inverter_fault_code claims that the last update was three days previously.

I opened a discussion about this (https://github.com/nathanmarlor/foxess_modbus/discussions/647) where someone also reported seeing this on an H3 inverter.

My setup is :

FoxESS H1-5.0-E-AUX inverter. Master : 1.63, Slave : 1.02, Manager : 1.70
HA Core 2024.7.2, Supervisor 2024.06.2, Operating System 12.4, Frontend 20240710.0

Can't find the modbus version in the suggested const.py file.  I assume I'm using the latest version as HA doesn't suggest updating and data is flowing even though I'm using HA2024.7.2.

I haven't included any logs since I'm guesing they won't show things that haven't happened.

FozzieUK, 2024-07-19:
ok, thanks - we'll need to do some testing to establish whether it is all alarm codes or just the 10 min avg fault not showing.

FozzieUK, 2024-07-19:
Hmmm, interestingly it works ok on my H1-6.0-E - AUX ? 
.
[image]
.
[image]
.
I'm on the same version of HA as you, and using integration version v1.12.1 (you'll find that in HACS and click on Foxess Modbus)

I have my intergration polling speed set to 6 seconds but I also did a modpoll to confirm the state was maintained and not just transient - it shows there consistently as well.
.
[image]

The only difference I can see is that I am on the latest firmware (Master 1.65, Manager 1.74)

Fox don't provide release notes so we don't know what the later versions include - but I think it would be wise to contact Fox and ask them to update your inverter firmware (and ask them to change your 10min volt average limit or disable the alarm)

MarkRS-UK, 2024-07-19:
Ok thanks, I'll get the updates and then keep an eye on it.  And yes, I'm on 1.12.1 of FoxESS - ModBus.

Why change the limit?  I'm in (very loose) contact with the supplier to monitor and if reqd. fix the voltage level.  Isn't that the better option? Attached graph seems to my not-very-informed eyes to show it's regularly a little too high.  During saver sessions it always gives a fault.  I'm assuming the current faults are due to export in the area (?)

MarkRS-UK, 2024-07-19:
Ooops.  Here's the graph
[image]

FozzieUK, 2024-07-19:
Ok keep this issue open and let us know how it goes after the update.

The high voltage is largely driven by solar and likely is the effect of your own inverter when you are exporting - the inverter will always show between 2 - 5V higher when exporting than the actual grid voltage when measured at your smart meter.

In all honesty 255V isn't that high, the DNO upper limit is 253V (lower 216.2V) and I suspect if they put a logger to measure your grid voltage at the meter it won't be exceeding that so they won't take any action - the nominal voltage your are seeing is around 245V which looks fine. You could safely increase your 10minAvg limit to 257V and avoid the alarms.

MarkRS-UK, 2024-07-19:
Dang!  They've done it already, just over an hour from my email to theirs confirming it done.  I'll let you know what happens.

canton7, 2024-09-18:
Closing as no update. @MarkRS-UK please re-open if this is still an issue.