nathanmarlor/foxess_modbus #651: BMS kWh Remaining sensor stuck at "max value" after BMS FW update.

Ndrinta, 2024-07-19:
Integration Version: 1.12.1

After updating my BMS (HV2600 v2) Battery Master version to the latest (to 1.014 from 1.013) the "BMS kWh Remaining" sensor is stuck at "max value" (9.87 kWh in my case).
This is most likely a problem for FoxESS to fix but an issue nonetheless, so it's here for us to track.
In the meantime if someone could confirm my findings would be appreciated so we can exclude an issue with limited to my system.

FozzieUK, 2024-07-19:
Hi @ryaaanmorgan - I think this is what you saw when you moved to the latest version, a broken kWh remaining sensor.. i’m guessing it’s not been fixed ?

ryaaanmorgan, 2024-07-19:
> Hi @ryaaanmorgan - I think this is what you saw when you moved to the latest version, a broken kWh remaining sensor.. i’m guessing it’s not been fixed ?

Yes this is what I’ve seen. It’s not been fixed unfortunately.

I always assumed the BMS stopped reporting the figure but could equally be a change to the register?

FozzieUK, 2024-07-20:
ok thanks, I had hoped that might have been a quick fix for them - but I guess there is so much new stuff coming out at the moment Fox are overwhelmed 😞 

The modbus register is still working my V1's are reporting BMS kWh remaining ok 👍

Ndrinta, 2024-07-22:
For whatever reason has now changed to a lower value.
I'm not aware of any changes to any systems or devices.

[image]

ryaaanmorgan, 2024-07-22:
Since the sensor stopped working on the 8th of June when I updated the software, mine has dropped a tiny amount but steadily over time.

I wonder if this is linked to cycles or some sort of battery aging factor in the calculation.

[image]

jamesog, 2024-07-31:
Ah, not just me! I have the same issue after I updated my ECS4100 stack to BMU 1.11 a bit over a month ago. Likewise it's occasionally dropping that value a little bit, although it's stayed steady for a few weeks now.

I'm wondering if I'll need to do some more full depth discharge/charge cycles to correct it, but pretty tricky at this time of year!

thetrestons, 2024-08-21:
I had my Mira-HV25 BMS updated yesterday and now the BMS remaining figure is fixed at 8.37 kWh, it was just over 12kWh before the upgrade. I have done a full to empty to full charge cycle and also rebooted the whole system and no changes in the figure for over 36 hours.

Ndrinta, 2024-08-22:
I sent an email to Fox support to see if we can get this sorted... I doubt but it's worth a try.

jamesog, 2024-08-25:
ECS4100 had new firmware versions released a few days ago (BCU 1.015, BMU 1.12), so I've updated my stack. So far it hasn't made a difference to the reported kWh remaining value, but I haven't quite been through a full discharge/charge cycle yet.

Stephens-ely, 2024-09-20:
I’m seeing this same behavior since a firmware update today on a BMS v2 going from 1.013 to 1.016. 

It’s going to screw up a lot of my HA automations!

Ndrinta, 2024-09-20:
Over the last few weeks I sent multiple emails about this specific issue and some other issues too but I am being completely ignored.

The FoxESS customer service has clear preferences when it comes to supporting its customers...
I would never recommend their products.

Ndrinta, 2024-10-02:
Apparently there's a solution to this.

> This is a known change with 1.014. Residual now shows your current battery capacity. To calculate Residual, multiply it by SoC. In my python library and HA integration, there is a switch that configures Residual Handling accordingly.
> You also need to update your Capacity in the battery settings in Energy Stats to match the value you are now seeing
> As previously posted before you updated, this was known and expected. This is how you get the true battery capacity that matches your SoH.

I fixed it this way for now:

``` 
## BMS kWh Remaining (SOH)

        bms_kwh_actual_remaining:
            friendly_name: "BMS kWh Remaining (SOH)"
            unique_id: '7df65af3-d1fd-4c39-a972-144d70636239'
            unit_of_measurement: 'kWh'
            device_class: energy
            value_template: >
              {%- set bms_kwh_remaining = states('sensor.bms_kwh_remaining')|float(default=0)  %}
              {%- set battery_soc = states('sensor.battery_soc')|float(default=0) %}
              {{ (bms_kwh_remaining * battery_soc / 100)|round(3,"floor",0) }}
            availability_template: >
              {{ states('sensor.bms_kwh_remaining')|float(default=0) > 0 and states('sensor.battery_soc')|float(default=0) > 0 }}
```

TonyHoyle, 2025-01-31:
In sensors.yaml