nathanmarlor/foxess_modbus #661: Parallel K series unable to Force Charge/Discharge

williamjeccles, 2024-08-15:
My setup is two KH10.5 wired in parallel (master/slave enabled) and connected to full stacks of HV2600 8* batteries per inverter.
The inverter and batteries are all on the very latest firmware using the latest version of HA and the modbus plugin.

If I set a force discharge/charge period via the Fox Cloud web UI the inverters are able to discharge/charge without issue, no fighting or flapping.

However, I can reproduce the issue if I use HA via the plugin to set the inverters into Force Charge workmode or Force Discharge workmode the inverters start flapping and fighting as per the video example.

https://youtu.be/1LQeiYCD9vs - Repro example video

I have a suspicion that the kW limit is causing the issue for systems that are parallel. Almost like its telling the systems to discharge at say 7kW total instead of per inverter.

[image]

I would like to bypass the UI limit as I believe this might be causing the inverters to flap/fight.
So I'm happy to test and rule out this suspicion if someone could point me which file and line I should comment or adjust to test out my theory.

I believe if I could set the force discharge limit to 21kW then it would allow both inverters to max out their charge/discharge amount instead of fight. It's not a perfect fix but at least it would unblock and validate my current belief.

williamjeccles, 2024-08-15:
@canton7 we discussed this a while back but would like to reopen the topic and see if we can solve it for all parallel users.

FozzieUK, 2024-08-15:
That's an interesting one, with parallel inverters do you set one of them to force charge/discharge and then they 'share' the output ? - as this might be a bit more complex to unravel.

Whilst we're waiting on Antony, it's a quick change to hard code the limit and test the theory, the file you are looking to change is 

https://github.com/nathanmarlor/foxess_modbus/blob/main/custom_components/foxess_modbus/inverter_profiles.py

I think a change in line 77 is all that is needed to test with i.e.

Change
```
            capacity = int(float(capacity_str) * 1000)
```

To this
```
            capacity = int(21000)
```

williamjeccles, 2024-08-15:
> That's an interesting one, with parallel inverters do you set one of them to force charge/discharge and then they 'share' the output ? - as this might be a bit more complex to unravel.
>
> Whilst we're [...]

Thank you! Testing shortly once the free octopus power hour is over.

canton7, 2024-08-15:
We have some inverters (IIRC it was the KH) which do entirely the wrong thing if you tell them to import / export more than their capacity. That's why we detect the capacity of the inverter, and we limit the import/export rate to that power. See https://github.com/nathanmarlor/foxess_modbus/issues/518#issuecomment-1956415753 and posts around it.

canton7, 2024-08-15:
It's entirely possibly that foxess cloud uses a different mechanism to doing force export than we have access to over modbus (which is limit to the remote control registers), and remote control doesn't work properly with master/slave setups.

williamjeccles, 2024-08-15:
Yeh I think it's just broken, the limit increase didn't resolve it from what I can see. Anythng you'd like to help troubleshoot or implement.

FozzieUK, 2024-08-16:
I've been doing some testing on my H1 to see if the schedules uses a similar method to control the inverter, and if so are there any differences.

For now i've only tested force discharge as that is the one of interest, it would appear both use the remote control registers albeit with some differences.

I know Antony knows this, but i'll write down what my testing has shown for the integration and for the scheduler and highlight any differences as i've learned a few things from the scheduler.

**Force Discharge - Integration first.**

The integration uses the remote control registers, specifically -

`Remote Enable` (44000) - master enable switch
`Timeout` (44001) - when this timeout expires it will terminate the action
`Active Power` (44002) - AC output power in watts (covers load first, remainder export)

`Remote Enable` is used as a master switch to enable the action
`Timeout` is set to poll speed * 2 - I assume so that if a poll is missed the next poll will set it prior to timeout.
`Active Power` is set to a positive value in watts for Force Discharge (Charge is negative signed)

When the force discharge mode is set the integration sets the `timeout` value to (poll *2), then `remote enable` to 1, then `active power` to x watts

The integration continually resets the `timeout` value every poll whilst in force discharge

When the work mode is changed, the integration sets `remote enable` to 0 and the force discharge ends.

**Force Discharge - Scheduler.**

The scheduler appears to use the remote registers but it does so in a different way

It manipulates the same 3 registers plus one other

`Remote Enable` (44000) - master enable
`Timeout` (44001) - when this timeout expires it will terminate the action
`Active Power` (44002) - AC output power in watts (covers load first, remainder export)
`Remote Take Effect` (44005) - if set zero no action will take place, when set to 1 timeout countdown starts

The scheduler sets `Timeout` to be the number of seconds that the action will run for i.e. 1 hour = 3,600
The scheduler sets `Remote Enable` to 1 to enable remote control (`Remote Take effect` is zero until the schedule is ready to run)
At the scheduled time it sets `Active Power` to a positive value in watts for Force Discharge which sets `Remote Take effect` to 1 - this starts the force discharge.

No further writes take place and the `timeout` counts downs to zero at which point it will terminate. This sets `Remote Take effect` to 0, which in turn sets `Remote Not Active Reason` (44006) to 2 ('host off line') but leaves all of the other values set

If the schedule is disabled `Remote Take effect` is set to zero the force discharge is terminated - all other registers stay as they were set, `Remote not Active Reason` shows the reason code.

**Differences?**

The integration sets `timeout` value every poll, whereas the scheduler sets it to the duration the action will run for.

The integration sets `Remote Enable` to turn on remote control, the scheduler sets it the first time it runs and leaves it set.

The integration/scheduler sets `Active Power` and as a result the inverter internally sets `Remote Take Effect`.

Once `Remote Take effect` is set the force discharge countdown will begin.

If a schedule is terminated early the `Remote Take effect` register is set to 0 (other registers stay as they were), the `Remote not Active Reason` is set to 1 ('enable set').

The integration writes the `Timeout` register every poll cycle, the scheduler does not change any of the registers once they have been set.

**Problems**

As I recall there may be a problem with reading the `Remote Take effect` register as the KH did not expose registers above 44003 - it would be useful to know if this is still the case and needs to be re-checked on a KH 

44004 is `Timeout Countdown` - this counts down each second, and at zero will terminate action with result in `Remote no active Reason`
44005 is `Remote take effect` - zero no action, set to 1 action starts countdown 
Note: This is a read only register, yet watching the scheduler it sets this to zero when a schedule is terminated - I cannot find how it is doing that as it leaves `Remote Enable` set
44006 is `Remote not Active Reason` - the reason why it's not running.

This command will try and read the 7 holding registers in one read 

modpoll.exe -m tcp -p 502 -a 247 -t 3 -0 -1 -c 7 -r 44000 192.168.x.x

[... 6 lines cut ...]
I'm not sure yet whether not being able to read them will a problem as these values seem to triggered by other actions - so useful when testing.

**Parallel running ?**

The parallel operation is likely managed as a higher function to the remote control, i.e. the scheduler simply asks for force discharge at x power for x seconds and the inverters work this out between them.

This is just a hypothesis but it is possible the action of repeatedly writing to the `Timeout` register is upsetting the higher management function which is why they are 'fighting' (approx every poll cycle)

If this is the case, the KH(s) may balance the force discharge correctly if the registers are set in the same way as the scheduler i.e. set `timeout` to number of seconds, but do not repeatedly write to the `timeout` register.

@williamjeccles the foxess integration can't set them this way as it has no concept of start and end times (yet), but the theory can be tested as an automation using the integrations modbus write register function.

Add the following 3 actions like this -

[image]

With these registers and values, in this order -

44001 set to 60 ' give it 60 seconds of force discharge
44000 set to 1 ' enable it
44002 set to 1000 ' 1000 watts discharge power - the active power setting automatically sets `remote take effect`

This will run a force discharge at 1000 watts for 60 seconds after which the timeout will expire terminating the action.

If it performs without fighting, clearly this mechanism works - if not, then that's a much bigger problem.

FozzieUK, 2024-08-16:
I've just tested remote control to Force Charge and setting a fixed power - it charged my batteries *and* kept solar enabled with spare being used to export.

So it looks like when Fox fixed the bug where force charge disabled solar power - it 'may' have been a bug in the way remote control worked (now resolved) and much of the complexity that had to be added to modulate the power limit to keep solar on, appears not to be required on the latest firmware versions - you just set a fixed negative power and the inverter does the rest.

williamjeccles, 2024-09-12:
Sorry for the delay @FozzieUK, just seen this. working through your last detailed reply and will update you.

williamjeccles, 2024-09-12:
Hey @FozzieUK / @canton7 Good news! writing directly to the registers as you showed in your screenshot worked!
I now have both inverters discharging without fighting. Amazing! I hope confirming this will enable better compatibility with the features in a future release for parallel owners. Maybe a flag during setup for is this a parallel connected system so features and code paths are separated from those single inverter users?  For now I'll just use an automation that writes to the registers directly. I've also tested it with a negative figure for force charging and it works well also without fighting.

FozzieUK, 2024-09-12:
That's brilliant, i'd hoped as much but couldn't test it on a single inverter 👍

williamjeccles, 2024-09-18:
Updated title as it appears to be solo and parallel k series systems that aren't working with the force charge/discharge implementation. Works great if using automation to write to the registers as shown above.

canton7, 2024-09-18:
Just catching up...

Note that "Remote take effect" is read-only. We can't write it from the integration. It's just a status register, saying what the remote control system is doing. The timeout is started (and reset) by writing a power. Re-reading your analysis with that in mind, I'm comfortable that the integration and the scheduler are doing the same thing (except that the scheduler writes a single power with a long time-out, and the integration continuously updates the power, with a short time-out).

So it does sound like it's specifically the logic which the integration adds to avoid the PV clipping which is causing you problems. While I'd love to just rip all of that out, I'm slightly wary of breaking people who currently rely on it... I guess I could make it an advanced setting, but I'm unsure on whether to enable or disable it by default...

canton7, 2024-09-18:
I strongly doubt that setting the power multiple times to the same value is what's confusing it. I rather suspect there's some weirdness with the feedback from the inverter which we use to set the remote control power. Some debug logs (which I don't think we've had yet, despite how long this has been open!) would help clarify!

Chip2786, 2024-09-25:
> With these registers and values, in this order -
>
> 44001 set to 60 ' give it 60 seconds of force discharge 44000 set to 1 ' enable it 44002 set to 1000 ' 1000 watts discharge power - the active power [...]

This ^ saved my life, my force charge/discharge stopped working weeks ago, using this I created my own "macro" to charge/ discharge to/ from grid. Thank you!

(Fox H1-5.0-E-G2)

canton7, 2024-09-28:
> As the KH battery power limit register doesn't exist, when in force charge work mode, I believe the remote control manager is only ever writing max_import_power, it's not able to calculate it as a variable - it will be the same value written to active power each cycle (which resets the timeout).

Ah yes, fair enough.

@williamjeccles Can you try changing [this line](https://github.com/nathanmarlor/foxess_modbus/blob/48b99d7ec859e32d1d271b5d212665f9551bee82/custom_components/foxess_modbus/modbus_controller.py#L120):

```diff
        self._remote_control_manager = (
-            RemoteControlManager(self, remote_control_config, poll_rate) if remote_control_config is not None else None
+            RemoteControlManager(self, remote_control_config, 60) if remote_control_config is not None else None
        )
```

See if that makes it behave any better? It might only go weird once a minute, or it might be able to handle it? Would be interesting to see.

I'm reluctant to have a timeout which is too long if we cna avoid it, as it means that your inverter will be stuck in force charge / force discharge if HA fails.

williamjeccles, 2024-10-20:
For those following with a KH and wanting to update the address directly use something like this as an automation:

actions:
  - parallel:
      - action: foxess_modbus.write_registers
        metadata: {}
        data:
          start_address: 44001 
          values: "3600" `how many seconds to run for`
          inverter: ea412fb93b7c45f7fbedc662fca0eac0
          
      - action: foxess_modbus.write_registers
        metadata: {}
        data:
          inverter: ea412fb93b7c45f7fbedc662fca0eac0
          values: "5000" `how many watts to discharge positive figure or charge negative figure`
          start_address: 44002

      - action: foxess_modbus.write_registers
        metadata: {}
        data:
          start_address: 44000
          values: "1" `enables the above to work like an on off switch I think`
          inverter: ea412fb93b7c45f7fbedc662fca0eac0

WyndStryke, 2025-01-24:
Just noticed in the modbus protocol document (20240516, v1.05) that it lists 46000-46020 as the remote control registers, and doesn't mention the 44000 range.

On my KH >1.33,  44000-44002 seems to contain identical data to 46001-46003.  I have to assume that means that they intend to depreciate 44000 in the future?

I see in the code that the H3 Pro is already looking at 46xxx.

With a 60s timeout, would register writes be a concern?  I note that Trevor has mentioned register writes quite a bit recently (albeit in the context of GivEnergy).

WyndStryke, 2025-01-28:
Regarding this earlier comment:

> As I recall there may be a problem with reading the Remote Take effect register as the KH did not expose registers above 44003 - it would be useful to know if this is still the case and needs to be re-checked on a KH

I've been trying to investigate the 46000 range to see if that would be useful.  It returns some data but not enough to be able to distinguish between force charge / force discharge if set via the fox app's manual schedule, for example. Feed-in is the only one I can reliably identify.  I can see the timeout counting down OK.

```
**Schedule - Force-charge 00:15-06:59**
changed to 46001=1,46002=23400,46003=0/0,46005=0/0,46007=23326,46018=0/0,46020=0/0,49203=1,46501=0/13000,46503=0,46504=0/13000,46506=0,46507=0,46508=0,46509=0,46510=0,46511=0,46512=0,46513=0,46514=0 triggered by automation WriteRemoteControlToLog triggered by time pattern
00:15:05 - 16 minutes ago

changed to 46001=1,46002=840,46003=0/0,46005=0/0,46007=232,46018=0/0,46020=0/0,49203=1,46501=0/13000,46503=0,46504=0/13000,46506=0,46507=0,46508=0,46509=0,46510=0,46511=0,46512=0,46513=0,46514=0 triggered by automation WriteRemoteControlToLog triggered by time pattern
00:10:05 - 21 minutes ago

changed to 46001=1,46002=840,46003=0/0,46005=0/0,46007=529,46018=0/0,46020=0/0,49203=1,46501=0/13000,46503=0,46504=0/13000,46506=0,46507=0,46508=0,46509=0,46510=0,46511=0,46512=0,46513=0,46514=0 triggered by automation WriteRemoteControlToLog triggered by time pattern
00:05:05 - 26 minutes ago

**Schedule - force-charge 00:01-00:14**
changed to 46001=1,46002=840,46003=0/0,46005=0/0,46007=826,46018=0/0,46020=0/0,49203=1,46501=0/13000,46503=0,46504=0/13000,46506=0,46507=0,46508=0,46509=0,46510=0,46511=0,46512=0,46513=0,46514=0 triggered by automation WriteRemoteControlToLog triggered by time pattern
00:00:05 - 31 minutes ago
25 January 2025

**Schedule - force-discharge, 23:48-23:59**
changed to 46001=1,46002=720,46003=0/0,46005=0/0,46007=230,46018=0/0,46020=0/0,49203=1,46501=0/13000,46503=0,46504=0/13000,46506=0,46507=0,46508=0,46509=0,46510=0,46511=0,46512=0,46513=0,46514=0 triggered by automation WriteRemoteControlToLog triggered by time pattern
23:55:05 - 36 minutes ago
changed to 46001=1,46002=720,46003=0/0,46005=0/0,46007=527,46018=0/0,46020=0/0,49203=1,46501=0/13000,46503=0,46504=0/13000,46506=0,46507=0,46508=0,46509=0,46510=0,46511=0,46512=0,46513=0,46514=0 triggered by automation WriteRemoteControlToLog triggered by time pattern
23:50:05 - 41 minutes ago

**Schedule - self-use, 23:38-23:47**
changed to 46001=1,46002=900,46003=0/0,46005=0/0,46007=9,46018=0/0,46020=0/0,49203=1,46501=0/13000,46503=0,46504=0/13000,46506=0,46507=0,46508=0,46509=0,46510=0,46511=0,46512=0,46513=0,46514=0 triggered by automation WriteRemoteControlToLog triggered by time pattern
23:40:05 - 1 hour ago

**Schedule - Feed-in, 23:28-23:37**
changed to 46001=1,46002=900,46003=0/0,46005=0/0,46007=9,46018=0/0,46020=0/0,49203=2,46501=0/13000,46503=0,46504=0/13000,46506=0,46507=0,46508=0,46509=0,46510=0,46511=0,46512=0,46513=0,46514=0 triggered by automation WriteRemoteControlToLog triggered by time pattern
23:30:05 - 1 hour ago

```

(This is from an automation that passively watches the registers on a 5 minute poll interval, while the fox app runs it's schedule).

WyndStryke, 2025-01-29:
Should point out that force charging / discharging all work fine via the integration on my KH, without needing assistance from automations, it's just reading the schedule or work-mode if it was configured from the app (rather than the integration) which doesn't seem to be possible.  I don't **need** to do that, it's just that I think it could potentially be useful if someone is using the integration primarily to monitor their system, rather than control it.

AndySmithers, 2025-04-04:
So glad I found this. This is exactly my issue with my H1 G2 5Kw inverter. Force Charge or discharge cycles randomly - presumably every write cycle?
Writing directly to the registers works correctly.
Is there any way I can get an integration that does this using the standard controls since it's going to complicate my PredBat somewhat to use direct register writes. If not, can I get an easy table showing the registers to write for force Charge and discharge?

WyndStryke, 2025-04-04:
Firstly make sure that the app does not have the mode scheduler enabled, otherwise they will be fighting for control.

Once that is turned off, then predbat should just work, assuming that your apps.yaml is good, and that you have set up the various automations to translate between W and kW.

For comparison purposes, this is my apps.yaml

`
pred_bat:
  module: predbat
  class: PredBat

# Based on the KH7 inverter, EC4300-H battery, and E-on Next Drive tariff.  Adjust as needed
# for your inverter, battery, and tariff.
#
# NECESSARY:
#
# Check your firmware version to ensure that BMS sensors and Back-up work-mode are supported.
# Email Fox support for a firmware update if necessary.
#
# Many sensors from the FoxESS MODBUS integration are provided in kW.  However Predbat needs
# these in W.  Install Tony's sensors + automations to do the conversion.
#   See https://github.com/TonyHoyle/foxess_predbat/blob/master/predbatt.yaml
#
#    - input_number.force_charge_power_w 
#    - input_number.force_discharge_power_w
#    - sensor.pv_power_w
#    - sensor.bms_actual_kwh_remaining
#
# Temperature charge curve either needs to be recalculated for your batteries voltage (or
# commented out)
#    battery_temperature_charge_curve
#
# Predbat settings:
#  * input_number.predbat_set_reserve_min should be set to 10% (90% DoD), as a minimum.
#       This is a hard limit, below this the system will draw from the grid.
#  * input_number.predbat_best_soc_min should be set to the kWh equivalent of 10% (90% DoD),
#       as a minimum
#  * input_number.predbat_best_soc_keep should be set to the kWh equivalent of a few % above
#       both the soc_min and reserve_min
#
# OPTIONAL:
#
# Install the carbon_intensity_uk integration as per predbat instructions, set#
#   input_number.predbat_carbon_metric and enable switch.predbat_carbon_enable.
#
# input_number.predbat_metric_battery_cycle can be set to (e.g.) 1p/kWh to represent the cost
#   of battery cycling.
#
# Increase history retention to 15 days in Homeassistant/configuration.yaml (this will allow
#   day 14 to be used for load estimation, since normally only 10 days are stored). e.g.:
#  recorder:
#    purge_keep_days: 15
#
# If too  much history is stored then the database size will escalate, this can be fixed by
#   purging less important sensors earlier via an automation which runs overnight.
#
# Periodic SoC Calibration - Needs an automation - track the times, and set min_soc, max_soc,
#   etc as needed  Fox inverters do not automatically calibrate.  Recommendation, at least once
#   every 7 days the battery should hit the 100% SoC mark, and at least monthly the battery
#   hit the 10% mark.

  # Sets the prefix for all created entities in HA - only change if you want to run more than once instance
  prefix: predbat

  # Timezone to work in
  timezone: Europe/London

  # Currency, symbol for main currency second symbol for 1/100s e.g. $ c or £ p or e c
  currency_symbols:
    - '£'
[... 578 lines cut ...]
  #watch_list:
  #  - '{octopus_intelligent_slot}'
  # - '{octopus_ready_time}'
  # - '{octopus_charge_limit}'
  #  - '{octopus_saving_session}'
  #  - '+[car_charging_planned]'
  #  - '+[car_charging_soc]'
  #  - '{car_charging_now}'
  
  # Carbon Intensity data from National grid
  carbon_intensity: 're:(sensor.carbon_intensity_uk)'

  # Watch list, a list of sensors to watch for changes and then update the plan if they change
  # This is useful for things like the Octopus Intelligent Slot sensor so that the plan update as soon as you plugin in.
  # If you update any predbat settings via automations, add them here.
#  watch_list:
#    - '{octopus_intelligent_slot}'
#    - '{octopus_ready_time}'
#    - '{octopus_charge_limit}'
#    - '{octopus_saving_session}'
#    - '+[car_charging_planned]'
#    - '+[car_charging_soc]'
#    - '{car_charging_now}'

`

AndySmithers, 2025-04-04:
Yeah, my apps.yaml is pretty much the same as yours (I worked through all your information today!). And I don't have the scheduler enabled.
However I have just gone through the exercise of writing directly to the Modbus registers and the force Charge/discharge works perfectly with this (as it does with the Fox scheduler). Setting force Charge/discharge using the integration cycles every few 10s of seconds.
Tomorrow I will change PredBat to call an automation for charge discharge. I'm now 100% convinced this is the issue.

WyndStryke, 2025-04-04:
> I'm now 100% convinced this is the issue.

The issue in this thread was resolved some time ago, so unless you have an old version of the integration, or perhaps old firmware, I don't think it is what you are seeing.

AndySmithers, 2025-04-05:
What can I say? My H1 G2 inverter is running 1.48 and I'm using the current 1.13.4 version of the integration.
When I set Force Charge or Force Discharge in the integration, I see the grid in/out cycling 0 - programmed power.

If I do the direct register writes detailed above, the Forced Charge or Forced Discharge works perfectly.
I don't have the Scheduler enabled in the app (before trying it yesterday to debug I had never used it).
When I get a chance to force discharge or charge again (clipping with sun today!) I will capture a debug session if it's useful.

[image]

AndySmithers, 2025-04-06:
Ok, I sat down with HA this morning to try to capture some meaningful logs.
The first thing I did was simply set the integration into force charge mode with debug on expecting to capture the cycling seen before.
To my great surprise it all worked perfectly, no cycling and the force charge stayed until I turned it off.
So then I tried force discharge.
Again this worked perfectly.
Baffled by why it had suddenly started working I let Predbat rip with Charge and Discharge.
This morning was a complicated one for Predbat because import rates were so low but Solcast was predicting a sunny day.
Thus Predbat did a lot of force charge/discharge.
And the integration performed flawlessly - no sign of the cycling that I had observed before every time I tried force charge/discharge.

In both cases the debug logs shoed exactly what I expected from the description of how force charge/discharge works.
(Oh, I was confused for a few seconds on the strange value for charge power, until it dawned on me that it was 2s complement!)

So, what has changed?
There's only 3 things different from the screen grab captured above:
1. That screen grab was in the middle of the night, so no sun. But, I'm 90% certain I've tried before when the sun is up.
2. Between it "not working" and "working" I have used the Schedule function of the Fox app for the first time ever (and then turned it off).
3. Between it "not working" and "working" I have tried the direct register write method for force charging - which worked perfectly.

Is it in any way feasible that there is a non-exposed register which needed to be written to by the scheduler before the integration force charge/discharge would work correctly?
I know it's a stretch to suggest that everyone else's H1s are only not showing this issue because they have been run in scheduler mode and mine was unique in not doing so - but I am running the latest firmware - 1.48, which is pretty new..???

Of course I've got to verify this still works as expected once the sun is down.

WyndStryke, 2025-04-06:
> 2\. Between it "not working" and "working" I have used the Schedule function of the Fox app for the first time ever (and then turned it off).

This is what I think it is most likely to be.  If either of the two cloud schedule systems (either the two force-charge timeslots **or** the module scheduler) was unknowingly running previously, then it would result in exactly the behaviour that you were seeing.   By turning it on, then off again, you'd be making sure that it was genuinely off.   If you didn't set it up yourself, the installer may have done so (mine was very keen on setting up the charge slots during the system demo).

williamjeccles, 2025-10-02:
I believe this was resolved in one of the firmware releases since I created it as mine is also now working great with the force charge and force discharge select box exposed by this integration.

If you're reading this in future please ensure your master and manager firmware are up to date as it should now be working.

closed 2025-10-02 by williamjeccles