nathanmarlor/foxess_modbus #669: EPS power moved in H3?

canton7, 2024-09-02:
I did a quick EPS test, and I think EPS Power is in holding register 39217 (and possibly 39216 if it's a 32-bit value - which would make the H1_G2 the same as the H3_PRO, counting against one of the 3 phases I guess). It may always have been there, or this may be new as of Manager 1.44 - I've no way of testing that unfortunately.

_Originally posted by @paultweedy77 in https://github.com/nathanmarlor/foxess_modbus/discussions/666#discussioncomment-10520750_

canton7, 2024-09-18:
@paultweedy77 I'd find it very odd if just the EPS power has moved out to 39217 (and things like EPS current have stayed put). Sure the H3 uses that register, but as you say it uses it for the T phase.

I guess they might have moved it out if they wanted to make it a 32-bit register and needed more room... Apparently the G2 has a max capacity of 33kW, which just exceeds a 16-bit register. I would have expected them to move rpower as well though, in that case.

Are you certain that the EPS power has moved out to 39217, and nothing else EPS-related (or power-related) has moved?

paultweedy77, 2024-09-18:
The only thing I'm certain about is that register contains the correct EPS power figure when in EPS mode, and is 0 otherwise. My suspicion is that it's unintentional and shouldn't be relied on, and that the change in behaviour of register 31012 in Manager 1.44 is a bug.