nathanmarlor/foxess_modbus #678: K series manager upgrade breaks integration

williamjeccles, 2024-09-27:
KH10.5 inverters upgraded to manager 1.33 and now seeing the following error in HA:

Error connecting to inverter (friendly name: "KHM"). Error reading registers. Type: RegisterType.HOLDING; start: 31039; count: 10; slave: 247 from tcp://192.168.1.164:502: Exception Response(131, 3, IllegalAddress)

williamjeccles, 2024-09-27:
So @TonyM1958 confirmed that address 31039 is PV3 voltage, moving a solar string over to pv3 and I can confirm that the address is showing 0v in home assistant using tony's project so assume this manager update has broken pv3 in someway.
Happy to test if someone can confirm how I can comment out the pv voltage registers and see if I get overcome the blocking faulty register?

williamjeccles, 2024-09-27:
Currently, the following are failing to respond and blocking the integration from loading cleanly.
Pv3 & PV4
31039
31040
31041
31042

31045 -rpower
31046 -rpower

EPSpower
31047
31048

williamjeccles, 2024-09-27:
Ok so Tony pointed me to a new[ v2 address register,](https://github.com/TonyM1958/HA-FoxESS-Modbus/blob/main/custom_components/HA-FoxESS-Modbus/modbusV2_RS485_LAN.yaml) I've tested out the new address on the pv3 voltage address which is the first failing address and it's working. I'm going to test out updating them each with the new values but it looks like more of an improvement to the address consistency than bugs introduced at this stage.

[image]

canton7, 2024-09-27:
Yay, I hate it when they break things. Looks like they're bringing it into line with the H3 Pro perhaps? That's the other inverter which uses 39074.

I think we're going to need to start selecting which registers to use based on the firmware version... We can't just keep breaking all KH users because one person upgrades. I'll try and find some time to implement something.

williamjeccles, 2024-09-27:
thanks @canton7, do you want me to commit a PR so we can offer a beta version for those who have upgraded while you work on a switch of some kind?

canton7, 2024-09-27:
Sure, if you've got the time to update the register map, that would be helpful!

Can you do what we did for the KH, and at least keep the old register map around? (For the KH, we did that as `KH_PR119`).

williamjeccles, 2024-09-27:
@canton7 versions are completely messed up also :(

[image]

I feel like this is going to need proper reengineering, maybe a switch in the config wizard or maybe we only support the latest firmware?

canton7, 2024-09-27:
🤦

TonyM1958, 2024-09-27:
Looks like the encoding of the versions has changed, it seems unlikely they got this wrong?

Previously, they have used decimal and then hex encoding. Don't think they expected to get to version 2.55. You may need to write out the values you read in binary to see if you can make any sense it.

canton7, 2024-09-28:
Re the version... We're currently interpreting this as hex, which seems to be the "new" standard. That means that "31.31" is 0x3131.

I can't work out any way in which that maps onto "1.33"....

It seems more likely to me that they've gone and moved the version registers, particularly if they're moving to parity with the H3 Pro. @williamjeccles can you check registers:

* 36001 (master version)
* 36002 (slave version)
* 36003 (manager version)

please?

TonyM1958, 2024-09-28:
ah OK. I just assumed you would have already tried that... may be not. There are other examples of versions in v2, such as Protocol Version Code that is 32-bit hex byte encoded.

canton7, 2024-09-29:
I've started having a crack at the KH-1.33 support. See https://github.com/nathanmarlor/foxess_modbus/pull/681.

I haven't put in anything which the spec says has 3 phases, as I'm not sure what the inverter reports there. @williamjeccles if you could have a poke around that would be great!

I also haven't decided how best to not break everyone who's on an older KH firmware version. They've gone and moved the version registers, which is very annoying.

williamjeccles, 2024-09-29:
[FoxESS Modbus Protocol--20240516 (V1.05.00.00)(1).pdf](https://github.com/user-attachments/files/17180041/FoxESS.Modbus.Protocol--20240516.V1.05.00.00.1.pdf)

@canton7 new modbus spec as I've been blessed to share it.

I'll test those changes and report back

williamjeccles, 2024-09-29:
[screencapture-192-168-1-33-8123-config-devices-device-de13d4d265ce5727f0012385c06a8326-2024-09-29-18_01_01.pdf](https://github.com/user-attachments/files/17180098/screencapture-192-168-1-33-8123-config-devices-device-de13d4d265ce5727f0012385c06a8326-2024-09-29-18_01_01.pdf)

So good news is that the integration loads when using commit #681, but fews things we need to fix.

As you can see from the attached, grid CT even when putting grid load on is showing 0

canton7, 2024-09-29:
Cool! Yep, as I said, I haven't put in anything which the latest spec says has 3 phases, as the KH is only single phase. If you can figure out how ids actually exposing that data (just using the R phase etc), I'll put them in.

canton7, 2024-09-29:
Could you also check the versions? The spec also claims to support force charge /discharge work modes natively, so I've enabled them and haven't enabled the remote control stuff - mins testing?

williamjeccles, 2024-09-30:
@canton7 any chance you push to a beta branch so I can easily import via hac instead of having to pull the commit?

williamjeccles, 2024-09-30:
I also have access to the old 1.31 so I can switch between 1.31 and 1.33 with ease.

williamjeccles, 2024-09-30:
@canton7 so using that commit, the biggest issues right now that I think we need to fix is the grid ct isn't working for me. I believe the address should be 39168. If you can update that and I'll pull the change and try it out again.

canton7, 2024-10-06:
@williamjeccles As soon as you can have a play with modpoll and let me know what the KH exposes on the R/S/T phase registers (i.e. what register of those 3 (if any) actually contains the information we need), I can get that in

williamjeccles, 2024-10-10:
@canton7 so if i've understood the ask correctly I've tested out the following:
[image]

Meter1/CT1 RPhase Active Power address: 38816 appears to be the ct1 negative number for export, positive for import

Meter1/CT1 SPhase Active Power address: 38818 - appears to respond but with no data
      
Meter1/CT1 TPhase Active Power address: 38820 - appears to respond but with no data

I hope that helps resolve the K series grid related registers in the next release.

FozzieUK, 2024-10-10:
@williamjeccles not sure if you have looked at this, but i'm intrigued to know if the BMS registers that went missing after the 1.19 update may actually have been moved to meet the new modbus spec 

Could you have a poke around at these addresses and see if you get a response please - particularly the items 24 - 32 & 39, 40, 42 on this list

[image]

williamjeccles, 2024-10-10:
Yeh so mv high low, cell temp high low, remaining kWh all working.
SOH I can't test as I've not got a v2 battery to test with but the modbus address works it just returns 0 for me.

FozzieUK, 2024-10-10:
Thanks, that’s great - be nice to have them working again 👍

williamjeccles, 2024-10-10:
@FozzieUK are you able to push a fix for the grid side stuff and these extra battery addresses?

canton7, 2024-10-10:
Thanks! I'll get those in when I get the chance (I'm afraid I'm a bit swamped until the middle of next week).

I'm keen not to release this until we have a way to switch between H3 Pro versions (either automatically or manually), as I don't want an update to just break our existing H3 Pro users.

FozzieUK, 2024-10-10:
No problem - I'll leave it to Antony - he'll be much quicker when he gets to it 👍 

(also in the middle of fixing Nathan's EM which has stopped working on 2024.10.x)

williamjeccles, 2024-10-10:
> Thanks! I'll get those in when I get the chance (I'm afraid I'm a bit swamped until the middle of next week).
> 
> I'm keen not to release this until we have a way to switch between H3 Pro versions (either automatically or manually), as I don't want an update to just break our existing H3 Pro users.

Thanks

ApronCricket, 2024-11-06:
Hi, any guidance on how to get a KH10 working again with fw Master:1.35?
If there is any information you might need that I can help with, let me know.
Or, if I'm best just waiting for a new release, is there any chance of an ETA?
Thanks.

ApronCricket, 2024-11-06:
> Yeh so mV high low, Cell temps high low, remaining kWh all working. SOH I can't test as I've not got a v2 battery to test with but the modbus address works it just returns 0 for me.
> 
> I think it's safe to say they've moved and are all working.

[image]

I've a brand new V2 battery and get 100(%) at the SOH address, so looks correct.

williamjeccles, 2024-11-06:
> ![image](https://private-user-images.githubusercontent.com/72933958/383629227-c721d16a-1b4d-490c-894c-0f7fe45a2ad1.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

thanks for confirming

rand12345, 2024-11-10:
Thank you for creating a fix @ocl80, it's working well here. 

@FozzieUK Do you have time to help bring these changes into @canton7 work and create a PR for this? I think many users are affected by this issue.

FozzieUK, 2024-11-10:
@rand12345 I think @williamjeccles has also done a lot of work on this as well to merge.

The biggest challenge is not to break the integration with earlier releases of firmware before 1.33 and @canton7 has already started work on that. 
Sadly i'm away for a couple of weeks so won't be able to get to look it for a while, I can see a number of users are needing this so hopefully @canton7 can find a bit of spare time to push this out 👍

DoingMyBit, 2024-12-04:
Hi All, 

Just wondering if there is an update on when an update to resolve this issue might be released? I know you do this out of the goodness of your hearts so thank you for all the efforts you put into this. 

All the best!

simondotm, 2024-12-11:
@ocl80 Sorry if this is a dumb question, and to side track the thread a little. I'm new to all this. In order to run your fork, how best to do this in HACS? Do I install your fork as a separate integration alongside this one, or is there a cleaner way to do it?

Just until the official fix is in, this might help me get things working as I have a new install with a KH7 and I'm getting the Illegal Address exception problem, so no data for my setup yet. Thanks!

ocl80, 2024-12-11:
Hi @simondotm I uninstalled the proper one for the time-being and installed my version instead.
After there is a proper fix, I will remove my version and reinstall this one; in doing so I _think_ that I will need to delete the entities created by my integration; this will allow the proper integration to recreate them again with the same names, and the historical stats in homeassistant will match up preserving the history. Hope that helps.

DoingMyBit, 2024-12-11:
@simondotm Just in case its installation process advice your after, I wasn't familiar with how to install forked repositories, so I basically followed this to install @ocl80 's fork:

[image]

@ocl80 thanks so much for doing this fork (legend!), while we wait for the other legends here to figure out how to support both new and old firmwares.

ocl80, 2024-12-11:
> @ocl80 thanks so much for doing this fork (legend!), while we wait for the other legends here to figure out how to support both new and old firmwares.

That's very kind, and indeed I have only respect for the real legends who will fix it properly ultimately, meanwhile I'm glad if this helps!

simondotm, 2024-12-11:
@ocl80 @DoingMyBit thanks both! I added the fork repo URL and downloaded it. The stats now seem to be flowing in. 👍

DoingMyBit, 2025-01-06:
Hi All, I’ve been using @ocl80 ‘s fork successfully for a while now (thanks again). 

Unfortunately HA core 2025.1 breaks it. I have rolled back to 2024.12.4 and it’s working again, so OK for the moment. 

I’m wondering though if the new update to the main integration fixes this issue with the K series inverters on newer firmwares as well as playing nicely with 2025.1. Please could someone advise?

As always, many thanks!

@canton7 @williamjeccles @FozzieUK @rand12345

ocl80, 2025-01-06:
@DoingMyBit , as far as I can see the newer K series firmware is still not working in the main version of the integration. 
But I merged in the updates from the main integration yesterday, into a new "v1.13.0-beta" release of my fork, which I am now running OK with HA 2025.1. 
To update to it, in HACS, click the ellipsis next to the integration, select _Redownload_, expand _Need a different version_ and select v1.13.0-beta.

helder-web, 2025-01-07:
> @DoingMyBit , as far as I can see the newer K series firmware is still not working in the main version of the integration. But I merged in the updates from the main integration yesterday, into a new [...]

Just tested this and numbers are back (KH9 here) :) 

Just a note (if this helps):
- PV3 and PV4 numbers are 0 (except Power Total) and this results in PV Power to also be half of what it should be (I have 4 strings). This also means PV Power (total) + battery discharge is less that Load Power (SelfUse mode, no grid usage)

And many thanks for this Beta release!

ocl80, 2025-01-08:
@helder-web I have published [v1.13.0-oc1-beta2](https://github.com/ocl80/foxess_modbus_ocl80/releases/tag/v1.13.0-oc1-beta2) for HACS; if you get a chance could you check if it works with PV3 and PV4 (and if it still works with PV1 and PV2) as they are all now aligned to the newer FoxESS ModBUS spec that @williamjeccles shared earlier. Let me know how you get on!

helder-web, 2025-01-08:
Thanks!! Just tested and all is working now!

[image]

It is night but I can see now voltage is correct for all 4 strings.

Also the version is now correct:
[image]

Can’t see anything wrong but I can double check tomorrow once the sun comes up.

BTW, after this firmware update I can see the inverter consumption dropped from 200-250W to 80W now, running with 2x EP11 with also latest BMS firmware.

ocl80, 2025-01-10:
Thanks @helder-web - did all 4 strings report OK during the day?

helder-web, 2025-01-10:
> Thanks @helder-web - did all 4 strings report OK during the day?

Hey @ocl80 , yes, I can confirm all is working as expected. Compared the values against FOX Cloud and they are matching.

DoingMyBit, 2025-01-11:
@ocl80 Thank you so much for the fix to get it working with the latest HA Core, and also for adding PV3 and PV4! Sending you many positive vibes - you’ve been a superstar getting this working for us non-technical people. All the best!

ocl80, 2025-01-14:
Thanks @DoingMyBit and thanks @helder-web for confirming. 

I have done a bit of tidying up of the code in [v1.13.0-oc1](https://github.com/ocl80/foxess_modbus_ocl80/releases/tag/v1.13.0-oc1) for >= HA 2025.1 and also [v1.12.2-oc1](https://github.com/ocl80/foxess_modbus_ocl80/releases/tag/v1.12.2-oc1) for HA < 2025.1. Functionally nothing should be different, so its not important to update it for those using the previous beta2, but I cleaned up my changes so that they are using a "KH_133" profile, and still retaining the record of the address maps for the older version as "KH_PRE133" as @canton7 mentioned (if I've understood how this was done before!). I had also previously merged your amendments @canton7 in [#681](https://github.com/nathanmarlor/foxess_modbus/pull/681) for the fault code set which I wouldn't otherwise have clocked. 

It will still break with the earlier KH firmwares unless something smarter is done in the flow to either discover the FW version and select a map accordingly, or allow the user to choose. Unfortunately that is beyond me!

canton7, 2025-02-01:
This should be fixed in v1.13.1.

closed 2025-02-01 by canton7

DoingMyBit, 2025-02-07:
@ocl80 Hey there. Thanks again for all the work you did for the interim fix fork. I just wanted to check in with you to see how one might migrate to the main version now. Is there a way to set it up that retains the data? Thanks!