nathanmarlor/foxess_modbus #683: H3 Pro 15kW - rFreq wrong scale

jschmitz-de, 2024-09-30:
Hi, 
i have a FoxEss H3 Pro 15kW and use Homeassist and your Plugin since the PV was installed in August.
Everything (that I can overview) works great - thanks 4 that!

When i look into the entity_descriptions.py the scale for all intverters is set to 
scale=0.01 (Line 1217)

In the ModBus description the gain (for H3 Pro) is given with 100 - but this seams to be wrong.

So it is possible to change the scale to 0.001 whitch results with correct frequence - but this is only a mitigation.

I don´t know how to "case" the scale and will be grateful if this could be implemented.

## Version of the custom_component

Used Version 1.12.2 

## Configuration
no special config

```yaml
Dieser Fehler wurde von einer benutzerdefinierten Integration verursacht

Logger: custom_components.foxess_modbus.entities.modbus_entity_mixin
Quelle: custom_components/foxess_modbus/entities/modbus_entity_mixin.py:177
Integration: FoxESS - Modbus (Dokumentation, Probleme)
Erstmals aufgetreten: 10:13:29 (84 Vorkommnisse)
Zuletzt protokolliert: 10:34:14

Value (49976: 499.76) for entity 'sensor.inverter_rfreq' address(es) '[38847, 38846]' failed validation against rule (Range : {'_min': 0, '_max': 60})
Value (49984: 499.84000000000003) for entity 'sensor.inverter_rfreq' address(es) '[38847, 38846]' failed validation against rule (Range : {'_min': 0, '_max': 60})
Value (50007: 500.07) for entity 'sensor.inverter_rfreq' address(es) '[38847, 38846]' failed validation against rule (Range : {'_min': 0, '_max': 60})
Value (50015: 500.15000000000003) for entity 'sensor.inverter_rfreq' address(es) '[38847, 38846]' failed validation against rule (Range : {'_min': 0, '_max': 60})
Value (50000: 500.0) for entity 'sensor.inverter_rfreq' address(es) '[38847, 38846]' failed validation against rule (Range : {'_min': 0, '_max': 60})

```

## Describe the bug
I got an error on the Log that the rFreq Value got a Range Error

FozzieUK, 2024-09-30:
Hi, thanks for this.

Could you do a couple of test please by reading the registers directly.

To do that in home assistant, go to Developer Tools, and select 'Actions' (or Services as it used to be called).

In the 'action' box start typing Foxess - Modbus and select 'Foxess - Modbus: Read registers'

Choose your inverter in the 'inverter' drop down field

Set the start address to 38846, the count to 2, the type to holding - and click on' perform action' as this -

[image]

If you could screen shot the response and paste it below please.

Then could you set the start address to 39139, the count to 1, the type to holding  - and click on' perform action'

Again if you could screen shot the response and paste below please.

jschmitz-de, 2024-09-30:
Yes - no Problem: 

**Register38846 +1**
[image]

**Register 39139:** 
[image]

FozzieUK, 2024-09-30:
Ok, thanks.

The latest modbus spec confirms that 39139 is the actual grid voltage register, whereas 38846 is the meter1/ct frequency - admittedly the latest spec is incorrect in that it says the 38846 (i32) register is a 100 scale but it should be reading 39139 instead which scales correctly at 0.01

There are likely to be a few changes being made over the next few week so it would be good if you could change line 1210 from this `ModbusAddressesSpec(holding=[38847, 38846], models=Inv.H3_PRO)` to this `ModbusAddressesSpec(holding=[39139], models=Inv.H3_PRO)` 

And restart your home assistant, it should then be reporting grid frequency correctly - if you can confirm that i'll do a PR for it, but as I say it might be a few weeks before it makes a formal release.

canton7, 2024-09-30:
The values that they're getting for the frequency do look correct (albeit 10x too large).

I'll check the spec I have for the H3 Pro when I can, see where that register and scale actually came from. This might have been broken in a recent firmware update or something, as I'm sure it would have been caught when we were first putting in H3-Pro support.

jschmitz-de, 2024-09-30:
Hi @FozzieUK  i changed the line - works like expected.
Thanks for that fast reply.

kr
joerg

jschmitz-de, 2024-09-30:
Mitigated

FozzieUK, 2024-09-30:
@canton7 it looks like the original testing may have found the meter1/ct frequency registers when testing and not the actual grid frequency register which looking at the latest modbus spec received yesterday is of the right scale (albeit the 38846, 38847 are also shown as 100 scale when in fact they are 1000). 
Let me know when you've had chance to review, happy to do the PR to update if it helps.

canton7, 2024-10-06:
Reopening, as we still need to fix this in the integration!

79Anni, 2024-11-21:
I get a much too high value for the PV4 Total. Is there a Solution?
Thank you very much

[image]

bradziszewski, 2024-12-23:
same here H3 pro:
[image]

also there are missing PV5 and PV6.

jschmitz-de, 2025-01-07:
Hi all, 
i´m a bit confused...
After Update 2025.1 and foxess_modbus Update to [v1.13.0] everything (Broken in 2025.1) seems to work 
Additionally the PV5 / PV6 Entities apeared (:-) - but in the Same time the rFrequenz Problem (validation against rule (Range : {'_min': 0, '_max': 60})) came back.
I changed line 1296 in entity/entity_descriptions.py to :  ModbusAddressesSpec(holding=[39139], models=Inv.H3_PRO),
and also the rFreq came back to valid Value.
Can u please put this into the next update ? 

Kind regards - and ... btw: THANK YOU FOR THIS GREAT INTEGRATION !!!!

canton7, 2025-08-13:
Fixed in #777

closed 2025-08-13 by canton7