nathanmarlor/foxess_modbus #706: Keep getting messages about no longer having a state class etc

letsstartbeinganon, 2024-11-21:
## Version of the custom_component

I'm using v1.12.2

## Configuration

Not sure how to do this but I've attached the debug logs below.

## Describe the bug

Every or multiple times per day if I clear I get a repair listed in Home Assistant saying that an entity no longer has a state class or that the unit has changed.  It doesn't appear to be doing anything that breaks the functionality but not sure how I'd know it is.  I do use this integration to help charge/discharge my battery at certain times of day so want assurance this bug isn't harming that.

## Debug log

[home-assistant_foxess_modbus_2024-11-21T11-50-04.216Z.log](https://github.com/user-attachments/files/17845161/home-assistant_foxess_modbus_2024-11-21T11-50-04.216Z.log)

[home-assistant_foxess_modbus_2024-11-19T16-20-34.679Z.log](https://github.com/user-attachments/files/17845162/home-assistant_foxess_modbus_2024-11-19T16-20-34.679Z.log)

[image]
[image]

letsstartbeinganon, 2025-01-03:
Still having this issue. Could it be something to do with PredBat, which I also have running?

canton7, 2025-01-31:
Can you go to Developer tools -> States and filter by sensor.bms_charge_rate? It should look like this:

[image]

letsstartbeinganon, 2025-01-31:
> Can you go to Developer tools -> States and filter by sensor.bms_charge_rate? It should look like this:
> 
> [image]

[image]

canton7, 2025-01-31:
I'm not sure where all of the attributes have gone. They're definitely properly defined in foxess_modbus. Do you have a template or something which is trying to provide that entity? Can you see "BMS Charge Rate" in the list of entities provided by foxess_modbus?

bfayers, 2025-02-01:
This looks like the behaviour of incorrectly configured predbat (it happened to me too).

At the moment predbat doesn't properly support setting charge power in kW rather than watts (https://github.com/springfall2008/batpred/issues/1717)

Personally, I made two input_number helpers for force charge (watts) and force discharge (watts) then use an automation to sync those values to the ones in this integration (here's a gist with that automation: https://gist.github.com/bfayers/82f5a45155281d187f2758c66872c162)

In predbat you then just set the charge_rate and discharge_rate to those input_numbers

```
  charge_rate:
    - input_number.force_charge_power_watts
  discharge_rate:
    - input_number.force_discharge_power_watts
```

This seems to all work fine for me.

canton7, 2025-08-13:
Thanks!

Closing this, as I don't think it's a problem with foxess_modbus.

closed 2025-08-13 by canton7