nathanmarlor/foxess_modbus #716: Feed-in and power consumption at the same time?

domerich, 2024-12-28:
I just updated from the old chestnut version to this, the installation is very nice.

One thing that I don't understand, it seems my inverter seems to both consume and feed-in engergy at the same time.

I cannot figure it out, does anyone know what this is?

[image]

FozzieUK, 2024-12-30:
As you have 3 phases they are independent of each other and if you have your H3 set with balance it will output equally across all it's phases - if one of your phases has a higher load than the amount of feed-in then you will see a grid consumption on that phase. 

Often your energy supply will be using net-metering so in the above example you are feed in (0+1.283+0)=1.283, and consuming (1.008+0+0.312)=1.32 so with net-metering your meter would see (1.32-1.283=0.037 import)

You can't import and export on the same phase at the same time.

domerich, 2024-12-30:
> As you have 3 phases they are independent of each other and if you have your H3 set with balance it will output equally across all it's phases - if one of your phases has a higher load than the [...]
 
 thanks for this explanation. My provider also uses a net meter, so all three phases are summed up. Is there a different way to set up the fox meter maybe for it to run more efficient? Consuming on one phase and feeding in on the other does not seem very efficient to me. I personally don’t care if it’s balanced or not.

closed 2025-08-13 by canton7