nathanmarlor/foxess_modbus #718: Foxess fails with newest core 2015.1.0

Twilek-de, 2025-01-03:
Foxxess Integration fails to setup with newest core with the following error:

`Logger: homeassistant.setup
Quelle: setup.py:334
Erstmals aufgetreten: 21:12:41 (1 Vorkommnisse)
Zuletzt protokolliert: 21:12:41

Setup failed for custom integration 'foxess_modbus': Unable to import component: cannot import name 'ModbusResponse' from 'pymodbus.pdu' (/usr/local/lib/python3.13/site-packages/pymodbus/pdu/__init__.py)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1015, in async_get_component
    comp = await self.hass.async_add_import_executor_job(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        self._get_component, True
        ^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1075, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 200, in protected_loop_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/foxess_modbus/__init__.py", line 20, in <module>
    from .client.modbus_client import ModbusClient
  File "/config/custom_components/foxess_modbus/client/modbus_client.py", line 16, in <module>
    from pymodbus.pdu import ModbusResponse
ImportError: cannot import name 'ModbusResponse' from 'pymodbus.pdu' (/usr/local/lib/python3.13/site-packages/pymodbus/pdu/__init__.py)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 334, in _async_setup_component
    component = await integration.async_get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1035, in async_get_component
    self._component_future.result()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1027, in async_get_component
    comp = self._get_component()
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1075, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 200, in protected_loop_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/foxess_modbus/__init__.py", line 20, in <module>
    from .client.modbus_client import ModbusClient
  File "/config/custom_components/foxess_modbus/client/modbus_client.py", line 16, in <module>
    from pymodbus.pdu import ModbusResponse
ImportError: cannot import name 'ModbusResponse' from 'pymodbus.pdu' (/usr/local/lib/python3.13/site-packages/pymodbus/pdu/__init__.py)`

stevetrease, 2025-01-03:
I'm seeing the same.

Woodmans52, 2025-01-03:
Same here, too

majornicety, 2025-01-03:
Same here. Rolled back to the backup.

mechanimal82, 2025-01-03:
Yep. I've been finding this since upgrading to 2024.12.x as mentioned here: https://github.com/nathanmarlor/foxess_modbus/discussions/714

I've been digging, as has icornish72 to find that the built in pymodbus has been updated to 3.7.4 from 3.6.9: https://github.com/home-assistant/core/pull/133175

Looking at this release, it states the removal of ModbusResponse: https://github.com/pymodbus-dev/pymodbus/releases/tag/v3.7.4

If I comment out that in modbus_client.py then I get another error: "No module named 'pymodbus.register_read_message"

So looks like there are multiple breaking changes in pymodbus in a few of the releases since 3.6.9, which I just don't know how to solve unfortunately. Hopefully someone that actually knows how this all works can do some magic and fix it!

Renz1982, 2025-01-04:
Same Problem :-(

Renz1982, 2025-01-04:
Where i can change ModbusResponse to ModbusPDU?

ITSpecialist111, 2025-01-04:
+1

Log details (ERROR)
Logger: homeassistant.setup
Source: setup.py:334
First occurred: 09:00:18 (1 occurrences)
Last logged: 09:00:18

Setup failed for custom integration 'foxess_modbus': Unable to import component: cannot import name 'ModbusResponse' from 'pymodbus.pdu' (/usr/local/lib/python3.13/site-packages/pymodbus/pdu/__init__.py)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1015, in async_get_component
    comp = await self.hass.async_add_import_executor_job(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        self._get_component, True
        ^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1075, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 200, in protected_loop_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/foxess_modbus/__init__.py", line 20, in <module>
    from .client.modbus_client import ModbusClient
  File "/config/custom_components/foxess_modbus/client/modbus_client.py", line 16, in <module>
    from pymodbus.pdu import ModbusResponse
ImportError: cannot import name 'ModbusResponse' from 'pymodbus.pdu' (/usr/local/lib/python3.13/site-packages/pymodbus/pdu/__init__.py)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 334, in _async_setup_component
    component = await integration.async_get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1035, in async_get_component
    self._component_future.result()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1027, in async_get_component
    comp = self._get_component()
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1075, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 200, in protected_loop_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/foxess_modbus/__init__.py", line 20, in <module>
    from .client.modbus_client import ModbusClient
  File "/config/custom_components/foxess_modbus/client/modbus_client.py", line 16, in <module>
    from pymodbus.pdu import ModbusResponse
ImportError: cannot import name 'ModbusResponse' from 'pymodbus.pdu' (/usr/local/lib/python3.13/site-packages/pymodbus/pdu/__init__.py)

=============================

Logger: homeassistant.util.loop
Source: util/loop.py:77
First occurred: 09:00:18 (1 occurrences)
Last logged: 09:00:18

Detected blocking call to import_module with args ('custom_components.foxess_modbus',) in /usr/src/homeassistant/homeassistant/loader.py, line 1075: ComponentProtocol, importlib.import_module(self.pkg_path) inside the event loop; This is causing stability issues. Please create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue For developers, please see https://developers.home-assistant.io/docs/asyncio_blocking_operations/#import_module Traceback (most recent call last): File "<frozen runpy>", line 198, in _run_module_as_main File "<frozen runpy>", line 88, in _run_code File "/usr/src/homeassistant/homeassistant/__main__.py", line 227, in <module> sys.exit(main()) File "/usr/src/homeassistant/homeassistant/__main__.py", line 213, in main exit_code = runner.run(runtime_conf) File "/usr/src/homeassistant/homeassistant/runner.py", line 154, in run return loop.run_until_complete(setup_and_run_hass(runtime_config)) File "/usr/local/lib/python3.13/asyncio/base_events.py", line 707, in run_until_complete self.run_forever() File "/usr/local/lib/python3.13/asyncio/base_events.py", line 678, in run_forever self._run_once() File "/usr/ [... cut ...] y", line 165, in async_setup_component result = await _async_setup_component(hass, domain, config) File "/usr/src/homeassistant/homeassistant/setup.py", line 334, in _async_setup_component component = await integration.async_get_component() File "/usr/src/homeassistant/homeassistant/loader.py", line 1027, in async_get_component comp = self._get_component() File "/usr/src/homeassistant/homeassistant/loader.py", line 1075, in _get_component ComponentProtocol, importlib.import_module(self.pkg_path)

muhkuh666, 2025-01-04:
Hi, same here. Kindly asking when will there be a fix?

mechanimal82, 2025-01-04:
> Where i can change ModbusResponse to ModbusPDU?

You can change it in custom_components > foxess_modbus > modbus_client.py

However, even making that change generates a new error: Unable to import component: No module named 'pymodbus.register_read_message'

then fixing that generates a new error... so there appear to be a lot of breaking changes and changes to locations that need to be adjusted in the 5 updates to pymodbus. For now, I'm leaving HA at 2024.11.3 until someone that can actually fix this can do something. I wish I was clever enough, but I'm just not

Chip2786, 2025-01-04:
+1 Same Issue, rolling back...

Andy256, 2025-01-04:
> You can change it in custom_components > foxess_modbus > modbus_client.py
>
> However, even making that change generates a new error: Unable to import component: No module named [...]

I've been testing different core versions this morning, latest core version that works for me appears to be 2024.12.4, 2024.12.5 breaks it.

canton7, 2025-01-04:
Currently looking at this, looks like pymodbus have done a bunch of breaking changes and HA just switched up the versions.

Please DO NOT comment with "+1" or "Same Here". It just fills my inbox with spam.

Thanks!

canton7, 2025-01-04:
Apologies for the delay. Version 1.13.0 should fix this. I've only had a chance to give it some cursory testing, so please open a new issue if you find anything broken.

Version 1.13.0 will only run on HA 2025.1.0, due to some fairly large backwards compat breaks which would be pretty hard to work around.

(As an aside, I really dislike the way that Python maintainers will break lots of things in a minor version bump. Looking at you, pymodbus...)

closed 2025-01-04 by canton7

Twilek-de, 2025-01-04:
No need to apologize. You are doing this in your free time. Thanks for the quick fix and have a nice weekend.

stevetrease, 2025-01-04:
An upgrade from HA 2024.12.5 to 2025.1.0 followed by using HACS to upgrade foxes_modbus to this afternoon's v1.13.0 seems to have worked for me.

Thanks for the swift resolution :-)

Renz1982, 2025-01-05:
Thanks for fast bugfix!

PR #720 (2025-01-04): Fix breakage caused by HA 2025.1.0
Fixes: #718