nathanmarlor/foxess_modbus #722: 1.1.30 Update - Failed to Start

saleemkhalil, 2025-01-04:
Hi

I've updated to the latest version HA, 2025.01. Right after there's an update to FoxEss Modbus. I've updated and after the restart the FoxEss module doesn't start. 

Logs are below

Logger: homeassistant.setup
Source: setup.py:334
First occurred: 9:02:35 PM (1 occurrences)
Last logged: 9:02:35 PM

Setup failed for custom integration 'foxess_modbus': Unable to import component: No module named 'pymodbus.pdu.register_read_message'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 334, in _async_setup_component
    component = await integration.async_get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1035, in async_get_component
    self._component_future.result()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1015, in async_get_component
    comp = await self.hass.async_add_import_executor_job(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        self._get_component, True
        ^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1075, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 200, in protected_loop_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/foxess_modbus/__init__.py", line 20, in <module>
    from .client.modbus_client import ModbusClient
  File "/config/custom_components/foxess_modbus/client/modbus_client.py", line 18, in <module>
    from pymodbus.pdu.register_read_message import ReadHoldingRegistersResponse
ModuleNotFoundError: No module named 'pymodbus.pdu.register_read_message'

I've rolled back HA and currently working. 

Also could help - my original FoxEss module was the one released by chestnut (stealthchesnut?) but I migrated to the nathanmarlor one when it was released last year. 

Thanks,

canton7, 2025-01-04:
Try uninstalling and reinstalling foxess_modbus (after updating HA to 2025.1). You won't lose your entities.

saleemkhalil, 2025-01-04:
> Try uninstalling and reinstalling foxess_modbus (after updating HA to 2025.1). You won't lose your entities.

I'll try this - if something breaks I can always restore from backup. Thanks for quick response.

saleemkhalil, 2025-01-04:
Hmmm, HA doesn't let me delete it from Devices and Services. 

I'll try deleting before updating and installing after the update.

canton7, 2025-01-04:
You can remove it from HACS.

canton7, 2025-01-04:
(Technically, for some reason, you've got the wrong version of pymodbus installed. I'm hoping that reinstalling foxess_modbus kicks HA into installing the right version)

saleemkhalil, 2025-01-04:
Still doesn't work. Same error. 

I'm able to delete it from hacs and do a redownload. 

I also redownloaded the version that worked to see if that updated the pymodbus.  Current version redownloaded fine and works. However, after updating HA and then to the new version of FoxEss still have the same issue...

canton7, 2025-01-04:
Do you have any other integrations installed which use modbus?

saleemkhalil, 2025-01-04:
Good shout, yes I do. 

FoxEss Energy Management by Nathan

canton7, 2025-01-04:
Ah, I suspect EM is forcing pymodbus to be downgraded in that case.

@nathanmarlor Could you do a new version of EM which has the right version of pymodbus in manifest.json?

miralemR, 2025-01-05:
Hi,
I uninstall Foxess EM and reinstall Foxess modbus. It was working for few minutes and stopped. I am getting this message...

Error connecting to inverter (friendly name: "Ngen h3"). Error reading registers. Type: RegisterType.HOLDING; start: 31000; count: 52; slave: 247 from tcp://192.168.68.78:502: Modbus Error: [Input/Output] Unable to decode request

screenagerbe, 2025-01-06:
> Hi, I uninstall Foxess EM and reinstall Foxess modbus. It was working for few minutes and stopped. I am getting this message...
>
> Error connecting to inverter (friendly name: "Ngen h3"). Error [...]

I've had the same issue this morning after a reboot when installing HACS 2.0.2. 
FoxESS-ModBus v1.13.0 was already installed last Saturday if I remember correctly.

so to resolve this I uninstalled Foxess-EM (too bad because I used it for the rather accurate prediction method)
restarted HA
after a while FoxESS-modbus gave some errors concerning writing and reading registers.
e.g.

```
This error originated from a custom integration.

Logger: custom_components.foxess_modbus.entities.modbus_entity_mixin
Source: custom_components/foxess_modbus/entities/modbus_entity_mixin.py:170
integration: FoxESS - Modbus (documentation, issues)
First occurred: 10:51:25 (3491 occurrences)
Last logged: 11:39:35

Value (254: 254) for entity 'sensor.foxess_time_period_2_start' address(es) '[41005, 41006]' failed validation against rule (Time : {})
Value (254: 254) for entity 'binary_sensor.foxess_time_period_2_enable_force_charge' address(es) '[41005]' failed validation against rule (Time : {})
Value (103: 103) for entity 'binary_sensor.foxess_time_period_1_enable_force_charge' address(es) '[41002]' failed validation against rule (Time : {})
Value (3326: 3326) for entity 'sensor.foxess_time_period_1_end' address(es) '[41003, 41002]' failed validation against rule (Time : {})
Value (103: 103) for entity 'sensor.foxess_time_period_1_start' address(es) '[41002, 41003]' failed validation against rule (Time : {})
```

and

```
This error originated from a custom integration.

Logger: custom_components.foxess_modbus.entities
Source: custom_components/foxess_modbus/entities/modbus_select.py:89
integration: FoxESS - Modbus (documentation, issues)
First occurred: 10:51:25 (500 occurrences)
Last logged: 11:42:45

Select option (60371) for address (41000) is not valid. Valid values: ({0: 'Self Use', 1: 'Feed-in First', 2: 'Back-up'})
Select option (60361) for address (41000) is not valid. Valid values: ({0: 'Self Use', 1: 'Feed-in First', 2: 'Back-up'})
Select option (60356) for address (41000) is not valid. Valid values: ({0: 'Self Use', 1: 'Feed-in First', 2: 'Back-up'})
Select option (60360) for address (41000) is not valid. Valid values: ({0: 'Self Use', 1: 'Feed-in First', 2: 'Back-up'})
Select option (60374) for address (41000) is not valid. Valid values: ({0: 'Self Use', 1: 'Feed-in First', 2: 'Back-up'})
```

I reloaded the integration and now it seems to be fine for the last hour

github-mr1903, 2025-01-08:
Unfortunately I have the same problem. Yesterday a new battery was installed and the firmware was updated.

HA: 2025.1.1
FoxESS - Modbus: 1.13.0

Dieser Fehler stammt von einer benutzerdefinierten Integration

Logger: custom_components.foxess_modbus.entities.modbus_entity_mixin
Quelle: custom_components/foxess_modbus/entities/modbus_entity_mixin.py:170
Integration: FoxESS - Modbus (Dokumentation, Probleme)
Erstmals aufgetreten: 08:45:22 (100 Vorkommnisse)
Zuletzt protokolliert: 10:49:52

Value (8420: 842.0) for entity 'sensor.foxess_485_pv1_current' address(es) '[11001]' failed validation against rule (Range : {'_min': 0, '_max': 100})
Value (20631: 2063.1) for entity 'sensor.foxess_485_pv1_current' address(es) '[11001]' failed validation against rule (Range : {'_min': 0, '_max': 100})
Value (1548: 154.8) for entity 'sensor.foxess_485_pv1_current' address(es) '[11001]' failed validation against rule (Range : {'_min': 0, '_max': 100})
Value (3653: 365.3) for entity 'sensor.foxess_485_pv1_current' address(es) '[11001]' failed validation against rule (Range : {'_min': 0, '_max': 100})
Value (19327: 1932.7) for entity 'sensor.foxess_485_pv1_current' address(es) '[11001]' failed validation against rule (Range : {'_min': 0, '_max': 100})
Dieser Fehler stammt von einer benutzerdefinierten Integration

Logger: custom_components.foxess_modbus.entities
Quelle: custom_components/foxess_modbus/entities/modbus_select.py:89
Integration: FoxESS - Modbus (Dokumentation, Probleme)
Erstmals aufgetreten: 08:45:22 (478 Vorkommnisse)
Zuletzt protokolliert: 10:59:52

Select option (15968) for address (41000) is not valid. Valid values: ({0: 'Self Use', 1: 'Feed-in First', 2: 'Back-up'})
Select option (15850) for address (41000) is not valid. Valid values: ({0: 'Self Use', 1: 'Feed-in First', 2: 'Back-up'})
Select option (15694) for address (41000) is not valid. Valid values: ({0: 'Self Use', 1: 'Feed-in First', 2: 'Back-up'})
Select option (16967) for address (41000) is not valid. Valid values: ({0: 'Self Use', 1: 'Feed-in First', 2: 'Back-up'})
Select option (17256) for address (41000) is not valid. Valid values: ({0: 'Self Use', 1: 'Feed-in First', 2: 'Back-up'})

screenagerbe, 2025-01-08:
> Unfortunately I have the same problem. Yesterday a new battery was installed and the firmware was updated.
> 
> HA: 2025.1.1 FoxESS - Modbus: 1.13.0

Did you try to reload the integration and then restart HA? Because that did fix it for me

You seem to be on the same versions as I do (only difference is my FoxESS installation is disconnected from the internet so I don't receive firmware updates).

github-mr1903, 2025-01-08:
> > Leider habe ich das gleiche Problem. Yesterday, a new battery was installed and the firmware was updated. > > HA: 2025.1.1 FoxESS - Modbus: 1.13.0 > > Have have to try re-laden the integration and then start HA reboot? Because that has it fixed for me > > They seem to be on the same versions that me (der einzige Unterschied ist, dass meine FoxESS-Installation vom Internet getrennt ist, so dass ich keine Firmware-Updates bekomme).  
Vielen Dank für Ihre Antwort. Ja, das habe ich versucht - leider hat es nicht funktioniert. Leider weiß ich nicht, was ich als nächstes tun soll. Ich kontrolliere meinen Batteriespeicher über HA. Da ich keine Perioden mehr einstellen kann, ist alles deaktiviert. Ich hoffe, dass jemand eine Lösung findet....

saleemkhalil, 2025-01-08:
You could always roll back your home assistant to a known working state until a fix is released...

Have have to re-laden the integration and then start HA re? Because that has it fixed for me

Thank you for your answer.
Yes, I tried that - unfortunately it didn't work.
Unfortunately I don't know what to do next. I control my battery storage via HA. ​​Since I can no longer set periods, everything is deactivated. I hope that someone finds a solution....

github-mr1903, 2025-01-09:
> You could always roll back your home assistant to a known working state until a fix is released... Sent from Outlook for Android<https://aka.ms/AAb9ysg>
> […](#)

Thanks, that worked :-) Unfortunately I lost 2 days of data as a result.

FozzieUK, 2025-01-09:
I’ll just leave this here in case anyone else decides to roll-back - you don’t need to restore a backup and lose data to do that, in HA open a terminal window and type this command 

`ha core update --version 2024.12.5`

Or replace the version number with the core version you want to go back to i..e 2024.11.2

It will say processing for a few minutes then restart and you will back to that version of core.

screenagerbe, 2025-01-09:
> Ah, I suspect EM is forcing pymodbus to be downgraded in that case.
> 
> @nathanmarlor Could you do a new version of EM which has the right version of pymodbus in manifest.json?

I'm not sure it is only FoxESS - EM which is causing this issue.
I have a second HA running with FoxESS - Modbus  without FoxESS - EM and I get exactly the same errors, but only after updating to HACS 2.0.2 yesterday. HA v2025.1.1 was already installed last Sunday.

saleemkhalil, 2025-01-17:
I've updated to the latest version of home assistant core and FoxEss modbus is still working. 

I haven't updated FoxEss though and also have not updated Hacs. 

I'll try hacs next!

screenagerbe, 2025-01-19:
Since updating pymodbus, the FoxESS - mobus integration is very unstable.

The error described here: https://github.com/nathanmarlor/foxess_modbus/issues/722#issuecomment-2572862654 is coming back at least once a day.

FozzieUK, 2025-01-20:
@screenagerbe looking at the post you referred to, that looks more like the data being read is being overflowed or another device is trampling it - is this a single adapter/ inverter install ?, and which adapter are you using ?

screenagerbe, 2025-01-20:
> [@screenagerbe](https://github.com/screenagerbe) looking at the post you referred to, that looks more like the data being read is being overflowed or another device is trampling it - is this a single adapter/ inverter install ?, and which adapter are you using ?

1 inverter
1 adapter: a WaveShare RS485 oto ETH POE (B).
2 HA installations.

It all worked flawlessly until that upgrade to HA v2025.1.0, with an active FoxESS - EM integration.
I've deleted EM first with reboot after, than I deleted Modbus with a reboot. Updated Modbus to v1.13.0 and initially it all looked OK, but after a day or three I started to get this error again. Values in HA are empty and the active workmode (Backup) was reset to default (Self Use).

FozzieUK, 2025-01-20:
> 1 inverter 1 adapter: a WaveShare RS485 oto ETH POE (B). 2 HA installations.
>
> It all worked flawlessly until that upgrade to HA v2025.1.0, with an active FoxESS - EM integration. I've deleted EM [...]

Hmmm if it works and then it doesn’t it sounds more of a timing thing - it is possible the latest pymodbus has slightly different response timing but I want to be sure I understand what you’ve got here - is that 2 instances of HA that are polling a single Waveshare on the same RS485 ?

If so, are the results consistent on both home assistants ?, is the Waveshare device set to multi-host mode?

screenagerbe, 2025-01-20:
> Hmmm if it works and then it doesn’t it sounds more of a timing thing - it is possible the latest pymodbus has slightly different response timing but I want to be sure I understand what you’ve got here - is that 2 instances of HA that are polling a single Waveshare on the same RS485 ?

yes

> If so, are the results consistent on both home assistants ?, is the Waveshare device set to multi-host mode?

yes and yes (if not mistaken it is by default multi-host when selecting Modbus TCP to RTU mode)

> When it happens, does it recover after a certain time or do you have to reset the HA, adapter or something else to get it back ?

I have to reload the Modbus integration in each HA instance.

thought
The firmware of my Waveshare is a bit old TBH, it's V1. 452, might consider to update that one first if I find some spare time

screenagerbe, 2025-01-21:
Both Poll Rate and Max Reads are not configured so there set to default values.
I've changed the latter to 40 and see what it gives.

saleemkhalil, 2025-01-24:
Just to update. I'm still on the older version of FoxEss modbus but on the latest version of HA and it's still working for me. Also have the the EM module installed with no issue.

booti101, 2025-01-24:
> [@booti101](https://github.com/booti101) (and [@saleemkhalil](https://github.com/saleemkhalil) ) the problem with modbus not working on HA 2025 is that the fix in v1.13.0 for foxess_modbus was to change to meet the new pymodbus spec and it specifies the latest version of pymodbus (>=3.7.4), whereas the EM application is specifying a much lower version (==3.1.3).
> 
> It is possible the integration that loads first wins that race and if EM gets in first the foxess_modbus won’t be able to load (or if EM loads second it works), but ultimately EM will need to change it’s version to co-exist properly on HA 2025

Thanks for the update and explanation

andrewmooreio, 2025-01-27:
No matter what I try, I can't get this integration working with 2025.1.4.

- I've uninstalled `FoxESS - Energy Management` to prevent pymodbus version conflicts with `FoxESS - Modbus`.
- Tried downgrading to v1.12.2 but that results in the error:
```
Setup failed for custom integration 'foxess_modbus': Unable to import component: cannot import name 'ModbusResponse' from 'pymodbus.pdu' (/usr/local/lib/python3.13/site-packages/pymodbus/pdu/__init__.py)
```
- Fresh reinstallation of v1.13.0 results in the error:
```
Setup failed for custom integration 'foxess_modbus': Unable to import component: No module named 'pymodbus.pdu.register_read_message'
```

Is there anything else I can try? I'd really rather avoid rolling back to Home Assistant 2024.12.5 if possible.

FozzieUK, 2025-01-27:
@andrewmooreio the error unable to import component with 2025.x on v1.12.2 is what we would expect after they changed it, the second error implies that something else is still forcing modbus to load a lower version - do you have any other modbus based applications ?.

With HA 2025 installed and v1.13.0, do a full reboot of the system (not a restart), so in Developer Tools, when you click the RESTART button drop the 'Advanced Options' and choose 'Reboot the System'

andrewmooreio, 2025-01-27:
Thanks @FozzieUK. I'll give it a system reboot rather than a restart.

I definitely don't have any other integrations that depend on pymodbus.

github-mr1903, 2025-01-27:
Hello 

Unfortunately I just can't get any further. I've tried various scenarios, but nothing works after the update :-(

1. update to 2024.1.12 -> 2025.1.4 = foxess_modbus can't load

2. System completely restarted 3x

3. Update foxess_modbus -> 1.13.0 = foxess_modbus load

4. System completely restarted 4x

Unfortunately without success.

FoxEss Workmode switches to unknown after a few seconds.

Manually changing the work mode causes errors.

I hope someone can help...

Thank you to everyone who is addressing the issue!!

>Dieser Fehler stammt von einer benutzerdefinierten Integration

Logger: custom_components.foxess_modbus.modbus_controller
Quelle: custom_components/foxess_modbus/modbus_controller.py:222
Integration: FoxESS - Modbus (Dokumentation, Probleme)
Erstmals aufgetreten: 13:55:50 (3 Vorkommnisse)
Zuletzt protokolliert: 13:56:28

Failed to write registers
Traceback (most recent call last):
  File "/config/custom_components/foxess_modbus/modbus_controller.py", line 222, in write_registers
    await self._client.write_registers(start_address, values, self._slave)
  File "/config/custom_components/foxess_modbus/client/modbus_client.py", line 193, in write_registers
    raise ModbusClientFailedError(
    ...<3 lines>...
    )
custom_components.foxess_modbus.client.modbus_client.ModbusClientFailedError: Error writing registers. Start: 41000; values: [0]; slave: 1. Received incorrect response type ReadInputRegistersResponse (2). Please ensure that your adapter is correctly configured to allow multiple connections, see the instructions at https://github.com/nathanmarlor/foxess_modbus/wiki from tcp://10.75.xx.xx:502: ReadInputRegistersResponse (2)

>Dieser Fehler stammt von einer benutzerdefinierten Integration

Logger: custom_components.foxess_modbus.entities
Quelle: custom_components/foxess_modbus/entities/modbus_select.py:89
Integration: FoxESS - Modbus (Dokumentation, Probleme)
Erstmals aufgetreten: 13:55:29 (4 Vorkommnisse)
Zuletzt protokolliert: 13:55:59

Select option (17264) for address (41000) is not valid. Valid values: ({0: 'Self Use', 1: 'Feed-in First', 2: 'Back-up'})
Select option (17574) for address (41000) is not valid. Valid values: ({0: 'Self Use', 1: 'Feed-in First', 2: 'Back-up'})

> Dieser Fehler stammt von einer benutzerdefinierten Integration

Logger: custom_components.foxess_modbus.client.modbus_client
Quelle: custom_components/foxess_modbus/client/modbus_client.py:148
Integration: FoxESS - Modbus (Dokumentation, Probleme)
Erstmals aufgetreten: 13:56:29 (2 Vorkommnisse)
Zuletzt protokolliert: 13:56:59

Error reading registers. Type: RegisterType.INPUT; start: 11000; count: 39; slave: 1. Received incorrect response type WriteRegisterResponse 41000 => 0. Please ensure that your adapter is correctly configured to allow multiple connections, see the instructions at https://github.com/nathanmarlor/foxess_modbus/wiki

FozzieUK, 2025-01-27:
@github-mr1903 as a quick test and to slow it down a bit in your HA go into Settings, Devices & Services, click on the > next to Foxess Modbus, then CONFIGURE, Advanced Settings and change your ‘Max Read’ value to 20

If you could post any error logs you then see please.

canton7, 2025-01-27:
@github-mr1903 You don't have the problem listed at the top of this thread: your problem is something different. Suggestions trying to fix @saleemkhalil's problem do not apply to you.

You say: 

> Yesterday a new battery was installed and the firmware was updated.

Are you sure that the inverter firmware update isn't the source of the problem?

github-mr1903, 2025-01-27:
> [@github-mr1903](https://github.com/github-mr1903) You don't have the problem listed at the top of this thread: your problem is something different. Suggestions trying to fix [...]

Hello and thank you for the message. I don't think (hope) that the firmware update is to blame, as everything works without problems under version 2024. 

Inverter hardware version: 
- Masters: 1.67
- Slave: 1.02
- Managers: 1.74

Thank you for your efforts!
GM

canton7, 2025-01-27:
@github-mr1903 Would you mind opening a new discussion, so that we can separate your problem out from the one addressed here?

screenagerbe, 2025-01-27:
> Both Poll Rate and Max Reads are not configured so there set to default values. I've changed the latter to 40 and see what it gives.

My issue seems to be resolved by reducing the Max Reads to 40 on one system and to 50 on the other.

miralemR, 2025-01-30:
Hy,
I tried every suggestions from here and still getting this...
Napaka izvira iz prilagojene integracije.

Logger: custom_components.foxess_modbus.modbus_controller
Source: custom_components/foxess_modbus/modbus_controller.py:269
integration: FoxESS - Modbus (documentation, issues)
First occurred: 06:55:03 (451 occurrences)
Last logged: 07:32:33

General exception when polling tcp://192.168.68.78:502 247: error('unpack requires a buffer of 4 bytes')
Traceback (most recent call last):
  File "/config/custom_components/foxess_modbus/modbus_controller.py", line 269, in _refresh
    reads = await self._client.read_registers(
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<4 lines>...
    )
    ^
  File "/config/custom_components/foxess_modbus/client/modbus_client.py", line 111, in read_registers
    response = await self._async_pymodbus_call(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<4 lines>...
    )
    ^
  File "/config/custom_components/foxess_modbus/client/modbus_client.py", line 212, in _async_pymodbus_call
    result = await self._hass.async_add_executor_job(_call)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/custom_components/foxess_modbus/client/modbus_client.py", line 209, in _call
    return call(*args)
  File "/usr/local/lib/python3.13/site-packages/pymodbus/client/mixin.py", line 103, in read_holding_registers
    return self.execute(no_response_expected, pdu_reg_read.ReadHoldingRegistersRequest(address=address, count=count, slave=slave))
           ~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/client/base.py", line 239, in execute
    return self.transaction.execute(no_response_expected, request)
           ~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transaction.py", line 241, in execute
    used_len, pdu = self.client.framer.processIncomingFrame(self.databuffer)
                    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/framer/base.py", line 77, in processIncomingFrame
    data_len, pdu = self._processIncomingFrame(data[used_len:])
                    ~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/framer/base.py", line 98, in _processIncomingFrame
    if (result := self.decoder.decode(frame_data)) is None:
                  ~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/pdu/decoders.py", line 104, in decode
    pdu.decode(frame[1:])
    ~~~~~~~~~~^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/pdu/register_write_message.py", line 271, in decode
    self.address, self.count = struct.unpack(">HH", data)
                               ~~~~~~~~~~~~~^^^^^^^^^^^^^
struct.error: unpack requires a buffer of 4 bytes

FozzieUK, 2025-01-30:
@miralemR if you can't get it working it's best to downgrade  - there are some strange things happening since HA core updated it's pymodbus version to 3.7.4 - for most it's working fine on core 2025.1.x (including me) but there are obviously some teething issues on some installations.

I'm hearing a few rumours that the pymodbus version may again be upgraded to 3.8.0 (because of bug fixes) in core 2025.2 but that may bring another set of breaking changes albeit hopefully more stable operation - if that is true i'd recommend for anyone having problems for now downgrade your core to 2024.12.5 and the integration to v1.12.2 and hold.

For everyone else keep a close eye on the changes in core 2025.2.x and don't upgrade if you see pymodbus mentioned.

miralemR, 2025-01-30:
Have already done that. 
So, few weeks ago after update to 2025.1 and all troubles I had , done the downgrades and let it downgraded and it was working fine till Tuesday... first it started with just few lost connection and then every hour more often till lost it totally. Since then I tried to upgrade, downgrade, change poll rate and max read.... nothing helps.

FozzieUK, 2025-01-30:
> Have already done that. So, few weeks ago after update to 2025.1 and all troubles I had , done the downgrades and let it downgraded and it was working fine till Tuesday... first it started with just [...]

What adapter are you using, if you downgraded and it worked again for a while - is it possible your adapter has locked up, changed it's IP address / config etc.. ?

miralemR, 2025-01-30:
> What adapter are you using, if you downgraded and it worked again for a while - is it possible your adapter has locked up, changed it's IP address / config etc.. ?

I'm using Elfin EW11 adapter. I tried ping from terminal in HA and it is responding

miralemR, 2025-01-31:
I figured out why it stopped working for me. My electricity provider, who also installed my solar system, updated the app for managing my H3 inverter. Now, in addition to setting the Threshold SOC, I can configure different power levels for peak shaving at various times (based on the hour, day of the week, months, and holidays). I can also set periods for charging and discharging the battery depending on electricity prices.

It's a great feature, but it's a shame I can't manage it via Home Assistant—I still have to manually adjust the Threshold SOC based on the weather forecast. It turns out the H3 inverter has a lot of potential.

Would this software be useful to anyone? I can ask my provider if they can share it.

canton7, 2025-02-01:
This should be fixed by v1.13.1.

closed 2025-02-01 by canton7