nathanmarlor/foxess_modbus #724: H3 Pro Latest Inverter firmware changes totals x 10

a900a900a900, 2025-01-06:
[image]

H3 pro 30 KW updated firmware to 1.25 and 1.56 and battery charge and discharge total has jumped by a decimal place 4800 to 48000. Feed in total, load energy and yield have also jumped. by the same
[Uploading Fox.log.txt…]()

Version of the custom_component
1.12.2

bradziszewski, 2025-01-08:
I’m experiencing the same issue on the H3 Pro with metrics like “Solar Generation Today,” “Solar Generation Total,” and several others.

[image]
[image]
[image]

FozzieUK, 2025-01-13:
@a900a900a900 ,  @bradziszewski I have checked the latest modbus spec and it is showing these registers as 'gain 100' whereas the entity is set to gain 10 scale (0.1). 

Which is a real pain as it is a departure from the scale 0.1 used for all of the inverters Fox have manufactured to date.... with the exception of the H3 Pro (and i'm guessing all models that will follow)

This seems as if Fox have realised and changed the latest H3 Pro firmware so that it matches the modbus spec - if you are able please could you confirm the correct scale by modifying your HA installation and the file `custom_components/foxess_modbus/entities/entity_descriptions.py` as follows 

Note: these line numbers are for integration version 1.12.2

entity solar_energy_total and change the scale from 0.1 to 0.01 (line #1404)

entity solar_energy_today and change the scale from 0.1 to 0.01 (line #1426)

entity battery_charge_total and change the scale from 0.1 to 0.01 (line #1448)

entity battery_charge_today and change the scale from 0.1 to 0.01 (line #1484)

entity battery_discharge_total and change the scale from 0.1 to 0.01 (line #1505)

entity battery_discharge_today and change the scale from 0.1 to 0.01 (line #1541)

entity feed_in_energy_total and change the scale from 0.1 to 0.01 (line #1562)

entity feed_in_energy_today and change the scale from 0.1 to 0.01 (line #1598)

entity grid_consumption_energy_total and change the scale from 0.1 to 0.01 (line #1619)

entity grid_consumption_energy_today and change the scale from 0.1 to 0.01 (line #1655)

entity total_yield_total and change the scale from 0.1 to 0.01 (line #1676)

entity total_yield_today and change the scale from 0.1 to 0.01 (line #1698)

entity input_energy_total and change the scale from 0.1 to 0.01 (line #1720)

entity input_energy_today and change the scale from 0.1 to 0.01 (line #1742)

entity load_power_total and change the scale from 0.1 to 0.01 (line #1764)

entity load_energy_today and change the scale from 0.1 to 0.01 (line #1807)

Save the file and restart HA - and if you could confirm that all of these sensors are now reporting the correct value please

a900a900a900, 2025-01-13:
This fix appears to fix the issue

> @a900a900a900 , @bradziszewski I have checked the latest modbus spec and it is showing these registers as 'gain 100' whereas the entity is set to gain 10 scale (0.1).
>
> Which is a real pain as it is [...]

MartB, 2025-01-16:
Did anyone test with the new H3 Pro 1.22 manager version?
Would be great to confirm this behavior.

AEG1974, 2025-03-05:
Fix Works With h3 Pro Manager 1.33 :-)

FozzieUK, 2025-03-07:
> Fix Works With h3 Pro Manager 1.33 :-)

Great, thanks very much for confirming - hopefully get these changes in the next release 👍

FozzieUK, 2025-03-14:
> It is unclear if the fix works with H3 Pro - Manager 1.33 correctly.
> Today, my H3-Pro 15.0 Inverter / Master / Slave was updated.
> 
> > Manager from v1.12 to v1.33
> > Master from v1.28 to v3.44
> > Slave from v1.06 to v2.62
> 
> 
> Immediately afterwards, I observed increased totals and today values.
> I resolved this by adjusting the scales mentioned above and began receiving correct values again.
The fix isn't in yet, so you will need to make the changes here for now.
We needed for the firmware to be formally released to confirm which manager version we were to apply it from.
Hopefully not long now to get this in the next release.

RP12Biker, 2025-03-15:
Can also confirm the global change with H3 Pro updating the master software from 1.53 to 1.58. No idea wheter the other components are patched too or not, was made from an external company. 
The patching inside entity_descriptions.py to the 0.01 also worked with foxess_modbus V1.13.4 with a little bit of searching the entities.

canton7, 2025-03-23:
@RP12Biker Do you know what your *manager* version was updated to? That's the important one.

canton7, 2025-03-23:
@FozzieUK Are we still waiting for a released version, or can we go with 1.33?

FozzieUK, 2025-03-23:
> [@FozzieUK](https://github.com/FozzieUK) Are we still waiting for a released version, or can we go with 1.33?

@canton7 yes we can go with 1.33 now - i’ll need to send you a quick message so will do in facebook.

RP12Biker, 2025-03-24:
> [@RP12Biker](https://github.com/RP12Biker) Do you know what your _manager_ version was updated to? That's the important one.

Sorry. Yes, its the manager version 1.33 and the company gave me quick instructions where to look up it inside the webinterface.

smash111111, 2025-05-01:
Had to migrate my H3 Pro from inverter Master Version 1.31 to 1.58 last Monday.
Inverter Version 1.31 was working fine with foxess-modbus integration 1.13.5b7. However, after migrating my inverter to 1.58 loads of entities came in with wrong scales. Hower, can confirm to fix it successfully with the solution described above from FozzieUK.

Thanks for it!

andyjenkinson, 2025-05-09:
@FozzieUK do you, @canton7 or anyone have a fork preparing an update for this? I can contribute a fix if it's not already in progress.

Do we know exactly which version it was introduced yet? Could it be 1.22 when some other changes were introduced? There's already a version split for that in the code when they changed the version number decoding. I don't think it can be as late as 1.33 as the initial report was for 1.25 (same version as #774 ). It would be a lot simpler to not have to create another version set...

There are some other fixes needed for H3 Pro too it seems (invalid registers - see #837) but perhaps we can work out what changed when to get a fix together.

FozzieUK, 2025-05-09:
@andyjenkinson the first formal release was Manager (ARM) 1.33 it was seen a bit earlier but only in development releases - everything between 1.22 and 1.33 are only field test releases and are unavailable to installers - the first official release is 1.33.

as I mentioned in #837 we were waiting for a few H3 Pro releases in case they changed it back as we now have conflicting scales in the old and new specs - It does need a PR doing to implement 'scale' for the total registers as all other inverter types have a scale of 10, the H3 Pro is now unique in that it has a scale of 100 (we still have no idea why).

As per my comment once you have changed your max read and implemented the changes above, I don't think you'll be seeing any invalid registers as we don't access anything above 37632 until we get to 37700, but any testing you can do to confirm this on the H3 Pro would be very helpful, thanks.

andyjenkinson, 2025-05-09:
@FozzieUK I was more thinking about ease of implementation; it may be a dev release but 1.22 already has a version split in the codebase so if the change is observed from that release it'd be easier to use it instead of 1.33. Do FoxESS actually publish releases properly? I don't see a source of what is or isn't an official release or any notes, so it's harder to contribute without being able to see what changed when.

I already reduced the number of registers to 40 and this made the entities actually load (it attempted to load 92 entities from 37609 onwards initially), I just figured the UX of that isn't great, I wasn't sure why it would attempt to query these registers in the first place based on the config, but I'm still in the learning curve to understand how the integration works.

I still see an attempt to read invalid register 37769 in my logs, running v1.13.5b7.

On mine (1.34), 37632 to 37635 are all valid registers. In those ranges you mentioned, these are valid:
37609 to 37635
37700 to 37718
However I can't determine what they mean just yet as everything from 37609 to 37635 reads zero on my installation (probably because my battery isn't connected yet), as does everything from 37700 to 37718. Then 37719 onwards are invalid.

In addition looking at the code, I'd expect some 410xx registers to work but all of these appear to be invalid now, not only the ones listed as invalid in H3_REGISTERS. I suspect the new H3_SMART_REGISTERS configuration in v1.13.5b7 is closer to what my version of H3 Pro now does, except 37636 is definitely invalid. I guess the ranges just need updating? I may do some tinkering.

What else do you need me to test? And let me know if you want any help with code after waiting for some more releases. It does seem to have persisted for now from 1.33 to 1.34 (and I think I saw someone else on 1.35), like you I would guess they intend to roll it out to other devices too.

andyjenkinson, 2025-05-09:
Oh by the way just for posterity: the for 1.13.5 already has scale parameterised for these entities, it's just only being used for the H3 SMART model at the moment, so the code change will be to add the newer H3 Pro to the list of matching models alongside H3 Smart. However permanent this change is, they've so far implemented it for H3 Smart and H3 Pro it appears.

FozzieUK, 2025-05-09:
> Oh by the way just for posterity: the for 1.13.5 already has scale parameterised for these entities, it's just only being used for the H3 SMART model at the moment, so the code change will be to add [...]

Ahhh you’re right it does, tbh I had some urgent surgery recently and had to spend 5 weeks in hospital - so I missed the H3 smart being added - it certainly does appear to follow the latest spec which will make the change to add the H3 Pro much so easier.

closed 2025-08-13 by canton7