nathanmarlor/foxess_modbus #727: H1 Gen 1 - Master firmware v1.67

ryaaanmorgan, 2025-01-10:
After updating from Master v1.65 to the latest Master v1.67 on my H1 G1 inverter, the integration shows the following errors under the connected status entity.

**After upgrading the firmware**
Error reading registers. Type: RegisterType.INPUT; start: 11000; count: 8; slave: 247 from udp://192.168.0.83:502: Modbus Error: [Input/Output] Unable to decode request

**After restarting HA**
Error reading registers. Type: RegisterType.INPUT; start: 10016; count: 3; slave: 247 from udp://192.168.0.83:502: Modbus Error: [Input/Output] timed out

And no matter how many restarts of HA or the inverter - I see the above error.

The logs don't seem to provide much more detail.

This error originated from a custom integration.

Logger: custom_components.foxess_modbus.modbus_controller
Source: custom_components/foxess_modbus/modbus_controller.py:347
integration: FoxESS - Modbus (documentation, issues)
First occurred: 1:31:59 pm (1 occurrences)
Last logged: 1:31:59 pm

udp://192.168.0.83:502 247 - 5 failed poll attempts: now not connected. Last error: Error reading registers. Type: RegisterType.INPUT; start: 10016; count: 3; slave: 247 from udp://192.168.0.83:502: Modbus Error: [Input/Output] timed out

Any ideas??

Serves me right I suppose haha

FozzieUK, 2025-01-10:
It's definitely not responding to any of those requests which looks more like a physical transport error (or the inverter is not replying) - but it's usually Manager that controls the modbus registers and Master is all of the real time things - what version of Manager are you on ?

There's no chance the adaptor has taken the opportunity to change it's IP address - can you address that on 192.168.0.83 ?

TonyM1958, 2025-01-10:
I'm running Master 1.68 and Manager 1.74 on my inverter with my integration and it's OK. I think this may be a configuration problem, assuming the hardware is still OK.

Fox has been dropping the Modbus Input registers that you appear to be using but keeping the holding registers. On my integration, most of the 11000 input registers were moved to 31000 holding registers a few months back. The only input registers that work are in the range 11034 - 11058 and mainly provide BMS related info like Residual, Cell Volts and Cell Temps.

https://github.com/TonyM1958/HA-FoxESS-Modbus/blob/main/custom_components/HA-FoxESS-Modbus/modbusH1G1_RS485.yaml

ryaaanmorgan, 2025-01-10:
Thnk you both for your responses!

You were spot on Dave.
I've just spent the best part of an hour up the bloody loft sorting out the USR-W610.

I don't know why but it did disconnect from my network. Not sure why that would have happened at the exact moment I upgraded the firmware - I'm sure there's an explanation but it's working again now.

Phew!

Thanks!

closed 2025-01-10 by ryaaanmorgan