nathanmarlor/foxess_modbus #748: foxess_modbus not working with Home Assistant 2025.2 rc - No module named 'pymodbus.pdu.register_read_message'

stevetrease, 2025-01-30:
Starting Home Assistant 2024.2 RC and foxess_modbus v1.13.0 results in an "invalid config" notification in Home Assistant.

Let me know if I can provide more information or help with testing a resolution....

## Version of the custom_component
- foxess_modbus v1.13.0 
- Home Assistant Core 2025.2.0b1

## Configuration

## Describe the bug
Starting Home Assistant 2024.2 RC and foxess_modbus v1.13.0 results in an "invalid config" notification in Home Assistant. 
```
Invalid config
The following integrations and platforms could not be set up:
foxess_modbus
Please check your config and [logs](https://hatest.tail5e5d.ts.net/config/logs).
```

## Debug log
Logger: homeassistant.setup
Source: setup.py:334
First occurred: 14:27:14 (1 occurrences)
Last logged: 14:27:14

Setup failed for custom integration 'foxess_modbus': Unable to import component: No module named 'pymodbus.pdu.register_read_message'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 334, in _async_setup_component
    component = await integration.async_get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1035, in async_get_component
    self._component_future.result()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1015, in async_get_component
    comp = await self.hass.async_add_import_executor_job(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        self._get_component, True
        ^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1075, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 201, in protected_loop_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/foxess_modbus/__init__.py", line 20, in <module>
    from .client.modbus_client import ModbusClient
  File "/config/custom_components/foxess_modbus/client/modbus_client.py", line 18, in <module>
    from pymodbus.pdu.register_read_message import ReadHoldingRegistersResponse
ModuleNotFoundError: No module named 'pymodbus.pdu.register_read_message'
```

canton7, 2025-01-30:
Do you mean HA 2025.2 or 2024.2? Your title and question disagree.

canton7, 2025-01-30:
Oh for goodness sake.

canton7, 2025-01-30:
I wonder whether we can just vendor pymodbus ourselves, and avoid taking a dependency on whatever version HA has decided to install. Or switch to one of the other modbus libraries out there (umodbus / modbus-tk).

stevetrease, 2025-01-30:
> Do you mean HA 2025.2 or 2024.2? Your title and question disagree.

Sorry, typo on my part, 2025.2. Released yesterday.

stevetrease, 2025-01-30:
> Do you have any other integrations which use modbus, such as foxess-em?

No. This is my test Home Assistant instance. Just Home Assistant and the foexess_Modbus integration - connecting to a Waveshare Modbus adapter over IP.

closed 2025-02-01 by canton7

stevetrease, 2025-02-01:
I see you've released v1.13.1 this afternoon. This is now working on my Home Assistant test system (running Home Assistant 2025.2.0b5).

Many thanks for the really fast turn round. 

Cheers,

Steve

canton7, 2025-02-01:
Thanks for confirming!

PR #749 (2025-02-01): Vendor pymodbus, rather than relying on the version which HA installs
This should protect us from HA breaking things in the future!

Fixes: #748
commit 405939ed (2025-02-01): Vendor pymodbus, rather than relying on the version which HA installs

This should protect us from HA breaking things in the future!

Fixes: #748