nathanmarlor/foxess_modbus #753: pymodbus error still occurs with Docker-based HA and v1.13.1

pz-earthling, 2025-02-01:
## Version of the custom_component

v1.13.1

## HA Version 
2025.1.4 docker image

## Configuration

nothing special

## Describe the bug

After updating to v1.13.1 foxess_modbus didn't start anymore. Downgrading to v1.13.0 also did not help.

After checking the HA logs and the vendoring PR, I downgraded the pymodbus version inside the home assistant docker container to 3.6.9.
Looks like just vendoring doesn't help if a newer pymodbus is installed in /usr/local/lib/python3.13/site-packages (as it is inside the docker container).

## Debug log

```text
2025-02-01 21:43:09.852 ERROR (MainThread) [homeassistant.setup] Setup failed for custom integration 'foxess_modbus': Unable to import component: No module named 'pymodbus.register_read_message'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 334, in _async_setup_component
    component = await integration.async_get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1035, in async_get_component
    self._component_future.result()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1015, in async_get_component
    comp = await self.hass.async_add_import_executor_job(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        self._get_component, True
        ^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1075, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 200, in protected_loop_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/foxess_modbus/__init__.py", line 19, in <module>
    from .client.modbus_client import ModbusClient
  File "/config/custom_components/foxess_modbus/client/modbus_client.py", line 23, in <module>
    from ..vendor.pymodbus import ModbusResponse
  File "/config/custom_components/foxess_modbus/vendor/pymodbus/__init__.py", line 13, in <module>
    from pymodbus.register_read_message import ReadHoldingRegistersResponse
ModuleNotFoundError: No module named 'pymodbus.register_read_message'

```

Badlee2020, 2025-02-01:
I got the error too, I had to delete it, install 1.13.0 then restart. Then re configure the integration.

canton7, 2025-02-01:
Mind trying v1.13.2b1?

Badlee2020, 2025-02-01:
1.13.2b1 starts, but I get this error...Error reading registers. Type: RegisterType.HOLDING; start: 31000; count: 91; slave: 196 from tcp://192.168.2.235:502: Exception Response(131, 3, IllegalAddress)

canton7, 2025-02-01:
OK great, that's something different. Mind opening a new ticket, and include your inverter model and manager version?

The inverter is successfully talking to us, it just doesn't like the register we're reading.

canton7, 2025-02-01:
Actually, leaving open until this is released as a non-beta version.

dropqube, 2025-02-02:
Adding to this -> I'm using Home Assistant Green and also get the pymodbus error

```
Logger: homeassistant.config_entries
Source: config_entries.py:3314

Error occurred loading flow for integration foxess_modbus: No module named 'pymodbus.register_read_message'
```

Using H3 inverter
Firmware versions:
Master: 2.0.5
Slave: 1.0.3
Manager: 1.7.7

canton7, 2025-02-02:
This should be fixed in v1.13.2.

closed 2025-02-02 by canton7

abuesen, 2025-02-02:
Hi @canton7.

In my environment it works. Thank you for the quick fix.

Best regards, A.