nathanmarlor/foxess_modbus #756: Issues with registers

Badlee2020, 2025-02-01:
With version 1.13.2b1 I get the following error:

Error reading registers. Type: RegisterType.HOLDING; start: 31000; count: 91; slave: 196 from tcp://192.168.2.235:502: Exception Response(131, 3, IllegalAddress)

With version 1.13.0 every was ok

Manager 1.62
Master 1.74
Slave 1.03

I have a  KOPP-KUARA 12.0-3H

canton7, 2025-02-01:
Right, I see. It looks like #756 added support for battery SoH, but this isn't supported by the Kuara. I'll fix.

canton7, 2025-02-01:
Can you just check quickly whether you can read holding registers 31102, 31103, 31123? Developer tools -> Actions -> FoxESS - Modbus: Read Registers.

canton7, 2025-02-01:
And do you know whether the Kuara supports Peak Shaving work mode?

Badlee2020, 2025-02-02:
31102, 31103, 31123 these registers produce errors. Peek shaving is not a work mode I have seen. More than willing to test it though:) the documentation with Kopp is quite 😔

jbarrett-fmad, 2025-02-02:
FYI I'm getting the same error on KH10.5 Manager version 1.21. Have requested firmware update from Fox to see if that resolves it.

hooktheboss, 2025-02-02:
H3-6.0-E - AUX no connection after Update to v1.13.1

Error connecting to inverter (friendly name: "FoxESSInverter"). Error reading registers. Type: RegisterType.HOLDING; start: 31000; count: 91; slave: 247 from tcp://192.168.178.17:512: Exception Response(131, 3, IllegalAddress)

Manager 1.56
Master 1.67
Slave 1.03

Reverted to v1.13.0 - working OK.

AndaleR, 2025-02-02:
Same here.

Worst case: Will all data be disappeared? Or is there a solution and it just needs to be updated and all the energy-data will be back in HomeAssistant.

canton7, 2025-02-02:
Right, I've releaed v1.13.2 which reverts those changes. I'll get them inot the next beta, and we can work out what firmware versions actually support them.

iancg, 2025-02-02:
> Right, I've released v1.13.2 which reverts those changes. I'll get them back into the next beta, and we can work out what firmware versions actually support them.

Would another approach be to try reading those registers as a test read after startup, and if success then add them in to the map, otherwise log and skip them? 

That may give a way to handle multiple versions without actually hard coding exactly which registers are available in which version.

It may need some validation to handle cases where you don't get illegal address, but just no data. Might not work if the same address is used for different things in different versions (but I don't think I've seen any evidence of that so far?)

AndaleR, 2025-02-02:
Thanks. I updated to 1.13.2 - but my energy-data are gone.

But actually it’s collecting data from the inverter.

WyndStryke, 2025-02-02:
> . I'll get them back into the next beta, and we can work out what firmware versions actually support them.

Presumably at least the KH1.33+ can read them, at the minimum.  I'm not in a position to test the earlier firmware.

Can the integration trap failures and disable individual sensors if they can't be read?  Or is it a hard failure?

canton7, 2025-02-02:
@iancg I've got a branch somewhere which attempts to do that. It's not particularly straightforward:

* We read large blocks of registers, but the inverter only tells us if one of the block was invalid, so we need logic to work out which particular register was bad
* We can't avoid creating the entity: we'll just have to leave it in an "Unknown" state
* It means we won't necessarily find out if we've got the register map incorrect, as people won't tell us

All solvable, but it just needs time to get it over the line. Feel free to pick it up!

canton7, 2025-02-02:
@AndaleR There's no reason HA just decide to delete all of your energy data just because an integration wasn't working for a bit. You won't have been collecting new data obviously, but that shouldn't affect old data. If your HA instance does delete old energy data when you disable an integration, you should raise that as a bug with HA.

canton7, 2025-02-02:
Right, closing this off, as v1.13.2 is out. Discussion on re-enabling these registers for the appropriate inverters is in #760.

closed 2025-02-02 by canton7