nathanmarlor/foxess_modbus #762: Error reading registers after latest update

AndyMillne, 2025-02-02:
## Version of the custom_component

1.13.2 but shows as 1.1.0

## Configuration

KH10.5
HA 2025.1.4

## Describe the bug

After upgrading to latest, entities are no longer available. 

Error reading registers. Type: RegisterType.HOLDING; start: 36001; count: 3; slave: 247 from tcp://192.168.1.254:502: Exception Response(131, 3, IllegalAddress)

## Debug log

```text

2025-02-02 22:45:20.373 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration teamtracker which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-02-02 22:45:20.374 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration foxess_modbus which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-02-02 22:45:20.374 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-02-02 22:45:20.375 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration myenergi which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-02-02 22:45:20.375 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration alarmo which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-02-02 22:45:20.375 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration octopus_energy which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-02-02 22:45:20.376 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration webrtc which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-02-02 22:45:24.590 WARNING (ImportExecutor_0) [homeassistant.core] Config was used from myenergi, this is a deprecated alias which will be removed in HA Core 2025.11. Use homeassistant.core_config.Config instead, please report it to the author of the 'myenergi' custom integration
2025-02-02 22:45:27.024 WARNING (MainThread) [homeassistant.helpers.frame] Detected that custom integration 'myenergi' calls `device_registry.async_get_or_create` referencing a non existing `via_device` 22702384, with device info: {'identifiers': {('myenergi', 22702384)}, 'manufacturer': 'myenergi', 'model': 'Zappi', 'name': 'zappi-22702384', 'sw_version': '3562S5.542', 'via_device': '22702384'} at custom_components/myenergi/sensor.py, line 689: async_add_devices(sensors). This will stop working in Home Assistant 2025.12.0, please create a bug report at https://github.com/cjne/ha-myenergi/issues
2025-02-02 22:45:27.067 WARNING (MainThread) [homeassistant.helpers.frame] Detected that custom integration 'myenergi' calls `device_registry.async_get_or_create` referencing a non existing `via_device` 22702384, with device info: {'identifiers': {('myenergi', 22702384)}, 'manufacturer': 'myenergi', 'model': 'Zappi', 'name': 'zappi-22702384', 'sw_version': '3562S5.542', 'via_device': '22702384'} at custom_components/myenergi/binary_sensor.py, line 62: async_add_devices(sensors). This will stop working in Home Assistant 2025.12.0, please create a bug report at https://github.com/cjne/ha-myenergi/issues
2025-02-02 22:45:27.072 WARNING (MainThread) [homeassistant.helpers.frame] Detected that custom integration 'myenergi' calls `device_registry.async_get_or_create` referencing a non existing `via_device` 22702384, with device info: {'identifiers': {('myenergi', 22702384)}, 'manufacturer': 'myenergi', 'model': 'Zappi', 'name': 'zappi-22702384', 'sw_version': '3562S5.542', 'via_device': '22702384'} at custom_components/myenergi/select.py, line 95: async_add_devices(devices). This will stop working in Home Assistant 2025.12.0, please create a bug report at https://github.com/cjne/ha-myenergi/issues
2025-02-02 22:45:27.083 WARNING (MainThread) [homeassistant.helpers.frame] Detected that custom integration 'myenergi' calls `device_registry.async_get_or_create` referencing a non existing `via_device` 22702384, with device info: {'identifiers': {('myenergi', 22702384)}, 'manufacturer': 'myenergi', 'model': 'Zappi', 'name': 'zappi-22702384', 'sw_version': '3562S5.542', 'via_device': '22702384'} at custom_components/myenergi/number.py, line 28: async_add_devices(devices). This will stop working in Home Assistant 2025.12.0, please create a bug report at https://github.com/cjne/ha-myenergi/issues
2025-02-02 22:45:31.503 WARNING (MainThread) [homeassistant.util.loop] Detected blocking call to open with args ('/usr/local/lib/python3.13/site-packages/pytz/zoneinfo/Africa/Abidjan', 'rb') inside the event loop by integration 'caldav' at homeassistant/components/caldav/coordinator.py, line 111: if not hasattr(event.instance, "vevent"): (offender: /usr/local/lib/python3.13/site-packages/pytz/__init__.py, line 108: return open(filename, 'rb')), please create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+caldav%22
For developers, please see https://developers.home-assistant.io/docs/asyncio_blocking_operations/#open
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/usr/src/homeassistant/homeassistant/__main__.py", line 227, in <module>
    sys.exit(main())
  File "/usr/src/homeassistant/homeassistant/__main__.py", line 213, in main
    exit_code = runner.run(runtime_conf)
  File "/usr/src/homeassistant/homeassistant/runner.py", line 154, in run
    return loop.run_until_complete(setup_and_run_hass(runtime_config))
  File "/usr/local/lib/python3.13/asyncio/base_events.py", line 707, in run_until_complete
    self.run_forever()
  File "/usr/local/lib/python3.13/asyncio/base_events.py", line 678, in run_forever
    self._run_once()
  File "/usr/local/lib/python3.13/asyncio/base_events.py", line 2033, in _run_once
    handle._run()
  File "/usr/local/lib/python3.13/asyncio/events.py", line 89, in _run
    self._context.run(self._callback, *self._args)
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 355, in async_refresh
    await self._async_refresh(log_failures=True)
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 379, in _async_refresh
    self.data = await self._async_update_data()
  File "/usr/src/homeassistant/homeassistant/components/caldav/coordinator.py", line 111, in _async_update_data
    if not hasattr(event.instance, "vevent"):

2025-02-02 22:45:31.551 ERROR (SyncWorker_2) [root] Ical data was modified to avoid compatibility issues
(Your calendar server breaks the icalendar standard)
This is probably harmless, particularly if not editing events or tasks
(error count: 1 - this error is ratelimited)
NoneType: None
2025-02-02 22:45:31.596 ERROR (SyncWorker_5) [root] Ical data was modified to avoid compatibility issues
(Your calendar server breaks the icalendar standard)
This is probably harmless, particularly if not editing events or tasks
(error count: 2 - this error is ratelimited)
NoneType: None
2025-02-02 22:45:31.597 ERROR (SyncWorker_2) [root] --- 
+++ 
@@ -5,7 +5,6 @@
[... 329 lines cut ...]
2025-02-02 23:27:31.996 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Modbus error when polling tcp://192.168.1.254:502 247: Exception Response(131, 3, IllegalAddress)
2025-02-02 23:27:41.770 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses on tcp://192.168.1.254:502 247: (31000, 43)
2025-02-02 23:27:41.887 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses on tcp://192.168.1.254:502 247: (32000, 24)
2025-02-02 23:27:41.966 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses on tcp://192.168.1.254:502 247: (36001, 3)
[... cut ...]
2025-02-02 23:28:21.891 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses on tcp://192.168.1.254:502 247: (32000, 24)
2025-02-02 23:28:21.971 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses on tcp://192.168.1.254:502 247: (36001, 3)
2025-02-02 23:28:22.001 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Modbus error when polling tcp://192.168.1.254:502 247: Exception Response(131, 3, IllegalAddress)

```

AndyMillne, 2025-02-02:
Rolled back to 1.13.0 which got me back up and running.

canton7, 2025-02-03:
Right, there's a couple of things going on.

The first is that you need to read the first section of the release notes:

If your inverter is not running the latest manager version, you may need to select your manager version in the integration settings. See [the wiki](https://github.com/nathanmarlor/foxess_modbus/wiki/Supported-Versions) for details.

| Inverter | Supported by default | Selectable in settings |
|---|---|---|
| H1 G1 / AC1 / AIO | All | |
| H1 G2 | v1.44 | All |
| H3 | All | |
| KH / KA | v1.33 | All |

Since you're running an older manager version on your KH, you need to select the correct version in the settings. v1.13.1 added support for the KH version 1.33, but you're on an older version.

The second is that there are a couple of other problems being highlighted in your log, which I need to fix.

closed 2025-02-03 by canton7