nathanmarlor/foxess_modbus #764: Error writing to registers

djdarlo, 2025-02-03:
When trying to update the Min SOC (On Grid) values using the modbus integration I got the following error.  I'm on the latest version of code from yesterday (02/02).  

Stopped because an error was encountered at February 3, 2025 at 12:30:54 AM (runtime: 6.12 seconds)

Error writing registers. Start: 41011; values: [100]; slave: 247 from /dev/ttyUSB0: Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 4 bytes (0 received)

Further info from the logs is below...

This error originated from a custom integration.

Logger: custom_components.foxess_em.fox.fox_modbus
Source: custom_components/foxess_em/fox/fox_modbus.py:89
integration: FoxESS - Energy Management (documentation, issues)
First occurred: 12:30:54 AM (1 occurrences)
Last logged: 12:30:54 AM

Error writing holding register - retry (1/5): Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 4 bytes (0 received)

This error originated from a custom integration.

Logger: custom_components.foxess_modbus.modbus_controller
Source: custom_components/foxess_modbus/modbus_controller.py:222
integration: FoxESS - Modbus (documentation, issues)
First occurred: 12:30:54 AM (1 occurrences)
Last logged: 12:30:54 AM

Failed to write registers
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 4 bytes (0 received)

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/config/custom_components/foxess_modbus/modbus_controller.py", line 222, in write_registers
    await self._client.write_registers(start_address, values, self._slave)
  File "/config/custom_components/foxess_modbus/client/modbus_client.py", line 180, in write_registers
    raise ModbusClientFailedError(message, self, response) from response
custom_components.foxess_modbus.client.modbus_client.ModbusClientFailedError: Error writing registers. Start: 41011; values: [100]; slave: 247 from /dev/ttyUSB0: Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 4 bytes (0 received)

This is the code I used.  The continue_on_error:true doesn't seem to be working either.

alias: Octopus Intelligent Go Switch On with Retry
description: ""
triggers:
  - trigger: numeric_state
    entity_id:
      - sensor.octopus_energy_electricity_XXXXX_XXXXX_current_rate
    below: 0.09
conditions: []
actions:
  - repeat:
      sequence:
        - device_id: 4012bd0489b8e4dfce98d801e02f43b7 (that's my H6 inverter)
          domain: number
          entity_id: 03c88a94be45c00e47b04b436421036c (that's the Min SOC on grid entity)
          type: set_value
          value: 100
          continue_on_error: true
        - condition: state
          entity_id: sensor.h6_min_soc_on_grid
          state: "100"
        - delay:
            seconds: 5
      until:
        - condition: state
          state: "100"
          entity_id: sensor.h6_min_soc_on_grid
  - data:
      message: >
        Octopus Intelligent Go Switch On:  {{ 'Successful' if
        is_state('sensor.h6_min_soc_on_grid', '100') else 'Failed after multiple
        retries' }}
    action: notify.persistent_notification
mode: single

canton7, 2025-02-03:
Thanks. 

Which inverter? What manager version? Have you been able to write to this entity before? Do you have any other integrations communicating over modbus (eg foxess_em or the modbus integration)?

djdarlo, 2025-02-03:
It's a H1-6.0 G2  Manager 1.71.  I was able to write the entity before and
was able to write it later.  I suppose I am wondering why my retry option
didn't working in the automation?  This bit....  I am also running Fox ESS
- Energy Manager to stop and start charging - planning to migrate this
across to modbus when I can work out how best to do this.

actions:
  - repeat:
      sequence:
        - device_id: 4012bd0489b8e4dfce98d801e02f43b7
          domain: number
          entity_id: 03c88a94be45c00e47b04b436421036c
          type: set_value
          value: 100
*          continue_on_error: true*
        - condition: state
          entity_id: sensor.h6_min_soc_on_grid
          state: "100"
        - delay:
            seconds: 5
      until:
        - condition: state
          state: "100"
          entity_id: sensor.h6_min_soc_on_grid

FozzieUK, 2025-02-03:
If your Manager is 1.71 it isn’t a G2, it’s a G1

Also your automation will not work as you cannot set the entity directly like that.

You should be using the action number:set and setting the number.h6_min_soc_on_grid entity, like this -

```
action: number.set_value
metadata: {}
data:
  value: "100"
target:
  entity_id: number.h6_min_soc_on_grid
```

canton7, 2025-02-03:
@FozzieUK It did work though -- we tried to write to the inverter. The inverter didn't respond for some reason, hence the error, but it definitely triggered us to try.

FozzieUK, 2025-02-03:
I've found when you try and update an entity like this you can get very strange results (but Number: set always works) - is it possible that changing state during a poll would see the write fail ?

djdarlo, 2025-02-03:
Could you help me by showing where I'd but this into my retry loop and I'll try it...

actions:
  - repeat:
      sequence:
        - device_id: 4012bd0489b8e4dfce98d801e02f43b7
          domain: number
          entity_id: 03c88a94be45c00e47b04b436421036c
          type: set_value
          value: 100
          continue_on_error: true
        - condition: state
          entity_id: sensor.h6_min_soc_on_grid
          state: "100"
        - delay:
            seconds: 5
      until:
        - condition: state
          state: "100"
          entity_id: sensor.h6_min_soc_on_grid
  - data:
      message: >
        Octopus Intelligent Go Switch On:  {{ 'Successful' if
        is_state('sensor.h6_min_soc_on_grid', '100') else 'Failed after multiple
        retries' }}
    action: notify.persistent_notification

FozzieUK, 2025-02-03:
Your repeat sequence would look like this, note i've changed the delay to 20 seconds - 5 seconds is way too short because of the polling time and that in itself could result in multiple writes taking place (that might have been why it errored).

Also note that continue on error won't function as the setting of the entity will always return true so it serves no purpose - your repeat sequence loop will set it again if it doesn't work first time.

```
repeat:
  sequence:
    - action: number.set_value
      metadata: {}
      data:
        value: "100"
      target:
        entity_id: number.h6_min_soc_on_grid
    - delay:
        hours: 0
        minutes: 0
        seconds: 20
        milliseconds: 0
  until:
    - condition: state
      entity_id: sensor.h6_min_soc_on_grid
      state: "100"
```

canton7, 2025-02-03:
> I've found when you try and update an entity like this you can get very strange results (but Number: set always works) - is it possible that changing state during a poll would see the write fail ?

They should come into the integration in exactly the same way: there's only one `ModbusNumber.async_set_native_value` method. And that method takes the central modbus lock, so it shouldn't conflict with a read.

djdarlo, 2025-02-04:
Hey guys - I tried the changes last night and hit the same error.

Stopped because an error was encountered at February 4, 2025 at 12:30:50 AM (runtime: 6.11 seconds)

Error writing registers. Start: 41011; values: [100]; slave: 247 from /dev/ttyUSB0: Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 4 bytes (0 received)

Interesting I ran the same command at 12:35, 12:40 and 12:45 and all executions worked so its feels like a conflict of some sort.  Do you want me to enable enhanced logging?  I have foxess em running which kicked off an foxess_em.start_force_charge_now call at 12:30:44 AM.  I'm also thinking, is there a way to replicate the start_force_charge_now in modbus?

djdarlo, 2025-02-04:
Some more log info if this helps....

2025-02-04 00:30:50.499 WARNING (MainThread) [custom_components.foxess_em.fox.fox_modbus] Error writing holding register - retry (1/5): Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 4 bytes (0 received)
2025-02-04 00:30:50.516 ERROR (MainThread) [custom_components.foxess_modbus.modbus_controller] Failed to write registers
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 4 bytes (0 received)

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/config/custom_components/foxess_modbus/modbus_controller.py", line 222, in write_registers
    await self._client.write_registers(start_address, values, self._slave)
  File "/config/custom_components/foxess_modbus/client/modbus_client.py", line 180, in write_registers
    raise ModbusClientFailedError(message, self, response) from response
custom_components.foxess_modbus.client.modbus_client.ModbusClientFailedError: Error writing registers. Start: 41011; values: [100]; slave: 247 from /dev/ttyUSB0: Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 4 bytes (0 received)
2025-02-04 00:30:50.532 ERROR (MainThread) [homeassistant.components.automation.octopus_intelligent_go_switch_on_with_retry_v2] Octopus Intelligent Go Switch On with Retry v2: Repeat at step 1: Error executing script. Unexpected error for call_service at pos 1: Error writing registers. Start: 41011; values: [100]; slave: 247 from /dev/ttyUSB0: Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 4 bytes (0 received)
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 4 bytes (0 received)

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 526, in _async_step
    await getattr(self, handler)()
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 764, in _async_call_service_step
    response_data = await self._async_run_long_action(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<9 lines>...
    )
    ^
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 727, in _async_run_long_action
    return await long_task
           ^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2795, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2838, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1006, in entity_service_call
    single_response = await _handle_entity_call(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^
        hass, entity, func, data, call.context
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1078, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/number/__init__.py", line 122, in async_set_value
    await entity.async_set_native_value(native_value)
  File "/config/custom_components/foxess_modbus/entities/modbus_number.py", line 118, in async_set_native_value
    await self._controller.write_register(self._address, int_value)
  File "/config/custom_components/foxess_modbus/modbus_controller.py", line 201, in write_register
    await self.write_registers(address, [value])
  File "/config/custom_components/foxess_modbus/modbus_controller.py", line 238, in write_registers
    raise ex
  File "/config/custom_components/foxess_modbus/modbus_controller.py", line 222, in write_registers
    await self._client.write_registers(start_address, values, self._slave)
  File "/config/custom_components/foxess_modbus/client/modbus_client.py", line 180, in write_registers
    raise ModbusClientFailedError(message, self, response) from response
custom_components.foxess_modbus.client.modbus_client.ModbusClientFailedError: Error writing registers. Start: 41011; values: [100]; slave: 247 from /dev/ttyUSB0: Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 4 bytes (0 received)
2025-02-04 00:30:50.552 ERROR (MainThread) [homeassistant.components.automation.octopus_intelligent_go_switch_on_with_retry_v2] Octopus Intelligent Go Switch On with Retry v2: Error executing script. Unexpected error for repeat at pos 1: Error writing registers. Start: 41011; values: [100]; slave: 247 from /dev/ttyUSB0: Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 4 bytes (0 received)
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 4 bytes (0 received)

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 526, in _async_step
    await getattr(self, handler)()
  File "/usr/src/homeassistant/homeassistant/helpers/trace.py", line 283, in async_wrapper
[... cut ...]
    await self._client.write_registers(start_address, values, self._slave)
  File "/config/custom_components/foxess_modbus/client/modbus_client.py", line 180, in write_registers
    raise ModbusClientFailedError(message, self, response) from response
custom_components.foxess_modbus.client.modbus_client.ModbusClientFailedError: Error writing registers. Start: 41011; values: [100]; slave: 247 from /dev/ttyUSB0: Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 4 bytes (0 received)

canton7, 2025-02-04:
It's a simple case of us sending the register write command to the inverter, but the inverter doesn't respond with anything. Maybe we could try retrying in that case, but I'm not sure what else we can do.

djdarlo, 2025-02-04:
For the time being I'm going to put a 30second delay in the modbus call so it doesn't clash with the FoxESS EM force charge call.  Is there anyway I can replicate the force charge call via modbus?

FozzieUK, 2025-02-04:
Now I see the problem, you are using EM and this integration together and they are not designed to co-exist, they don't know about each other so both can write together at the same time and this will happen.

To make it worse you are doing at exactly the same time as EM is making it's writes to setup the charge, and to make it even worse EM works by setting the charge and the minsoc, it then manages minsoc to achieve it's balance - if you set minsoc to something else it will throw EM and it will attempt to set it back again - it simply cannot work like this, you have to let one do it or the other - introducing a 30 second will make the write work but EM will then set it back shortly after, you can't do this in both integrations.

Probably best to start again and look at what you want to do and go from there - you can set the charge periods via foxess modbus if you want and disable all of EM's write functions which would stop this happening.

canton7, 2025-02-04:
Ah yes, I missed that you were using EM at the same time, too. 

How are you even getting that to work, given that you have a usb-to-serial adapter? Those will only let one process open them at once, so it shouldn't be possible for both foxess_modbus and EM to be using it at the same time.

canton7, 2025-02-04:
If you want to start a force charge using foxess_modbus, set the Work Mode to Force Charge.

FozzieUK, 2025-02-04:
The call to set force charge is this one (you'll have to paste in your device ID as it is unique -

```
data:
  enable_force_charge: true
  enable_charge_from_grid: true
  start: "00:30:00"
  end: "05:30:00"
  charge_period: "1"
  inverter: 84609677d0bc2a5f9fb29beba3f27565
action: foxess_modbus.update_charge_period
```

and the same again to clear it

```
data:
  enable_force_charge: false
  enable_charge_from_grid: false
  start: "00:00:00"
  end: "00:00:00"
  charge_period: "1"
  inverter: 84609677d0bc2a5f9fb29beba3f27565
action: foxess_modbus.update_charge_period
```

If in EM you set the 'disable auto charge' switch - it will perform all of it's calculations it just won't attempt to write anything.

canton7, 2025-02-04:
I don't even understand how they've got EM reading anything from the inverter, given that this is a usb-to-serial adapter! Even if that *did* work, responses to reads would get mixed up between the two integrations, and we'd get unexpected values back.

djdarlo, 2025-02-04:
I don't use EM's full functionality - that is I have the auto charge functionality disabled.  I only call the two services calls - Start force charge now and Stop force charge based on a trigger.  This has been working for a good year now ;)  I think I might try to migrate these two calls to modbus ones instead. Maybe my setting the work mode to force charge then back to self-use - would this work?

canton7, 2025-02-04:
> (you'll have to paste in your device ID as it is unique -

(Note that you can use your inverter's friendly name there as well. Including `""` if you didn't give it a friendly name).

canton7, 2025-02-04:
(Note that the "modbus" integration is [this one](https://www.home-assistant.io/integrations/modbus/), which is different to foxess_modbus).

You can set up a force charge either by using charge periods (as @FozzieUK showed you), or by setting the work mode. Both work.

But again, how is EM actually reading anything from the inverter in the first place, given that you have a dumb usb-to-serial adapter, which should only allow one process to have it open at a time?

FozzieUK, 2025-02-04:
> I don't use EM's full functionality - that is I have the auto charge functionality disabled. I only call the two services calls - Start force charge now and Stop force charge based on a trigger. [...]

Just use the calls I posted above, that replicates the service call to force charge and stop charge - you could use change work mode to force charge / self-use as well - it's a bit simpler.

>I don't even understand how they've got EM reading anything from the inverter, given that this is a usb-to-serial adapter! Even if that did work, responses to reads would get mixed up between the two integrations, and we'd get unexpected values back.

If the disable auto charge switch is on it won't read or write (apart from at start up) - but the service call to start and stop force generates the writes

canton7, 2025-02-04:
Ah, I thought it was reading things like the battery SoC

FozzieUK, 2025-02-04:
> Ah, I thought it was reading things like the battery SoC

actually yes good point, i'll check - it's been a while since I looked at it

FozzieUK, 2025-02-04:
> > Ah, I thought it was reading things like the battery SoC
> 
> actually yes good point, i'll check - it's been a while since I looked at it

ok, yep - it doesn't make those calls, it simply uses whatever integration you have installed (cloud or modbus) and it monitors the sensors battery_soc and load_power

canton7, 2025-02-04:
Anyway, I'm going to close this off as it isn't a bug in foxess_modbus. Thanks!

closed 2025-02-04 by canton7