nathanmarlor/foxess_modbus #774: H3 Pro version numbers

martinprior, 2025-02-05:
The Master, Slave, & Manager versions of an H3 Pro reported in the integration are wrong:
The numbers reported are the integer values of registers 36001 (master), 36002 (slave), and 36003 (manager)
The integer value of those registers should first be converted into a hex string, then that string should be converted back to a number, then divided by 100.

The master version currently reported for me is 3.42   (register value is 342)
The version in the About section on the Inverter display is 1.56 ( &H0156 / 100) 

Cheers for your work on this integration, btw 
Martin

canton7, 2025-02-06:
Annoyingly FoxESS keep flip-flopping between how those versions are interpreted: it varies based on inverter model and manager version, and can be different between the 3 different version registers.

What's the manager version of your inverter? Either your on a new one and FoxESS changed this recently, or you're on an old version and they've changed it since.

martinprior, 2025-02-06:
Oh, OK :)
We've got manager 1.25
It's a rather large system, two H3-Pro-25 with 2 stacks of batteries per inverter, in parallel, with a smart logger and emergency power system. To fix an issue getting everything to play nicely Fox updated the manager software to something not released yet.
Master version is 1.56, Slave version is 1.06

canton7, 2025-02-06:
OK thanks! It's starting to look like they changed it in v1.22.

MontyPud, 2025-02-10:
Hi.
Possibly I should make a new issue, but it seems so similar.  I suspect fox have rolled out a bunch of updates. The modbus link with my H1 stopped working last night. 
It is a
Model H1-3.7-E
Version_Master 1.63
Version_Slave 1.02
Version_ARM 1.70

Or it might be something totally different.

canton7, 2025-02-10:
@MontyPud Sounds like something completely different. Please open a new discussion.

canton7, 2025-02-13:
Superseded by #777

closed 2025-02-13 by canton7

martinprior, 2025-02-13:
Yes, versions all match reality now.