nathanmarlor/foxess_modbus #785: Support for "I-X10"

Trustmania, 2025-02-20:
Hello,

I want to use this nice integration. I have a I-X10, which seams to be the same as the "H3-10.0-Smart" but with a branding of the German comanpy "Enpal". 

The data sheet of the "H3-10.0 Smart": https://de.fox-ess.com/wp-content/uploads/2024/08/DE-H3-Smart-Datasheet-V1.2-8.5.pdf
The data sheet of the "I-X10":  https://www.transfernow.net/dl/20250220kLZo7hCr (The link is only 7 days available, I can reupload)

For me it looks complete the same.

The following log I can see in HomeAssistant:

```
Logger: custom_components.foxess_modbus.modbus_controller
Quelle: custom_components/foxess_modbus/modbus_controller.py:538
Integration: FoxESS - Modbus ([Dokumentation](https://github.com/nathanmarlor/foxess_modbus), [Probleme](https://github.com/nathanmarlor/foxess_modbus/issues))
Erstmals aufgetreten: 21:59:16 (2 Vorkommnisse)
Zuletzt protokolliert: 22:12:26

Did not recognise inverter model 'I-X10' ([18733, 22577, 12288, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0])
```

and

```
Logger: custom_components.foxess_modbus.modbus_controller
Quelle: custom_components/foxess_modbus/modbus_controller.py:539
Integration: FoxESS - Modbus ([Dokumentation](https://github.com/nathanmarlor/foxess_modbus), [Probleme](https://github.com/nathanmarlor/foxess_modbus/issues))
Erstmals aufgetreten: 21:59:16 (2 Vorkommnisse)
Zuletzt protokolliert: 22:12:26

Autodetect: failed to connect to (tcp://192.aaa.bbb.ccc:502)
Traceback (most recent call last):
  File "/config/custom_components/foxess_modbus/modbus_controller.py", line 539, in autodetect
    raise UnsupportedInverterError(full_model)
custom_components.foxess_modbus.common.exceptions.UnsupportedInverterError: Inverter model not supported: 'I-X10'
```

I had a short look into the code, it seams to be clear, as the "I-X10" is not added to the `inverter_profiles.py`

canton7, 2025-02-21:
That'll be in the next beta, released hopefully later today. Let me know if you have any problems!

closed 2025-02-21 by canton7

Trustmania, 2025-02-21:
Wow, awesome! Thank you so much. Will have a look and give some feedback :-)

canton7, 2025-02-21:
You're lucky -- you caught me on my day off!

fabian1512, 2025-04-26:
@Trustmania : Could you please reupload the Datasheet? I am trying to setup my H3-G2 (smart) via configuration.yaml

Thank you in advance