nathanmarlor/foxess_modbus #817: KeyError when selecting LAN for KH8 inverter (ConnectionType.LAN missing from profile)

Bramasole11, 2025-03-30:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### FoxESS - Modbus version

V1.13.4 & V1.13.5b6

### Inverter model

KH8|

### Inverter manager version

1.41

### Describe the bug

I’m using a KH8 inverter, which is listed as explicitly supported in the foxess_modbus documentation. I’m configuring the integration via HACS (UI-based), and selecting LAN as the connection type. This leads to the following crash at runtime: 

File "/config/custom_components/foxess_modbus/inverter_profiles.py", line 425, in inverter_connection_type_profile_from_config
return INVERTER_PROFILES[inverter_config[INVERTER_BASE]].connection_types[inverter_config[INVERTER_CONN]]
KeyError: <ConnectionType.LAN: 'LAN'>

It appears the KH8 inverter profile is missing the ConnectionType.LAN entry in its connection_types dictionary.

Since the KH series is listed as supported in the docs, and LAN is selectable in the UI, I believe this is a bug.

Suggested fix:

	•	Add LAN support to the KH8 profile in inverter_profiles.py

canton7, 2025-03-30:
I'm sure we had an open issue for this...

canton7, 2025-03-30:
> Since the KH series is listed as supported in the docs, and LAN is selectable in the UI, I believe this is a bug.

The problem there is that some inverters expose different registers over different connection types, so we need to know the connection type *before* we can query the inverter for its model number. It's therefore possible for the user to choose the wrong connection type, and most of the time we'll still manage to query the inverter for its model, and only *then* realise that that inverter model shouldn't support the connection type.

canton7, 2025-03-30:
v1.13.5b7 should fix that.

closed 2025-03-30 by canton7

Bramasole11, 2025-03-30:
Hi Nathan,Thanks for your response Nathan.First of all, I’m very impressed with what you’ve done with the ModBus integration, it is of immense importance to FoxESS system owners, since as you well know, the cloud app is virtually unusable in HA automations as well as lacking control of things like Work Mode.Yes, I did see that, however the specific point I wanted to make is not for you to fix the code to accept direct Ethernet connections, but to simply remove the option from the integration list because, again as you well know if you select this option you get a fatal runtime error, however if you select option 3) ModBus to TCP it works fine.Best regardsKeith Harvey

Bramasole11, 2025-03-30:
Hi Antony,My apologies for getting your name wrong. Plus thanks for the version feedback, that’s brilliant.Best regardsKeith Harvey

Closed #817 as completed via 9d2b156.

canton7, 2025-03-30:
As I noted, I can't remove the Direct Ethernet Connection option from the setup. This is because we need to know how you're connecting to the inverter *before* we try to query it, and we only know what model it is *after* we query it.

So if I removed that option, anyone who was actually using a direct ethernet connection woudn't be able to select it, which menas we wouldn't be able to query their inverter to find out that it's one which supports a direct ethernet connection.

Bramasole11, 2025-03-30:
Hi Antony,Thanks for that. It is never as simple as it appears and your the guru.Best regardsKeith Harvey

So if I removed that option, anyone who was actually using a direct ethernet connection woudn't be able to select it, which means we wouldn't be able to query their inverter to find out that it's one which supports a direct ethernet connection.

PR #819 (2025-03-30): Support ConnectionType.LAN implicitly for all inverter models
It looks like we're not really aware of which inverters support LAN and which don't, and there's nothing to stop a user from selecting the wrong one during setup anyway.

If we're looking for a connection type of LAN on a profile, and we don't find it, then look for AUX as a backup.

Fixes: #817
commit 9d2b1566 (2025-03-30): Support ConnectionType.LAN implicitly for all inverter models

It looks like we're not really aware of which inverters support LAN and
which don't, and there's nothing to stop a user from selecting the wrong
one during setup anyway.

If we're looking for a connection type of LAN on a profile, and we don't
find it, then look for AUX as a backup.

Fixes: #817