nathanmarlor/foxess_modbus #818: Sonnenkraft HWR 15 - invalid Registers

Flirina, 2025-03-30:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### FoxESS - Modbus version

1.13.5b6

### Inverter model

SK HWR 15

### Inverter manager version

Master 1.50; Slave 1.06, Komm 1.33

### Describe the bug

Hi, i set up the modbus connection to the Sonnenkraft HWR 15 with that integration using an ELFIN EW11 and was also able to connect to the HWR15.
Nevertheless I get an warning in HA that I have invalid registers.

Details: [(31000, 52)], [(32000, 24)], [(41000, 1)], [(41007, 5)].

I tried to exclude the registers in the inverter_profile.py but either i get no change in the error or I get something like "foxess address overlaps an invalid range".

the only entities I get back it is:

Version: Manager
31.35

Version: Master
131.04

Version: Slave
131.30

all other entities are unknown.
Nevertheless I can change the working mode of the inverter via the integration (from e.G: self use -> backup), and its also reflected in the pv-hub 2.0 app.

can anyone guide me in the right direction.
i assume cabling is correct since I can at least change sth..

canton7, 2025-03-30:
Interesting. The inverter model at address 30000 must be working as well, or you'd have never made it through the setup. The inverter is rejecting reads or pretty much all of the other registers, though.

Could you post a full debug log, see if that has any clues?

@krema445, what manager version is your SK-HWR on?

Flirina, 2025-03-30:
DEBUG LOG:

2025-03-30 09:26:28.640 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-03-30 09:26:28.641 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration foxess_modbus which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-03-30 09:26:40.737 WARNING (MainThread) [custom_components.foxess_modbus.modbus_controller] tcp://192.168.6.56:502 247: register 31000 is invalid
2025-03-30 09:26:40.806 WARNING (MainThread) [custom_components.foxess_modbus.modbus_controller] tcp://192.168.6.56:502 247: register 31001 is invalid
2025-03-30 09:26:40.875 WARNING (MainThread) [custom_components.foxess_modbus.modbus_controller] tcp://192.168.6.56:502 247: register 31002 is invalid
2025-03-30 09:26:40.963 WARNING (MainThread) [custom_components.foxess_modbus.modbus_controller] tcp://192.168.6.56:502 247: register 31003 is invalid
2025-03-30 09:26:41.052 WARNING (MainThread) [custom_components.foxess_modbus.modbus_controller] tcp://192.168.6.56:502 247: register 31004 is invalid
2025-03-30 09:26:41.122 WARNING (MainThread) [custom_components.foxess_modbus.modbus_controller] tcp://192.168.6.56:502 247: register 31005 is invalid
2025-03-30 09:26:41.191 WARNING (MainThread) [custom_components.foxess_modbus.modbus_controller] tcp://192.168.6.56:502 247: register 31006 is invalid
2025-03-30 09:26:41.260 WARNING (MainThread) [custom_components.foxess_modbus.modbus_controller] tcp://192.168.6.56:502 247: register 31007 is invalid
2025-03-30 09:26:41.330 WARNING (MainThread) [custom_components.foxess_modbus.modbus_controller] tcp://192.168.6.56:502 247: register 31008 is invalid
2025-03-30 09:26:41.399 WARNING (MainThread) [custom_components.foxess_modbus.modbus_controller] tcp://192.168.6.56:502 247: register 31009 is invalid
2025-03-30 09:26:41.468 WARNING (MainThread) [custom_components.foxess_modbus.modbus_controller] tcp://192.168.6.56:502 247: register 31010 is invalid
2025-03-30 09:26:41.536 WARNING (MainThread) [custom_components.foxess_modbus.modbus_controller] tcp://192.168.6.56:502 247: register 31011 is invalid
2025-03-30 09:26:41.606 WARNING (MainThread) [custom_components.foxess_modbus.modbus_controller] tcp://192.168.6.56:502 247: register 31012 is invalid
2025-03-30 09:26:41.675 WARNING (MainThread) [custom_components.foxess_modbus.modbus_controller] tcp://192.168.6.56:502 247: register 31013 is invalid
2025-03-30 09:26:41.743 WARNING (MainThread) [custom_components.foxess_modbus.modbus_controller] tcp://192.168.6.56:502 247: register 31014 is invalid
2025-03-30 09:26:41.812 WARNING (MainThread) [custom_components.foxess_modbus.modbus_controller] tcp://192.168.6.56:502 247: register 31015 is invalid
2025-03-30 09:26:41.882 WARNING (MainThread) [custom_components.foxess_modbus.modbus_controller] tcp://192.168.6.56:502 247: register 31016 is invalid
2025-03-30 09:26:41.951 WARNING (MainThread) [custom_components.foxess_modbus.modbus_controller] tcp://192.168.6.56:502 247: register 31017 is invalid
2025-03-30 09:26:42.020 WARNING (MainThread) [custom_components.foxess_modbus.modbus_controller] tcp://192.168.6.56:502 247: register 31018 is invalid
2025-03-30 09:26:42.089 WARNING (MainThread) [custom_components.foxess_modbus.modbus_controller] tcp://192.168.6.56:502 247: register 31019 is invalid
2025-03-30 09:26:42.158 WARNING (MainThread) [custom_components.foxess_modbus.modbus_controller] tcp://192.168.6.56:502 247: register 31020 is invalid
2025-03-30 09:26:42.227 WARNING (MainThread) [custom_components.foxess_modbus.modbus_controller] tcp://192.168.6.56:502 247: register 31021 is invalid
2025-03-30 09:26:42.296 WARNING (MainThread) [custom_components.foxess_modbus.modbus_controller] tcp://192.168.6.56:502 247: register 31022 is invalid
2025-03-30 09:26:42.364 WARNING (MainThread) [custom_components.foxess_modbus.modbus_controller] tcp://192.168.6.56:502 247: register 31023 is invalid
2025-03-30 09:26:42.434 WARNING (MainThread) [custom_components.foxess_modbus.modbus_controller] tcp://192.168.6.56:502 247: register 31024 is invalid
2025-03-30 09:26:42.502 WARNING (MainThread) [custom_components.foxess_modbus.modbus_controller] tcp://192.168.6.56:502 247: register 31025 is invalid
2025-03-30 09:26:42.570 WARNING (MainThread) [custom_components.foxess_modbus.modbus_controller] tcp://192.168.6.56:502 247: register 31026 is invalid
2025-03-30 09:26:42.639 WARNING (MainThread) [custom_components.foxess_modbus.modbus_controller] tcp://192.168.6.56:502 247: register 31027 is invalid
2025-03-30 09:26:42.709 WARNING (MainThread) [custom_components.foxess_modbus.modbus_controller] tcp://192.168.6.56:502 247: register 31028 is invalid
2025-03-30 09:26:42.778 WARNING (MainThread) [custom_components.foxess_modbus.modbus_controller] tcp://192.168.6.56:502 247: register 31029 is invalid
2025-03-30 09:26:42.846 WARNING (MainThread) [custom_components.foxess_modbus.modbus_controller] tcp://192.168.6.56:502 247: register 31030 is invalid
2025-03-30 09:26:42.916 WARNING (MainThread) [custom_components.foxess_modbus.modbus_controller] tcp://192.168.6.56:502 247: register 31031 is invalid
[... cut ...] 31042, 31044, 31045, 31047, 31048, 31049, 31050, 31051, 41011, 41010}
2025-03-30 15:05:52.577 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Refresh of tcp://192.168.6.56:502 247 complete - notifying sensors: set()
2025-03-30 15:06:02.579 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Refresh of tcp://192.168.6.56:502 247 complete - notifying sensors: set()

i have the feeling that the RX cable might be the issue... I will change the cable today...

canton7, 2025-03-30:
Thanks. I'll go through that properly later.

I don't think it is a wiring problem. We're talking to the inverter just fine: it's explicitly rejecting our attempts to read most registers

canton7, 2025-03-30:
The debug log looks as I'd expect given what we know: the reads are failing with IllegalAddress, except the ones you notes.

@williamjeccles I don't suppose you know whether 1.33 is a reasonably up-to-date manager version for this inverter?

closed 2025-08-15 by Flirina