nathanmarlor/foxess_modbus #822: Strange end of day issue with home assistant energy tab

LeighAS, 2025-04-07:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### FoxESS - Modbus version

Master 1.54 slave 1.02

### Inverter model

6kW H1 G1

### Inverter manager version

1.57

### Describe the bug

[image]

This is what is happening as the sun goes down. I have yet to commission the battery storage part of my setup. So when the PV contactor's are opening back up the data being logged in the energy tab is getting some huge 10x approximately bigger value than reality which is shown in the normal dash board below. As you can see this knackers the, (really clear up till then), graph of the days production.

[image]

Anyway thanks for any help that could given here.

Using the elfin ew11a WiFi modbus device to collect data.

WyndStryke, 2025-04-09:
There was an issue with the 3-phase inverter returning 10x the figures:

https://github.com/nathanmarlor/foxess_modbus/issues/724

So I wonder if the same code is being added into the single phase inverters too.

LeighAS, 2025-04-11:
I have now set the time and date. It was 8 hours ahead which would tally with when the sundown was happening too. Will keep using the template but hopefully this will sort the last gotcha.

WyndStryke, 2025-04-11:
In theory the inverter time should sync automatically rather than needing manual intervention, and from the logs it looks like it has been doing it successfully on mine.

Which timezone are you in, does the 8 hour offset correspond to that?

I do have a crude automation which will correct small time differences, I needed for a while in January because the clock syncs weren't happening at that time.   It could do with being re-written, given that the variable scoping rules changed recently, which would simplify it a lot.

`inverter device ID` Needs to be replaced with the inverter's device ID from the integration.

```
input_text.foxess_rtc_adjustments

alias: rtc_update_and_log
description: Write date & time from inverter into HA log
triggers:
  - trigger: time
    at: "23:30:30"
  - trigger: time
    at: "11:30:30"
conditions: []
actions:
  - variables:
      RTCRegister: null
  - action: foxess_modbus.read_registers
    data:
      start_address: 40000
      type: input
      inverter: **inverter device ID**
      count: 6
    response_variable: RTCRegister
  - choose:
      - conditions:
          - condition: template
            value_template: >-
              {% set year = RTCRegister['values'].40000| int % 100 %} {% set
              month = RTCRegister['values'].40001| int %} {% set day =
              RTCRegister['values'].40002| int %} {% set hour =
              RTCRegister['values'].40003| int %} {% set minute =
              RTCRegister['values'].40004| int %} {% set second =
              RTCRegister['values'].40005| int %} {% set new_year =
              now().strftime('%y') | int   %} {% set new_month =
              now().strftime('%m') | int  %} {% set new_day =
              now().strftime('%d') | int  %} {% set new_hour =
              now().strftime('%H') | int  %} {% set new_minute =
              now().strftime('%M') | int  %} {% set new_second =
              now().strftime('%S') | int  %}   {% set safe_to_update = ( year ==
              new_year and month == new_month and hour == new_hour and
              (((minute-new_minute)|abs) < 5)) %} {% set need_to_update = not
              (year ==  new_year and month == new_month and hour == new_hour and
              minute == new_minute and ((second == new_second) or (((second -
              new_second) | abs) < 2))) %} {% set time_valid = not ((new_year <
              25) or (new_month < 1)  or (new_month > 12) or (new_day < 1) or
              (new_day > 31) or (new_hour < 0) or (new_hour > 23) or (new_minute
              < 0) or (new_minute > 59) or (new_second < 0)  or (new_second >
              59)) %} {{ need_to_update and safe_to_update and time_valid }}
        sequence:
          - action: foxess_modbus.write_registers
            metadata: {}
            data:
              start_address: 40000
              values: >-
                {{ now().strftime('%y')|int}},{{ now().strftime('%m')|int}},{{
                now().strftime('%d')|int}},{{ now().strftime('%H')|int}},{{
                now().strftime('%M')|int}},{{ now().strftime('%S')|int}}
              inverter: **inverter device ID**
          - action: input_text.set_value
            metadata: {}
            data:
              value: >-
                {% set val = RTCRegister['error'] %} {% if 'Error reading
                registers' not in val %}
[... 58 lines cut ...]
            {{new_year}}/{{new_month}}/{{new_day}}
            {{new_hour}}:{{new_minute}}:{{new_second}}
              {%- if year == new_year and month == new_month and hour == new_hour and minute == new_minute and ((second == new_second) or (((second - new_second) | abs) < 2)) %}
            Time diff < 2 seconds, do not update.
              {%- else %}
            Inverter RTC time does not match HA time and needs to be updated.
              {%- endif %}
              {%- if year == new_year and month == new_month and hour == new_hour and (((minute-new_minute)|abs) < 5) %}
            Safe to update.
              {%- else %}
            Unsafe to update (time diff > 5 min).
              {%- endif %}
            {%- else %} No RTC Time, received {{ RTCRegister['error'] }} {%-
            endif %}
        target:
          entity_id: input_text.foxess_rtc_adjustments
      - action: logbook.log
        metadata: {}
        data:
          name: automation.rtc_update_and_log
          message: "{{states('input_text.foxess_rtc_adjustments')}}"
          entity_id: automation.rtc_update_and_log
mode: single

```

LeighAS, 2025-04-12:
I changed the time yesterday and still had a weird issue but today seems OK so far but haven't hit PV off time yet so we shall see.

closed 2025-08-13 by canton7