nathanmarlor/foxess_modbus #832: H3 Battery_Soc and Battery_Temperature "unknown"

Gridomin, 2025-04-24:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### FoxESS - Modbus version

1.13.4

### Inverter model

H3

### Inverter manager version

1.85

### Describe the bug

Modbus via Waveshare USB to RS485 (B).
Inverter H3 Version 1.85
nearly all entities are showing and working except Battery_SOC and Battery_temperature which are showing state "unknown".

marcinzoladek, 2025-04-29:
I have the same issue. Seems to be a generic problem for all H3 inventers

FozzieUK, 2025-04-29:
Juts as a quick test try reducing your max_reads to 40 - you do that in Settings, Devices & Services, click '>' next to Foxess - Modbus, then 'CONFIGURE',  Advanced Settings, and set Max Reads to 40

Gridomin, 2025-04-29:
Reduced Max Reads to 40. 
Reloaded integration and restartet HA.
No effect, still "unknown".

rejnder, 2025-04-29:
> Reduced Max Reads to 40. Reloaded integration and restartet HA. No effect, still "unknown".

Done the same - but still "unknown" value

FozzieUK, 2025-04-29:
What errors do you see in the HA logs that relate to Foxess - Modbus ?

Gridomin, 2025-04-29:
In logs of Home Assistant Core and Supervisor there are no warnings or errors relating to Foxess.
I have startet the debuglog of FoxEss-Modbus, reloaded the integration and stopped the debug protocol. This shows up:

> IndexError: list index out of range
2025-04-29 10:53:39.636 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses on /dev/ttyACM0 247: (31000, 39)
2025-04-29 10:53:39.780 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses on /dev/ttyACM0 247: (31041, 11)
2025-04-29 10:53:39.865 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses on /dev/ttyACM0 247: (31090, 34)
2025-04-29 10:53:40.001 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses on /dev/ttyACM0 247: (32000, 24)
2025-04-29 10:53:40.113 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses on /dev/ttyACM0 247: (41000, 1)
2025-04-29 10:53:40.182 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses on /dev/ttyACM0 247: (41007, 1)
2025-04-29 10:53:40.249 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses on /dev/ttyACM0 247: (41008, 1)
2025-04-29 10:53:40.318 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses on /dev/ttyACM0 247: (41009, 1)
2025-04-29 10:53:40.385 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses on /dev/ttyACM0 247: (41010, 1)
2025-04-29 10:53:40.452 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses on /dev/ttyACM0 247: (41011, 1)
2025-04-29 10:53:40.520 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Refresh of /dev/ttyACM0 247 complete - notifying sensors: {32000, 32001, 32002, 32003, 32004, 32005, 32006, 32007, 32008, 32009, 32010, 32011, 32012, 32013, 32014, 32015, 32016, 32017, 32018, 31123, 32019, 32020, 32021, 32022, 31000, 31001, 31002, 31003, 31004, 31005, 32023, 31009, 31010, 31011, 31012, 31013, 31014, 41000, 31022, 41007, 41008, 41009, 31026, 31027, 31028, 31029, 31030, 31031, 31032, 31033, 31034, 31035, 31036, 31037, 31038, 31041, 31042, 31044, 31045, 31047, 31048, 31049, 31050, 31051, 41011, 41010, 31090, 31100, 31101, 31102, 31103}
2025-04-29 10:53:49.637 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses on /dev/ttyACM0 247: (31000, 39)
2025-04-29 10:53:49.780 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses on /dev/ttyACM0 247: (31041, 11)
2025-04-29 10:53:49.870 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses on /dev/ttyACM0 247: (31090, 34)
2025-04-29 10:53:50.004 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses on /dev/ttyACM0 247: (32000, 24)
2025-04-29 10:53:50.115 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses on /dev/ttyACM0 247: (41000, 1)
2025-04-29 10:53:50.183 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses on /dev/ttyACM0 247: (41007, 1)
2025-04-29 10:53:50.250 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses on /dev/ttyACM0 247: (41008, 1)
2025-04-29 10:53:50.317 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses on /dev/ttyACM0 247: (41009, 1)
2025-04-29 10:53:50.385 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses on /dev/ttyACM0 247: (41010, 1)
2025-04-29 10:53:50.453 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses on /dev/ttyACM0 247: (41011, 1)
2025-04-29 10:53:50.521 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Refresh of /dev/ttyACM0 247 complete - notifying sensors: {32000, 32001, 32002, 32003, 32004, 32005, 32006, 32007, 32008, 32009, 32010, 32011, 32012, 32013, 32014, 32015, 32016, 32017, 32018, 31123, 32019, 32020, 32021, 32022, 31000, 31001, 31002, 31003, 31004, 31005, 32023, 31009, 31010, 31011, 31012, 31013, 31014, 41000, 31022, 41007, 41008, 41009, 31026, 31027, 31028, 31029, 31030, 31031, 31032, 31033, 31034, 31035, 31036, 31037, 31038, 31041, 31042, 31044, 31045, 31047, 31048, 31049, 31050, 31051, 41011, 41010, 31090, 31100, 31101, 31102, 31103}
2025-04-29 10:53:59.638 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses on /dev/ttyACM0 247: (31000, 39)
2025-04-29 10:53:59.783 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses on /dev/ttyACM0 247: (31041, 11)
2025-04-29 10:53:59.873 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses on /dev/ttyACM0 247: (31090, 34)
2025-04-29 10:54:00.012 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses on /dev/ttyACM0 247: (32000, 24)
2025-04-29 10:54:00.128 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses on /dev/ttyACM0 247: (41000, 1)
2025-04-29 10:54:00.197 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses on /dev/ttyACM0 247: (41007, 1)
2025-04-29 10:54:00.263 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses on /dev/ttyACM0 247: (41008, 1)
[... cut ...]  31012, 31013, 31014, 41000, 31022, 41007, 41008, 41009, 31026, 31027, 31028, 31029, 31030, 31031, 31032, 31033, 31034, 31035, 31036, 31037, 31038, 31041, 31042, 31044, 31045, 31047, 31048, 31049, 31050, 31051, 41011, 41010, 31090, 31100, 31101, 31102, 31103}
2025-04-29 10:54:07.975 DEBUG (MainThread) [custom_components.foxess_modbus.client.modbus_client] Closing connection to modbus on /dev/ttyACM0

Not sure if the first row (IndexError) is of FoxEss or the module before that (Xiaomi Cloud Map)

Gridomin, 2025-04-29:
it was enabled. Did as u said: Disabled, waited 20 sec, enabled again and waited 30 sec. All values updated.
Battery SOC still unknown.

Also unknown are: Battery SOH, Battery Temp, BMS Cell mV (High+Low), BMS Cell Temp (High+Low), BMS kWh Remaining.

Gridomin, 2025-04-29:
I am sorry, but that didn't work either.
Selected the pre_180 version. Disabled battery coc, waited and reactivated - waited. Reloaded the integration, restartet HA. Still "unknown".

Gridomin, 2025-05-11:
hey @mschierh. No solution at this time, sorry.
maybe @FozzieUK is checking and finds a solution on it. hope so :)

mhabio, 2025-06-08:
Hi

Same issue here on a new april-2025 installation:
Battery_SOC, Battery_temperature, both unknown. setting max_reads to 20 and 10 gives no change, no error in the standard HA logs

[image]

EW11 wifi/modbus adaptor integration - worked out of the box following instruction here.

WyndStryke, 2025-06-08:
> Testing the registers, they are read, SoC reads fine, but the temperature is off (19 and not 26C)

There are a number of 'battery' temperatures, which one are you looking at?

* bms_cell_temp_high
* bms_cell_temp_low
* battery temperature (not actually the battery temperature, it's the BMS temperature)

Right now, on my system, they are 25.1c, 20.4c, and 28c, respectively.  The app shows bms_cell_temp_low.

Baratux, 2025-06-09:
Until that’s fixed, drop the following **trigger-template sensor** into `configuration.yaml` to replace it:

```yaml
# configuration.yaml ───────────────────────────────────────────────
template:
  - trigger:
      - platform: time_pattern
        minutes: "/1"

    action:
      - action: foxess_modbus.read_registers
        data:
          start_address: 31038
          count: 1
          type: holding
          inverter: abcdefghijklmnop123456 # replace
        response_variable: modbus

    sensor:
      - name: "FoxESS Battery SoC"
        unique_id: foxess_battery_soc
        device_class: battery
        unit_of_measurement: "%"
        state_class: measurement
        state: >-
          {% set values = modbus.get('values', {}) if modbus is defined else {} %}
          {% set soc    = values.get(31038) if 31038 in values else values.get('31038') %}
          {{ soc if soc is not none else this.state }}
        availability: >-
          {{ modbus is defined and modbus.get('values') is mapping }}

mhabio, 2025-06-10:
> There are a number of 'battery' temperatures, which one are you looking at?
>
> * bms_cell_temp_high
> * bms_cell_temp_low
> * battery temperature (not actually the battery temperature, it's the BMS [...]

Not sure, will implement the work-around from [Baratux](https://github.com/Baratux) and compare.

kharas80, 2025-06-16:
Have the same issue.
Workaround is also not working for me. Just for doublecheck: Where can i find my inverter name?

mhabio, 2025-08-03:
> Not sure, will implement the work-around from [Baratux](https://github.com/Baratux) and compare.

Tested workaround and works fine. With the new update a number of sensors are still unknown. 

[image]

devpwnz, 2025-08-07:
Did anyone manage to identify the problem? I'm understanding that the Battery SoC is read correctly from the 31038 address but somewhere along the way it's not written correctly in the entity?

FozzieUK, 2025-08-07:
> Did anyone manage to identify the problem? I'm understanding that the Battery SoC is read correctly from the 31038 address but somewhere along the way it's not written correctly in the entity?

Yes that’s it, i’ve checked some of the more obvious things and jumped over the code but I can’t see why it isn’t working, the address and definition is correct and it works on the other inverters apart from the H3.

It really needs someone with an H3 to enable debugging on the integration and see what logs are generated

devpwnz, 2025-08-07:
I have a H3-12.0-E and I'd be willing to try to get to the bottom of this, but I'll need some guidance on how to help.

b10c77, 2025-08-12:
Facing the same issue with a P3-6.0-E (same as the H3). Only Battery and BMS values are unknown. Elfin EW11 as well. 

Reading register 31038 (Battery SoC) returns the value I see on the foxESS app.

canton7, 2025-08-13:
Can you read register 31042 and tell me what value it has?

Is it just battery_soc and battery_temperature, or all of the BMS-related sensors?

canton7, 2025-08-13:
Give v1.14.0b2 a try, and re-open if that doesn't fix it. Thanks!

devpwnz, 2025-08-13:
Not only did it fix the Battery SoC / SoH, but it seems to me that I also have several new items regarding the BMS.