nathanmarlor/foxess_modbus #841: Support for H3-12.0-Smart

fabian1512, 2025-05-09:
Hi,
I would also like to use this integration. I have an inverter with the above designation:

[image]

As has been discussed #785 and #807, my inverter is probably the same as the ones from Enpal and 1komma5. I have already integrated various registers manually from the [entity_descriptions.py](https://github.com/nathanmarlor/foxess_modbus/blob/main/custom_components/foxess_modbus/entities/entity_descriptions.py) for H3_smart or and _pro via the modbus integration. The data I get back is also correct.

But apparently my h3-smart is not recognised as such:

Geändert zu Error reading registers. Type: RegisterType.HOLDING; start: 31000; count: 52; slave: 247 from tcp://192.168.1.xyz:502: Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 8 bytes (0 received)

as the register 31000 seems not to exist for the H3_smart

Can you please help me to get the integration to work?

fabian1512, 2025-05-16:
helped myself. Version in forked repo is working
https://github.com/fabian1512/foxess_modbus

fhoeborn, 2025-05-16:
Wouldn't it make sense to open a Pull Request with your changes?

fhoeborn, 2025-05-17:
I tried it out (H3 Smart 12 as well) and it seems to work without a problem - how did you find out what registers are the correct one? Is there a documentation somewhere?

fabian1512, 2025-05-17:
that's what I thought at first - but then I realised that my H3-smart is not recognised as such because the registers are assigned incorrectly.
as I said, I made several attempts with copilot until I saw in the log that it was recognised as a different H3.
apparently the recognition only runs via the "H3" and not via the full name "H3-smart".
have a look at my branch https://github.com/fabian1512/foxess_modbus/tree/fabian1512-working-version
and copy these files into your corresponding directories undercustom_components/foxess_modbus:
modbus_controller.py
inverter_profiles.py
entity_descriptions.py
remote_control_description.py
types.py
I'm curious to see if the data is displayed again after a restart.

FozzieUK, 2025-05-17:
Ok thanks, that makes more sense - it suggests the inverter model is different than we are expecting (which is why it fails) - much of your re-work would not to be needed in the latest release (it appears to have used an older release as it's base) and it didn't support the H3_SMART which is why you've had to make so many changes in entities etc..

I think once we confirm the inverter model, correct the order it is processed within inverter_profiles and detect it correctly to be an H3_SMART it should work as the latest release has the updated entity register maps for H3_SMART

Can you please read the model number registers back and post it here :

use the Developer Tools, Actions, Foxess - Modbus Read Registers enter the Start Address as 30000 and count 16, holding and if you could post an image of what it returns please.

fabian1512, 2025-05-17:
here you go:
values:
  "30000": 18483
  "30001": 11569
  "30002": 12846
  "30003": 12333
  "30004": 21357
  "30005": 24946
  "30006": 29696
  "30007": 0
  "30008": 0
  "30009": 0
  "30010": 0
  "30011": 0
  "30012": 0
  "30013": 0
  "30014": 0
  "30015": 0

FozzieUK, 2025-05-17:
ok, thanks - that seems to imply the model type is 'H3-12.0-Smt' (note lower case on 'mt')

Can you have a look in your V2 app, under Device, Inverter and click the serial number under the image of the inverter - what do you see the 'Model' as ?

fabian1512, 2025-05-17:
Thank you for your efforts in advance

[image]

FozzieUK, 2025-05-17:
Thanks for confirming, actually it was my mistake the Smt was wrong (I didn't include the 24946 line which translates as ascii 61,72 or a, r) 

It is actually sending `H3-12.0-Smart` (the 'mart' is lower case), as soon as I get chance i'll review the code for this.

fabian1512, 2025-05-17:
sounds great, no hurry - for now my fork is working for me.

one more remark: you can get the H3-Smart in these versions:
H3-5.0-Smart
H3-6.0-Smart
H3-8.0-Smart
H3-9.9-Smart
H3-10.0-Smart
H3-12.0-Smart
H3-15.0-Smart

Turbo87, 2025-05-18:
I've opened a PR to fix this: https://github.com/nathanmarlor/foxess_modbus/pull/847 :)

closed 2025-05-26 by fabian1512

Nour-Said1976, 2025-08-10:
@fabian1512 
[image] 
The Modbus connection currently only shows 9 entities – that seems a bit low to me.
Could this be a bug?
Sorry, I'm not that well-versed on this topic.

fabian1512, 2025-08-10:
yes, that is because of the faults in the source-code and my source is outdated I assume

so delete the folder in config/custom_components/foxess_modbus and just download the source from here via hacs and edit the two files on your HA like mentioned above.

fabian1512, 2025-08-10:
okay, seems, you don't have hacs.
I set the repo up on my fabian1512-repo and correct the two files.
just [download](https://github.com/fabian1512/foxess_modbus/archive/refs/heads/main.zip) an put the content of custom_components to your custom_components-folder on HA, restart, set up and enjoy

Nour-Said1976, 2025-08-10:
@fabian1512 Thank you so much. It worked :-) I have a second problem connecting a Siemens wallbox. Do you have any experience with this?

fabian1512, 2025-08-10:
no, mine is a Schneider pro AC - but anyway, I can try to help. But not here. Are you in any German forum? 
e.g https://community-smarthome.com - you can find me with the same nick-name

Nour-Said1976, 2025-08-10:
Thanks for the tip!
I registered there and am logged in under the same name. Unfortunately, I can't contact you directly :-(
I'll look around the forum, maybe I'll find something about the Siemens Wallbox and dynamic electricity tariff. Best regards

skwal8r, 2025-11-20:
@fabian1512 i have the H3-15 SMART and trying to connect via the integrated WiFi connection. 

Error reading registers. Type: RegisterType.HOLDING; start: 31000; count: 91; slave: 247 from tcp://IP_ADDRESS:502: Modbus Error: [Input/Output]

Should this issue have been fixed in the current version

fabian1512, 2025-11-22:
> [@fabian1512](https://github.com/fabian1512) i have the H3-15 SMART and trying to connect via the integrated WiFi connection.
>
> Error reading registers. Type: RegisterType.HOLDING; start: 31000; [...]

@skwal8r 
What makes you so sure that the mistake lies in the integration?