nathanmarlor/foxess_modbus #842: Modbus Error: [Input/Output] [Errno 104] Connection reset by peer

The-Web-Designer, 2025-05-12:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### FoxESS - Modbus version

1.13.4

### Inverter model

H3-Pro

### Inverter manager version

1.31

### Describe the bug

Hello, I have an Elfin EE11A, i.e. the LAN version of the Elfin, because I thought it would be more stable, but now I get the following error message every day:

`changed to Error reading registers. Type: RegisterType.HOLDING; start: 37002; count: 1; slave: 247 from tcp://IP:502: Modbus Error: [Input/Output] [Errno 104] Connection reset by peer`.

And of course all "displays from HA" are gone, i.e. battery level, PV power etc. A restart of the HA does not help, a reconfiguration of the FoxEss Modbus does not help either, I have to log into the Elfin and restart it via the web interface.

I have already tried a second Elfin, which has exactly the same problem, I have also changed the power supply, to power supply from the inverter, to external power supply - I have also tried different power supplies, it doesn't matter, the error occurs about 1-2 times a day.

It is always the same error.

WyndStryke, 2025-05-14:
Perhaps try this, I saw it suggested on a different thread:

> In HA go to Settings, Devices & Services, click on the '>' to the right of Foxess - Modbus, click CONFIGURE, Advanced Settings and reduce the Max Read value to 40 - submit and it will be actioned.

It reduces the amount of work that the adaptor is having to do during each request.

The-Web-Designer, 2025-05-14:
Yes, that was my first thought too, so I reduced the value to 10. Unfortunately, that doesn't help.