nathanmarlor/foxess_modbus #844: H3-Pro no ModBus response since firmware upgrade to Master 1.62 and ARM 1.34

qstadlmeier, 2025-05-13:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### FoxESS - Modbus version

1.135b7

### Inverter model

H3-Pro-20

### Inverter manager version

1.62

### Describe the bug

ModBus communication stopped with new Master 1.62 / Arm 1.34 - Previous 1.58 / 1.33 was running fine.

2025-05-13 13:34:52.962 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Modbus error when polling tcp://192.168.178.143:502 247: Modbus Error: [Input/Output] No Response received from the remote slave/Unable to decode response
2025-05-13 13:35:08.829 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses on tcp://192.168.178.143:502 247: (36001, 3)
2025-05-13 13:35:08.830 DEBUG (SyncWorker_6) [custom_components.foxess_modbus.client.custom_modbus_tcp_client] Connecting to ModbusBaseSyncClient._params(retries=3, retry_on_empty=True, broadcast_enable=False, reconnect_delay=None, source_address=None)
2025-05-13 13:35:22.941 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Modbus error when polling tcp://192.168.178.143:502 247: Modbus Error: [Input/Output] No Response received from the remote slave/Unable to decode response
2025-05-13 13:35:22.942 WARNING (MainThread) [custom_components.foxess_modbus.modbus_controller] tcp://192.168.178.143:502 247 - 5 failed poll attempts: now not connected. Last error: Error reading registers. Type: RegisterType.HOLDING; start: 36001; count: 3; slave: 247 from tcp://192.168.178.143:502: Modbus Error: [Input/Output] No Response received from the remote slave/Unable to decode response

Additional log from ModBusProxy:
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
Preparing to run modbus-proxy
Upstream: 192.168.178.176:502
Listen: 502
Timeout: 10
Connection Time: 0.1
Loglevel: DEBUG
Generated Config
devices:
  - modbus:
      url: 192.168.178.176:502 # device url (mandatory)
      timeout: 10 # communication timeout (s) (optional, default: 10)
      connection_time: 0.1 # delay after connection (s) (optional, default: 0)
    listen:
      bind: 0:502 # listening address (mandatory)
logging:
  version: 1
  formatters:
    standard:
      format: "%(asctime)s %(levelname)8s %(name)s: %(message)s"
  handlers:
    console:
      class: logging.StreamHandler
      formatter: standard
  root:
    handlers: ['console']
    level: DEBUG
2025-05-13 13:45:57,416     INFO modbus-proxy: Starting...
2025-05-13 13:45:57,432     INFO modbus-proxy.ModBus(192.168.178.176:502): Ready to accept requests on 0:502
2025-05-13 13:46:08,859     INFO modbus-proxy.Client(192.168.178.143:60568): new client connection
2025-05-13 13:46:08,861    DEBUG modbus-proxy.Client(192.168.178.143:60568): received b'\x00\x1a\x00\x00\x00\x06\xf7\x03\x8c\xa1\x00\x03'
2025-05-13 13:46:08,862     INFO modbus-proxy.ModBus(192.168.178.176:502): connecting to modbus...
2025-05-13 13:46:08,869     INFO modbus-proxy.ModBus(192.168.178.176:502): connected!
2025-05-13 13:46:08,870     INFO modbus-proxy.ModBus(192.168.178.176:502): delay after connect: 0.1
2025-05-13 13:46:08,971    DEBUG modbus-proxy.ModBus(192.168.178.176:502): sending b'\x00\x1a\x00\x00\x00\x06\xf7\x03\x8c\xa1\x00\x03'
2025-05-13 13:46:18,975    ERROR modbus-proxy.ModBus(192.168.178.176:502): write_read error [1/2]: TimeoutError()
2025-05-13 13:46:18,976     INFO modbus-proxy.ModBus(192.168.178.176:502): closing connection...
2025-05-13 13:46:18,978     INFO modbus-proxy.ModBus(192.168.178.176:502): connection closed
2025-05-13 13:46:18,979     INFO modbus-proxy.ModBus(192.168.178.176:502): connecting to modbus...
2025-05-13 13:46:18,983     INFO modbus-proxy.ModBus(192.168.178.176:502): connected!
2025-05-13 13:46:18,984     INFO modbus-proxy.ModBus(192.168.178.176:502): delay after connect: 0.1
2025-05-13 13:46:19,086    DEBUG modbus-proxy.ModBus(192.168.178.176:502): sending b'\x00\x1a\x00\x00\x00\x06\xf7\x03\x8c\xa1\x00\x03'
2025-05-13 13:46:29,093    ERROR modbus-proxy.ModBus(192.168.178.176:502): write_read error [2/2]: TimeoutError()
2025-05-13 13:46:29,094     INFO modbus-proxy.ModBus(192.168.178.176:502): closing connection...
2025-05-13 13:46:29,095     INFO modbus-proxy.ModBus(192.168.178.176:502): connection closed
2025-05-13 13:46:29,097     INFO modbus-proxy.Client(192.168.178.143:60568): closing connection...
2025-05-13 13:46:29,098     INFO modbus-proxy.Client(192.168.178.143:60568): connection closed
2025-05-13 13:46:38,860     INFO modbus-proxy.Client(192.168.178.143:52912): new client connection
2025-05-13 13:46:38,860    DEBUG modbus-proxy.Client(192.168.178.143:52912): received b'\x00\x1b\x00\x00\x00\x06\xf7\x03\x8c\xa1\x00\x03'
2025-05-13 13:46:38,861     INFO modbus-proxy.ModBus(192.168.178.176:502): connecting to modbus...
2025-05-13 13:46:38,865     INFO modbus-proxy.ModBus(192.168.178.176:502): connected!
2025-05-13 13:46:38,866     INFO modbus-proxy.ModBus(192.168.178.176:502): delay after connect: 0.1
2025-05-13 13:46:38,968    DEBUG modbus-proxy.ModBus(192.168.178.176:502): sending b'\x00\x1b\x00\x00\x00\x06\xf7\x03\x8c\xa1\x00\x03'
2025-05-13 13:46:48,971    ERROR modbus-proxy.ModBus(192.168.178.176:502): write_read error [1/2]: TimeoutError()
2025-05-13 13:46:48,972     INFO modbus-proxy.ModBus(192.168.178.176:502): closing connection...
2025-05-13 13:46:48,974     INFO modbus-proxy.ModBus(192.168.178.176:502): connection closed
2025-05-13 13:46:48,976     INFO modbus-proxy.ModBus(192.168.178.176:502): connecting to modbus...
2025-05-13 13:46:48,981     INFO modbus-proxy.ModBus(192.168.178.176:502): connected!
2025-05-13 13:46:48,982     INFO modbus-proxy.ModBus(192.168.178.176:502): delay after connect: 0.1
2025-05-13 13:46:49,086    DEBUG modbus-proxy.ModBus(192.168.178.176:502): sending b'\x00\x1b\x00\x00\x00\x06\xf7\x03\x8c\xa1\x00\x03'
2025-05-13 13:46:59,092    ERROR modbus-proxy.ModBus(192.168.178.176:502): write_read error [2/2]: TimeoutError()
2025-05-13 13:46:59,093     INFO modbus-proxy.ModBus(192.168.178.176:502): closing connection...
2025-05-13 13:46:59,095     INFO modbus-proxy.ModBus(192.168.178.176:502): connection closed
2025-05-13 13:46:59,096     INFO modbus-proxy.Client(192.168.178.143:52912): closing connection...
2025-05-13 13:46:59,097     INFO modbus-proxy.Client(192.168.178.143:52912): connection closed

FozzieUK, 2025-05-14:
The inverter simply isn't responding to the modbus request, assuming that the firmware update was the only thing done - it might be worth checking the slave id in inverter settings to make sure it hasn't been reset, and probably do a full restart of the inverter.

Also if you are using the USRW610 it has an annoying ability to change it's IP address when you least expect it.

qstadlmeier, 2025-05-14:
FoxESS confirmed in the meantime it is a bug in the new firmware. They are working on it.

closed 2025-08-13 by canton7