nathanmarlor/foxess_modbus #850: 1Komma5 Direct lan connection

BenySur, 2025-05-25:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### FoxESS - Modbus version

Master 1.06 Slave 1.0 Arm 1.05

### Inverter model

1k5-HI-8-V1

### Inverter manager version

Fox Cloud and Foxess App also 1komma5

### Describe the bug

Hello I have newly installed 1komma5 solar system, I can connect over cloud and did the integration to Home assistant but I'm looking for granularity so interested in Modbus, I have tried with the Wave share but wasn't successful I was getting the error 1k5-HI-8-V1 not compatible error, so I sent that back then after reading some more I recognized that I'm able to directly communicate over lan and I can poll data from the inverter. as seen on the picture below, and I have already manually integrated 1 sensor. Looking at the 1komma5 test data I believe the Adresse seem correct, what is the way that I can bypass the 1k5-HI-8-V1 not compatible error to use the addresses and sensors in the integration ?

Appreciate the support.
Regards.

[image]
[image]
[image]
[image]

fabian1512, 2025-05-26:
try to use modbus rtu over tcp

BenySur, 2025-05-26:
@FozzieUK Thanks a lot for the input it seems it was downloading latest stable version not the latest version

closed 2025-05-26 by BenySur