nathanmarlor/foxess_modbus #858: Max SoC setting will not be applied

domerich, 2025-06-06:
Hi guys, in summer I don't need all battery capacity so I want to cap it at 80%. But with multiple tries, it will accept the value but the batteries are still charged to 100%.

what could be the issue? I set the values now with foxcloud v1 instead but I don't want this to be the solution.

FozzieUK, 2025-06-06:
The inverter MaxSoC setting only affects grid charge, solar will continue to charge the batteries.

Best way to do handle this is to trigger on an SoC and either switch the workmode to Feed in or reduce the battery maximum charge current to 0 - obviously remembering to return it to the default value at the end of the day.

domerich, 2025-06-06:
thanks for this, a bit disappointing but with homeassistant I think I can do it.