nathanmarlor/foxess_modbus #878: v1.13.5 - brings back scaling issue on H3 Pro inverter

Damu74, 2025-08-03:
### Is there an existing issue for this?

- [ ] I have searched the existing issues

### FoxESS - Modbus version

1.13.5

### Inverter model

H3 Pro

### Inverter manager version

1.33

### Describe the bug

It seems the scaling (x10)  issue with *_total and *_today entities has returned in the 1.13.5 update. My H3 Pro inverter manager is running version 1.33. 
I immedatelly rolled back to foxess_modbus version 1.13.4 and updated the scale values as described here: https://github.com/nathanmarlor/foxess_modbus/issues/724#issuecomment-2586922485. 
Now, all *_total and *_today values are accurate again.

MattiasLindahl, 2025-08-04:
I have the same issue with my H3-PRO-20 running manager firmware 1.34. Every update I do for this integration I need to edit the scaling in file custom_components/foxess_modbus/entities/entity_descriptions.py

a900a900a900, 2025-08-05:
Same issue. 

[image] I’m not sure I pulled that much today. All the false data is reported from Fox ESS integration

hertelhaustechnik, 2025-08-06:
I simply edited the file as MattiasLindahl wrote, as indicated in the link from Damu74.
The line numbers are only slightly higher, but if you simply search for "H3_PRO", you will easily find the relevant lines and change the scale from 0.1 to 0.01.

Damu74, 2025-08-09:
nice hint @hertelhaustechnik.

I write down the necessary changes here... 

_solar_energy_total: change 0.1 to 0.0.1 **in line 1560**
_solar_energy_today: change 0.1 to 0.01 **in line 1591**
_battery_charge_total: change 0.1 to 0.01 **in line 1622**
_battery_charge_today: change 0.1 to 0.01 **in line 1673**
_battery_discharge_total: change 0.1 to 0.01 **in line 1704**
_battery_discharge_today: change 0.1 to 0.01 **in line 1748**
_feed_in_energy_tota: change 0.1 to 0.01 **in line 1780**
_feed_in_energy_today: change 0.1 to 0.01 **in line 1825**
_grid_consumption_energy_total: change 0.1 to 0.01 **in line 1857**
_grid_consumption_energy_today: change 0.1 to 0.01 **in line 1902**
_total_yield_total: change 0.1 to 0.01 in line: change 0.1 to 0.01 **in line 1934**
_total_yield_today: change 0.1 to 0.01 **in line 1966**
_input_energy_total: change 0.1 to 0.01 **in line 1998**
_input_energy_today: change 0.1 to 0.01 **in line 2030**
_load_power_total: change 0.1 to 0.01 **in line 2064**
_load_energy_today: change 0.1 to 0.01 **in line 2113**

Another Hint: 
HA core update to version 18.0 breaks the older version (1.13.4) of this integration completely. 
At least for my H3 Pro 15 the v 1.13.4 integration did not read any modbus register anymore. I updated to 1.13.5 and changed to line written here, and everything is working as expected.

Damu74, 2025-08-11:
same issue with v1.13.6

closed 2025-08-13 by canton7