nathanmarlor/foxess_modbus #884: Wrong readouts at invertor startup (  morning )  and invertor pre-shutdonw ( evening)

NOELV70, 2025-08-06:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### FoxESS - Modbus version

Version 1.13.5

### Inverter model

H3

### Inverter manager version

1.87

### Describe the bug

Hi,

Since the latest Home Assistant updates, an old issue with my inverter has resurfaced.

Normally, in the early morning—when the inverter is just starting up and the sun isn't fully active yet—I would see small negative power readouts (around -50 watts), which I assumed were normal idle readings.

However, after the recent updates, those small negative readings now seem to trigger very large positive power spikes instead. It appears the system is misinterpreting or mishandling the data during this early wake-up phase of the inverter.

Has anyone else experienced this? Any suggestions on how to fix or debug this would be appreciated.

Thanks!

[image]
[image]
[image]
[image]
[image]

NOELV70, 2025-08-11:
Hello,

The "Solar Generation Today"  (as shown in the pictures ) is displaying data "spikes" — (this has happened before  but disapeared with one of the updates 'fox-ess-pluging).

Is there a way to log or export the raw data so I can provide you with accurate information? Otherwise, it may turn into a trial-and-error process until the issue is fixed.

Thank you,
Noel

[image]

[image]

NOELV70, 2025-08-13:
Hello 

I have reviewed the code and added additional debug output to capture raw Modbus reads.

[image]

Initially, I assumed the issue was related to negative power readings, but I no longer believe that to be the case. It appears that something else is going wrong, specifically during inverter–solar startup in the morning.

The spikes observed in today’s energy generation are visible in the raw logs.

I have attached the logs for your reference, which include the raw Modbus data.

[image]

[image]

Best regards,
Noel

[raw-log.txt](https://github.com/user-attachments/files/21747105/raw-log.txt)
[log-grep-32000.txt](https://github.com/user-attachments/files/21747104/log-grep-32000.txt)

canton7, 2025-08-13:
You're right, that doesn't look like a negative value.

The inverters are known to do weird things on power-on. We have validation to reject some of these.

I guess we can fairly confidently say that a H1 can't generate more than 190kWh in 24h (the largest H1 has a capacity of 7.8kW), so we could tighten up the validation here.

canton7, 2025-08-13:
Please re-open if that doesn't fix it

closed 2025-08-13 by canton7

NOELV70, 2025-08-13:
> Please re-open if that doesn't fix it

This morning, I configured the system to limit output to 100 kW for my specific case (H3-8.0 inverter with 8 kW of solar panels) to test whether this would suppress the “error/no-response” readout.
For reference, the highest output I have observed over the past two years is ~70 kW, so a 100 kW limit should be sufficient for my use case.

The issue was not present at the time of configuration, so I will need to wait until the next occurrence to verify whether this limitation resolves it.

If the test is successful, my next step will be to add a second RS485 interface to the same Modbus network to monitor communication at the protocol level. The goal is to confirm whether the root cause lies within the inverter itself or elsewhere in the system code.

!!! **_Notably, this behavior only began appearing after the most recent Home Assistant updates; prior to that, the system ran without issue for over a year_.  ( same hardware config / setup) **!!! 

 {FYI : A similar issue did occur more than a year ago, but it was infrequent and I did not investigate it further at the time.}  If it reappears now, I want to capture the exact cause rather than rely on guesswork.

_{if you want worst case i can revert back to older HA ( 2024.06.x) version - have full backup - of the disk - to verify if it happens too on the older HA}_

Based on my current observations, the issue appears to occur during the inverter’s startup (warm-up) phase. In some cases, the inverter fails to respond entirely; in others, the response time is long enough that the next Modbus request is sent before the previous one completes. This behavior has never been observed during normal daytime operation—only during startup.

canton7, 2025-08-13:
> The goal is to confirm whether the root cause lies within the inverter itself or elsewhere in the system code.

We've seen this reported with inverters before, as they power-up at dawn (and don't have batteries).

NOELV70, 2025-08-16:
> > The goal is to confirm whether the root cause lies within the inverter itself or elsewhere in the system code.
> 
> We've seen this reported with inverters before, as they power-up at dawn (and don't have batteries).

I did revert back to older ha-2024.06.x ( and plugin version ) and guess what for 2 days the issue did NOT appear !! (solar invertor  firmware not changed) 

Did not find time yet to connect 2nd rs485 interface ( in listen only mode)

canton7, 2025-08-24:
From what we've seen, the inverter genuinely outputs nonsense values. I don't think there's anything we can do, other than try to spot them.

NOELV70, 2025-08-27:
The strange thing is that the problem has completely disappeared for the past three days. 

( yesterday i removed the 'error' filtering -- so if it will happen again it will also show up in energy control panel - makes it easy to spot if it happens)

About a week ago, I added raw-byte logging (both RX and TX), and at that time, the issue was still showing up—during inverter startup in the morning. 

But for some reason, it hasn’t occurred at all in the last three days. Interestingly, when I overthink it, something similar happened to me last year, probably around the same time of year ( and also disapeard by-itselfs).

NOTE: 'the' patch in the code-base here at github for H1 doesn’t work for me because I’m using H3. As mentioned before, it doesn’t really “filter”—if the values fall within the filter range, they will still pass through.

_If you ask for my opinion, I’d suggest removing the filtering code altogether, at least until we figure out what’s actually going on. Right now, it feels more like a half-working workaround than a real solution._

I’ll keep monitoring the situation and prepare some extra hardware as soon as possible—specifically, an additional RS485 interface so I can capture logs directly from the wire. 

_FYI: Just as an observation, I noticed that HA was creating automatic backups around the time the error readings occurred. Maybe that’s related, though at this point it’s just a wild guess—but you never know.

Regards 
Noel

commit 4c4f7a6b (2025-08-13): Limit solar_energy_today to 150kHw for H1

Fixes: #884