nathanmarlor/foxess_modbus #887: BMS kWh Remaining - Static Value

JAD92, 2025-08-08:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### FoxESS - Modbus version

1.13.5

### Inverter model

H1

### Inverter manager version

1.52

### Describe the bug

Apologies if not a bug - really appreciate this integration for monitoring the FoxEss hardware via modbus. 

For the entity BMS kWH Remaining (sensor.bms_kwh_remaining), I notice that the value is static, rather than showing charge left. 

I added a new battery slave yesterday, value increased as expected to 14.27 (approximately correct on 2.88 kWh modules). However this doesn't reduce as the battery is used throughout the evening (I was expecting the kWh remaining to reduce with use). 

If I've misunderstood the purpose of this sensor, please let me know (is it usable kWh left assuming degradation?)

canton7, 2025-08-13:
This is read directly from the inverter, as you say.

I have a H1 v1.70 and it seems to perform as expected:

[image]

Maybe try updating your inverter?

devpwnz, 2025-08-13:
I have a H3-12.0-E AUX and it's also static to me - 10.36. Reading the register at 31123 gives 1036. I'm using the latest 1.14.0b2 version that fixed Battery SoC and SoH. Also, It's been 20 minutes and it hasn't updated yet, so it seems that I only got the value at startup.

FozzieUK, 2025-08-13:
Discussed here https://github.com/nathanmarlor/foxess_modbus/issues/651#issuecomment-2388489149

Since October 24 all new V2 BMS firmware variants report residual battery differently (V1 BMS firmware is unchanged) - it now shows your current battery capacity. To calculate Residual, multiply it by SoC.

canton7, 2025-08-13:
Aha, thanks!