nathanmarlor/foxess_modbus #911: FoxESS KH10.5 EPS and Load entities reversed?

mjb-is, 2025-08-25:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### FoxESS - Modbus version

1.13.5

### Inverter model

FoxESS KH10.5

### Inverter manager version

1.43

### Describe the bug

In Home Assistant it seems that the Inverter Load and EPS data is reversed. I have my whole house on the EPS and have a secondary Zigbee based CT energy monitor for Grid and House.

The EPS entity seems to reflect the whole inverter load throughput, which includes the PV Power in, any Grid Power in during Force Charge, and Grid Power out during Force Discharge. Conversely, the Load entity agrees with my house monitor CT albeit its higher and seems to include the inverter internal load of 300-300W (2 inverters)

The FoxESS Cloud app seems correct on these figures so I can

[image]

 only surmise that the register data in the ModBus integration might have these reversed?

dave-seegreen, 2025-08-30:
I'm getting similar issues, i.e. what claims to be load is in fact EPS, however not sure they are reversed because in my case the EPS circuit is outputting very very similar values to the EPS values despite property having a considerable load at the time.

mjb-is, 2025-10-26:
> Yes I found a discrepancy for H3 and the correct registers to use, i will create a pr on the weekend.

Mine is on the K Series, KH10.5.

nathanmarlor, 2026-03-19:
Is this still an issue on the latest version (v1.14.0b3)? This may be related to #669 (EPS power register changes on newer firmware). If you can confirm the register values you're seeing for EPS vs Load, that would help with diagnosis.

nickw444, 2026-05-10:
This seems to still be a problem, running off of master (897c8026dd3a677ed5b5fc49e52fc72d14210776). EPS Power ~= Load Power. Voltage and current doesn't add up: `6 * 240 = 1440`

[image]

[image]

[image]