nathanmarlor/foxess_modbus #921: H3 Smart 10KW

kai19692024-sys, 2025-09-17:
I have an H3 Smart inverter. And I get the following error message:

Invalid registers detected
Fehler · Von FoxESS - Modbus gemeldet
Inverter (friendly name: "Fox H3Smart") has some registers which we expect to read, but can't. Details: [(31000, 15)], [(31022, 30)], [(32000, 24)], [(41000, 1)], [(41007, 5)].

Under Device Info it says H3-10.0-M - AUX von FoxESS. Has an incorrect inverter been detected?
Can the inverter be entered manually?

FozzieUK, 2025-09-17:
What version of the integration are you running ?

kai19692024-sys, 2025-09-17:
Version 1.14.0b3

phoet, 2025-09-18:
After restarting and upgrading all the things, I managed to get somewhat reasonable data from the integration, but there is still a register that cant be read:

Inverter (friendly name: "H3-Pro-15.0") has some registers which we expect to read, but can't. Details: [(37636, 64)].

Versions are Master 1.66, Slave 1.06, Manager 1.34

AndyDerNixKann, 2025-09-18:
Got this problem H3 Smart 8kw

Connection Status geändert zu Error reading registers. Type: RegisterType.HOLDING; start: 31000; count: 23; slave: 247 from tcp://192.168.178.65:502: Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 8 bytes (0 received)

FozzieUK, 2025-09-18:
> I reinstalled home assistant and I'm getting the H3-10.0-M - AUX von FoxES
> 
@kai19692024-sys 
Can you try and read the model number registers back please and post back what you get here :

To do that in home assistant use the 'Developer Tools', 'Actions', in the action box type 'Foxess - Modbus Read Registers', select your inverter in the drop down and enter the Start Address as 30000 and with a count 16, type = 'holding' 
f you could post an image of what it returns please.

FozzieUK, 2025-09-18:
`18483` = `48`(H), `33`(3)
`11569` = `2D`(-), `31`(1)
`12334` = `30`(0), `2E`(.)
`12333` = `30`(0), `2D`(-)
`19712` = `4D`(M), `00`(nul)

It really is reporting H3-10.0-M

Ok let's try something to see if the 'M' actually behaves like an 'H3 SMART', delete the Device in the integration

Edit the file `homeassistant/custom_components/foxess_modbus/inverter_profiles.py`

Change line 342 from
 `    InverterModelProfile(InverterModel.H3_SMART, r"^H3-([\d\.]+)-Smart").add_connection_type(`
to
`    InverterModelProfile(InverterModel.H3_SMART, r"^H3-([\d\.]+)-M").add_connection_type(`

Save the file, restart Home Assistant and re-add the device to the integration and let us know what you get.

AndyDerNixKann, 2025-09-18:
[image]

So i cant even find a line that is written like your example.

kai19692024-sys, 2025-09-18:
Now it works. Thank you for your help.

closed 2025-09-21 by kai19692024-sys