nathanmarlor/foxess_modbus #923: Wrong scale for Inverter Power?

phoet, 2025-09-20:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### FoxESS - Modbus version

latest

### Inverter model

H3-15 Pro

### Inverter manager version

1.66

### Describe the bug

It looks like Inverter Power is not scaled correctly. I'm new to Homeassistant, so a solution might be flipping a switch somewhere, but for now I'm stuck with weird graphs when comparing power consumption and generation.

The comment here suggests that mW is used https://github.com/nathanmarlor/foxess_modbus/blob/2b3107016bcc0cc2a563c0fb2dabcd573a2a1a9d/custom_components/foxess_modbus/entities/entity_descriptions.py#L868

But it looks to me like it's not.

[image]

The values in the graphs should be up to 15kW.

MartB, 2025-10-30:
Hey @phoet! Can you double check your H3 Pro manager version? It should be 1.33 or 1.39 or anything close to that, 1.66 seems like a master version to me.

phoet, 2025-11-16:
@MartB ah, that's exactly what I just did 😸

nathanmarlor, 2026-03-19:
Confirmed the `0.000001` scale is still on line 869 of `entity_descriptions.py` for the total inverter power on H3-PRO/H3-SMART (registers 39135/39134). The per-phase R/S/T registers all use `0.001`.

PR #962 proposes changing this to `0.001` but was requested to provide hardware evidence (register readings) before merging, since the original code has an explicit comment saying it was deliberately set. @phoet — can you share actual register values showing the current scale is wrong?